Koenkk/zigbee2mqtt #21399: IKEA Devices (SYMFONISK E2123 and TRADFRI GU10 CWS) Connectivity Issues

Lymestone, 2024-02-14:
### What happened?

Dear all, 

I am experiencing connectivity issues with two IKEA devices in my Zigbee2MQTT setup. While other devices like Lidl Smart Lights, Tuya Thermostat, and IKEA TRADFRI ON/OFF switch are working correctly and the server receives payloads from them without issues, however the IKEA SYMFONISK Button (E2123) and the IKEA TRADFRI Bulb GU10 CWS 345lm encounter problems maintaining a connection.

**I am using Zigbee2mqtt on HomeAssistant OS, Raspberry Pi 4 with a Conbee III stick.** 

[image]

**IKEA SYMFONISK Button (E2123):** This device connects to the network initially but fails to sustain the connection. No button inputs are recognized at all.
**IKEA TRADFRI Bulb GU10 CWS 345lm:** Similar to the SYMFONISK Button, this device can connect to the network but fails to update or maintain the connection afterward (can't change color, turn off) 

Troubleshooting Steps Tried:

Ensured that Zigbee2MQTT is up to date.
Checked the range and interference, placing the problematic devices closer to the coordinator.
Restarted Zigbee2MQTT and the MQTT broker.
Removed and re-paired the devices multiple times.
Checked logs for any errors related to these devices but found none that could explain the connectivity issues.

The thing bothering me me most is definitely the Tradfari button. It connects but when connected does not receive anything from the button. 

Thanks in advance. 

### Zigbee2MQTT version

1.35.3.1

### Adapter firmware version

0x264f0900

### Adapter

Conbee III 

### Setup

Home Assistant OS, Raspberry Pi 4

### Debug log

Error 2024-02-14 18:40:42Publish 'set' 'brightness' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genLevelCtrl.moveToLevelWithOnOff({"level":157,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (112))'
Info 2024-02-14 18:40:42MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":157,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (112))'","meta":{"friendly_name":"0x30fb10fffe0479f5"},"type":"zigbee_publish_error"}'
Error 2024-02-14 18:40:43Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (114))'
Info 2024-02-14 18:40:43MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (114))'","meta":{"friendly_name":"0x30fb10fffe0479f5"},"type":"zigbee_publish_error"}'
Error 2024-02-14 18:40:43Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (113))'
Info 2024-02-14 18:40:43MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (113))'","meta":{"friendly_name":"0x30fb10fffe0479f5"},"type":"zigbee_publish_error"}'
Error 2024-02-14 18:40:44Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (116))'
Info 2024-02-14 18:40:44MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (116))'","meta":{"friendly_name":"0x30fb10fffe0479f5"},"type":"zigbee_publish_error"}'
Error 2024-02-14 18:40:44Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (115))'
Info 2024-02-14 18:40:44MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (115))'","meta":{"friendly_name":"0x30fb10fffe0479f5"},"type":"zigbee_publish_error"}'
Error 2024-02-14 18:40:45Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (120))'
Info 2024-02-14 18:40:45MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (120))'","meta":{"friendly_name":"0x30fb10fffe0479f5"},"type":"zigbee_publish_error"}'
Error 2024-02-14 18:40:45Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received (119))'
Info 2024-02-14 18:40:45MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (119))'","meta":{"friendly_name":"0x30fb10fffe0479f5"},"type":"zigbee_publish_error"}'
[... cut ...]
Info 2024-02-14 18:40:55MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'color' to '0x30fb10fffe0479f5' failed: 'Error: Command 0x30fb10fffe0479f5/1 lightingColorCtrl.moveToColor({\"transtime\":0,\"colorx\":24313,\"colory\":15663}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (126))'","meta":{"friendly_name":"0x30fb10fffe0479f5"},"type":"zigbee_publish_error"}'

ctournayre, 2024-02-21:
+1 Same problem here, same SYMFONISK and TRADFRI with Conbee III.

zurnib, 2024-02-24:
+1 same here with STYRBAR and Slaesh's CC2652RB stick

ctournayre, 2024-02-26:
FIXED, found a workaround. For LED just after pairing successfully (all appears in HA) I switch off the LED physically (by turning the physical switch OFF). Then turn back ON and all the LED are working great now. No more desync.

Lymestone, 2024-02-26:
Does that also fix the issue with the Symfonisk (pulling battery shortly after pairing)?
If so, gonna give it a try when i get back from work.

closed 2024-09-24 by github-actions[bot]