Koenkk/zigbee2mqtt #21401: Moes switches ZS-B-EUX-XX-MS Missing functionality

jlpedrosa, 2024-02-14:
### What happened?

Switches: [ZS-B-EU1-GY-MS](https://moeshouse.com/products/eu-2-way-zigbee-light-switch-no-capacitor?variant=39796985692241) from MOES (1 indicates number of rockers [1,3]) GY indicates color.

The device is connected to the network correctly, identified as [Tuya TS003](https://www.zigbee2mqtt.io/devices/TS0013.html#tuya-ts0013). 

Issues found:
* The device, according to the manual has support for 2 way configuration, which means the relay must be settable to "decoupled" mode, but it's not possible.

* Unable to bind: I am not sure if this is an issue or not, but I was not able to bind the switch to a bulb (Ikea) for on/off. The UI disables it. There are 3 clusters reported on the UI, but on the enpoints there are 6 listed: 1,2,3,left,center,right, I tried all combinations.

### What did you expect to happen?

Be able to configure the switch in "decoupled" mode.
Be able to create bindings (unsure)

### How to reproduce it (minimal and precise)

* Add the device, connect zigbee2mqtt.
* Check missing controls on device page
* Go to bind section, select any endpoint 1->3, left->right and select any device, binding is not possible.

### Zigbee2MQTT version

1.35.2 commit: 06b3395

### Adapter firmware version

20230507

### Adapter

zStack3x0 (development board)

### Setup

Container standalone amd64

mauritsivs, 2024-03-19:
Same here, I have many of these MOES devices, with two and three buttons. The three button version used to be identified correctly. But now is identified as tuya "no neutral" ts0013, while it is a MOES with neutral wire.
I am no longer able to change the led back light mode as before. 
No decoupling possible anymore
No binding possible. 
Somewhere between December and today, something has changed. My entire HA z2m setup was turned off for months because of renovations of my house.
I can't find it either in "supported devices" list anymore... 
What happened?

grabbou, 2024-07-14:
I have problem with the 3gang specifically identifying as Tuya and not working properly. I'll try to figure this out and post an update

mlewando, 2024-09-15:
Same here, no decoupled mode, no way to bind... :( But I think that it's device or device frimware issue... Anyway, if anyone would manage to fix/improve this, that would be awesome :)

thepaulf, 2024-09-15:
> Same here, no decoupled mode, no way to bind... :( But I think that it's device or device frimware issue... Anyway, if anyone would manage to fix/improve this, that would be awesome :)

I've solved it with mine, will let you know in a few hours when I have the time how I did it.

mlewando, 2024-09-17:
I found some alternative frimware (https://github.com/doctor64/tuyaZigbee) but it's only for batteries powered pilots afaik and I'm a bit scared to not brick the device...

how did you mange to solve the problem @thepaulf ?

thepaulf, 2024-09-17:
But can you connect the switch to your instance? Set it in connection mode by holding the left button? Or try pressing 10 times.

I fixed the functionality of the switch with a custom config as far as I remember, but if you can't connect that won't work.

mlewando, 2024-09-18:
I am able to connect them and in general, they work... the issues are:
- binding is not supported, as the genOnOff cluster seems to be only input one
- I cannot switch the button without switching rely, which is a bit annoying and you can hear a "click" sound on every click, even when the switch is meant to handle eg. AC (I don't need a relay for this). So, no "decoupled mode"

there is also one more small inconvenience - you cannot push two buttons on 2-gang switch at once (eg. to turn off 2 lights), you have to do it one-by-one

are you able to fix any of those issues with those custom converters?

thepaulf, 2024-09-18:
Yes that is fixed with the converter. Only add the converter you need and it will work perfectly

closed 2025-01-07 by github-actions[bot]

mrdarrengriffin, 2025-01-20:
Does anyone have a way of achieving decoupled mode using ZHA?

wizzyto12, 2026-01-14:
> Good

Holy smokes man, I was so disappointed of these switches I thought of giving them away to friends, but I just flashed one and its everything I ever ask it to be! Thanks a lot for enlightening me about this repo. 

Жив и здрав!