Koenkk/zigbee2mqtt #21402: [New device support]:

Justdigit, 2024-02-14:
### Link

https://nl.aliexpress.com/item/1005006267870912.html?spm=a2g0o.order_list.order_list_main.122.5d2f79d2mJKxwN&gatewayAdapt=glo2nld

### Database entry

cant find folders

### Comments

**This is how the RGB  controller is regonized.**

const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0503B'],
    model: 'TS0503B',
    vendor: '_TZ3210_odlghna1',
    description: 'Automatically generated definition',
    extend: [light({"color":{"modes":["xy","hs"]}})],
    meta: {},
};

module.exports = definition;

**But its actually this device:**

https://www.zigbee2mqtt.io/devices/GL-C-003P.html

### External definition

```shell
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0503B'],
    model: 'TS0503B',
    vendor: '_TZ3210_odlghna1',
    description: 'Automatically generated definition',
    extend: [light({"color":{"modes":["xy","hs"]}})],
    meta: {},
};

module.exports = definition;
```

mrmaximas, 2024-02-15:
@Justdigit depend of you OS please find z2m folder and file like this /zigbee2mqtt/data/database.db

Justdigit, 2024-02-15:
> @Justdigit depend of you OS please find z2m folder and file like this /zigbee2mqtt/data/database.db

Found the entry!!
{"id":13,"type":"EndDevice","ieeeAddr":"0x00158d0002cb2681","nwkAddr":23540,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b186acn01","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,25,65535,18],"outClusterList":[0,4,3,5,25,65535,18],"clusters":{"genBasic":{"attributes":{"65281":{"1":3175,"3":31,"4":424,"5":29,"6":[0,1],"8":5129,"10":0},"modelId":"lumi.remote.b186acn01","appVersion":9,"manufacturerName":"LUMI","powerSource":3}},"genMultistateInput":{"attributes":{"presentValue":2}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":9,"interviewCompleted":true,"meta":{},"lastSeen":1707863086056}

mrmaximas, 2024-02-15:
it’s not a TS0503B db record

Justdigit, 2024-02-15:
> it’s not a TS0503B db record

Well that's the problem.... It's a different device... The device with picture doesnt correspond with the actual item.

mrmaximas, 2024-02-16:
You need to find record for 0x4c13855c9112a68

Justdigit, 2024-02-16:
> You need to find record for 0x4c13855c9112a68

ahhhh sorry i thought "ID 13" corresponded with device 13 in ZIgbee2mqtt Frontend.

{"id":14,"type":"Router","ieeeAddr":"0xa4c13855c9112a68","nwkAddr":27466,"manufId":4417,"manufName":"_TZ3210_odlghna1","powerSource":"Mains (single phase)","modelId":"TS0503B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0503B","manufacturerName":"_TZ3210_odlghna1","powerSource":1,"zclVersion":3,"appVersion":101,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":9,"colorMode":1,"currentX":28436,"currentY":32820,"currentHue":41,"currentSaturation":54}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":101,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1708024823788}

closed 2024-09-14 by github-actions[bot]