Koenkk/zigbee2mqtt #21405: Legrand 412015 reports old firmware with installed_version greater than latest_version, can't update OTA

Ivano62, 2024-02-14:
### What happened?

Installed firmware on Legrand 412015 is old (build date is 20190709), firmware version appears as unknown characters (see image belove), but firmware installed_version is greater than latest_version, so OTA update is not possible.

Example json:

{
> "last_seen": "2024-02-14T23:12:36+01:00",
    "linkquality": 255,
    "power": 261,
    "power_alarm_active": false,
    "power_alarm_active_value": 0,
    "power_alarm_enabled": 1,
    "power_alarm_wh_threshold": 6600,
    "power_apparent": null,
    "update": {
        "installed_version": 54531584,
        "latest_version": 5456895,
        "state": "idle"
    },
    "update_available": false
>
}

[image]

### What did you expect to happen?

Correct build numbers so that I can update the firmware or a workaround

### How to reproduce it (minimal and precise)

Legrand 412015 with old firmware installed, go to OTA page on Zigbee2Mqtt and "check for new updates", a notification with "OTA: Firmware on '_... addr ..._' ( DIN power consumption module) is newer than latest firmware online." will appear.

### Zigbee2MQTT version

1.35.3-dev commit: 8a9477f

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

plain on docker container

pickard1, 2024-03-08:
Same issue here with a Legrand shutter 067776

[image]

How can i force an update with the online version ?

Kali-, 2024-04-17:
I have the same issue, but my firmware is like 3m newer then your

[image]

kaleofr, 2024-04-22:
Same problem here:
I have 11 Legrand 067776 that have been updated from Fw 01a (067776_inverted) to Fw 002f (067776) without problems. 
I am currently updating them to Fw 003C, no problems.
 
But I also have 6 other Legrand 067776 with Fw "014  !����!" from "20190627" that can't update (z2m says FW newer than the latest online)
Is there a simple way to force fw update?

Regards,

Kaléo.

Harckange, 2024-04-23:
Hello,

Same issue here with Legrand 064882 :( 

[image]

kaleofr, 2024-04-24:
Hello,
So, there is no easy way to force a FW update (button to force, selection of an existing compatible Fw for exemple)... but the way to do it is not hard.
I did a local OTA update following this: 
https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files
You can also find some tuto on the web.

1-Add in the configuration.yaml
```
ota:
  zigbee_ota_override_index_location: my_index.json

```
2-Create a my_index.json file in the same directory of configuration.yaml
here is mine:
```
[
    {
       "url": "ota/1021-0013-002f4203-NLV.zigbee",
        "path": "ota/1021-0013-002f4203-NLV.zigbee",
	"force": true
    }
]
```

I choose only the oldest FW available in the repository (002F -> 1021-0013-002f4203-NLV.zigbee for my 067776) 
Next updates should work without local OTA update forcing.

3-Create a directory (ota/ in the same place than configuration.yaml for me, but it's as you want) and place your fw file inside.
4-Restart z2m (maybe not usefull, I don't know)
5-Start your update
6-You can remove the local OTA update configuration (1,2, 3, and maybe 4)

I hope this may help you.

Kaléo.

pickard1, 2024-04-25:
This has worked for me ... thanks for helping

Fabrice

Ivano62, 2024-04-25:
> Hello, So, there is no easy way to force a FW update (button to force, selection of an existing compatible Fw for exemple)... but the way to do it is not hard. I did a local OTA update following this: https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files You can also find some tuto on the web.
> 
> 1-Add in the configuration.yaml
> 
> ```
> ota:
>   zigbee_ota_override_index_location: my_index.json
> ```
> 
> 2-Create a my_index.json file in the same directory of configuration.yaml here is mine:
> 
> ```
> [
>     {
>        "url": "ota/1021-0013-002f4203-NLV.zigbee",
>         "path": "ota/1021-0013-002f4203-NLV.zigbee",
> 	"force": true
>     }
> ]
> ```
> 
> I choose only the oldest FW available in the repository (002F -> 1021-0013-002f4203-NLV.zigbee for my 067776) Next updates should work without local OTA update forcing.
> 
> 3-Create a directory (ota/ in the same place than configuration.yaml for me, but it's as you want) and place your fw file inside. 4-Restart z2m 5-Start your update 6-You can remove the local OTA update configuration (1,2, 3, and 4)
> 
> I hope this may help you.
> 
> Kaléo.

Ok, thanks, but where I can find the firmware fo rmy Legrand 412015?

Harckange, 2024-04-25:
> Ok, thanks, but where I can find the firmware fo rmy Legrand 412015?

https://github.com/Koenkk/zigbee-OTA/tree/master/images/Legrand

But you have to identify the kind of device (NLC = Cable Outlet, NLM = Micromodule, ...) : 

https://developer.legrand.com/forums/topic/list-of-hw_type-for-home-control-api/

Harckange, 2024-04-25:
> Hello, So, there is no easy way to force a FW update (button to force, selection of an existing compatible Fw for exemple)... but the way to do it is not hard. I did a local OTA update following [...]

Thank you for your answer. I followed the (simple) steps but Z2M still tells me that the firmware is newer bla bla bla....
I don't see where I'm wrong... Basically you tell Z2M to force update this firmware on this device and... voila but nope....

My configuration.yml :

[image]

my_index.json (devide is a cable outlet 064882 so NLC device) : 

[image]

Ivano62, 2024-04-25:
> https://github.com/Koenkk/zigbee-OTA/tree/master/images/Legrand
>
> But you have to identify the kind of device (NLC = Cable Outlet, NLM = Micromodule, ...) : [...]

my Legrand 412015 is a [DIN power consumption module](https://www.zigbee2mqtt.io/devices/412015.html#legrand-412015), so 

I can't find it on the page you kindly indicated.

It seems a "DIN power consumption module" -  POWER - NLPC - LG-412015 BT-F20T60A (https://developer.legrand.com/local-interoperability/#Compatible%20products).

Could you help me?

Thanks

Harckange, 2024-04-26:
> my Legrand 412015 is a [DIN power consumption module](https://www.zigbee2mqtt.io/devices/412015.html#legrand-412015), so
>
> I can't find it on the page you kindly indicated.
>
> It seems a "DIN power [...]

Maybe have a look here : https://developer.legrand.com/production-firmware-download/

acidric, 2024-09-03:
Hi,
Exactly the same problem here. I tried Harckange's solution but unfortunately, it's doesn't work for me (with 
images/Legrand/1021-0011-005a45ff-NLP.zigbee )
[image]

I'have this result:
[image]

if anyone has any ideas please ?

closed 2025-01-07 by github-actions[bot]