Koenkk/zigbee2mqtt #21408: PIR sensor HW500A not working on Zigbee2MQTT

eabraga69, 2024-02-15:
### Link

https://expo.tuya.com/product/897029

### Database entry

{"id":12,"type":"EndDevice","ieeeAddr":"0xa4c13861029a1c7d","nwkAddr":59929,"manufId":4417,"manufName":"_TZ3000_bsvqrxru","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x04cd15fffec080b2","zoneState":1,"currentZoneSensitivityLevel":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x04cd15fffec080b2","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":491743244},"lastSeen":1665616820098,"defaultSendRequestWhen":"immediate"}

### Comments

HW500A PIR sensor does not work well on Zigbee2MQTT. It is possible to insert it, but it does not allow sensitivity settings. sometimes he also acts without presence.

### External definition

```shell
const {batteryPercentage} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0202'],
    model: 'TS0202',
    vendor: '_TZ3000_bsvqrxru',
    description: 'Automatically generated definition',
    extend: [batteryPercentage()],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-02-15:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/6e17a3d2aa0b2d07dd026b92442450b5): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

eabraga69, 2024-02-18:
Thanks a lot @Koenkk .  The funcionality "occupacy timeout" is now available on specific configurations of z2mqtt.

closed 2024-02-19 by Koenkk