Koenkk/zigbee2mqtt #21409: All Zigbee devices stop working after upgrade to 1.35.3.1 from 1.35.2.1

miguelpineiroo, 2024-02-15:
### What happened?

I have 1.35.2.1 working with Conbee II and after upgrade to 1.35..3.1, all Zigbee devices failed to response. HA reset didn’t help. I needed to revert back to 1.35.2.1 and they are working again 

### What did you expect to happen?

1.35.3.1 working properly as 1.35.2.1

### How to reproduce it (minimal and precise)

Upgrade to 1.35.3.1 from 1.35.2.1 with Conbee II on Raspberry PI 4B with HA OS

### Zigbee2MQTT version

1.35.2.1

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Setup

Ad on on raspberry pi 4B x64

niziak, 2024-02-15:
Confirmed. Restarting add-on multiple times doesn't help. **Reboot of machine (not HA restart) helps**.
After restarting of add-on log file from add-on shows lots of MQTT messages with devices list from Conbee II adapter. But then any communication with Zigbee devices fails.

stephdust, 2024-02-15:
same problem for me, from 1.35.1 to 1.35.3, with a conbee as well.
hopefully I made a backup of 1.35.1 before

MarkusFelix, 2024-02-15:
Same problem but stand-alone installation with two different instances, two different networks and two different coordinators - my whole house is gone since yesterday...

fidodido1202, 2024-02-15:
I can also confirm updates from 1.35.2 to 1.35.3 and all devices failed to communicate to zigbee with an  Conbee II

s3frank, 2024-02-15:
I don’t think this is fixed yet. I’ve tried all versions from the official channel (not beta) and not been able to maintain the connection to the conbee2. I’m staying on 1.35.1-1 for now which is what I was on before this broke.

MrOmnipotenz, 2024-02-15:
Can confirm exactly Miguels issue, Conbee II, same Versions, HP ProDesk instead if RasPi

kev300, 2024-02-16:
Also for me. In addition: the web interface is also broken.

miguelpineiroo, 2024-02-16:
Solved for me with HA core 2024.2.1 (previously I had the last version of Jan): I updated the HA core version, then moved to Zigbee2MQTT 1.35.3 and now everything is fine.

stephdust, 2024-02-16:
I use HA core 2024.2.1 already but no luck :-(

s3frank, 2024-02-16:
When you say solved does that mean stable for more then 24 hours?

Best regards,

-FF

miguelpineiroo, 2024-02-16:
Only 14h stable, not 24 yet. Finger crossed. If fails again, I will update here :)

AiYoriAoshi, 2024-02-17:
I'm running HA 2024.2.2, Z2M 1.35.3-1 and ConBee 2 and now also 3. Regardless of the dongle, all Mains powered devices and like half of the battery powered devices are unreachable. When running deconz on a Windows laptop next to the server, it immediately starts picking up devices, mains and battery powered with both dongles. Restoring HA to 2024.1 and Z2M to 1.34.0 did NOT help anymore. 

I also set up a new HA VM with 2024.2.2 and Z2M 1.35.3-1 with only MQTT and Z2M installed it no devices were picked up. The closest mains powered device is 10cm away. Thinking of trying ZHA on the new VM.

psa-anddev, 2024-02-17:
Same here. I'm running HomeAssistant OS in a VM in my Proxmox server with a Conbee II stick. Running HomeAssistant 2024.02.2 (but it was already happening with 2024.02.1). The Zigbee2MQTT version is 1.35.3.1. It started with automations randomly failing. At this stage I can see the initial status of all my devices but sometimes they don't update and when HomeAssistant is trying to send commands to them, I get errors in the logs of Zigbee2MQTT.  

```
Zigbee2MQTT:info  2024-02-17 19:15:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to 'Hallway accent light' failed: 'Error: Command 0x003c84fffed9ce76/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (no response received (190))'","meta":{"friendly_name":"Hallway accent light"},"type":"zigbee_publish_error"}'
```

I also see errors from Deconz when I try to control my blinds using HomeAssistant. These errorrs end up referrencing the library used to interact with Zigbee. When I use other means to change the state of devices, these state changes are not reflected in the Zigbee2MQTT interface (which is broken in this version). 

Lastly, I had the motion sensors (Aqara P1) which I use to control some lighting stopped updating. I restarted the whole Home Assistant VM and manage to get them to update again but everything else is not really working. 

I'm sorry if I sound harsh. I really really like Zigbee2MQTT and it's been working for me great since I switched to it when I moved last year. It's always been reliable and I really love the product. If I manage to collect more data about the problem, I'll make sure I post it.

miguelpineiroo, 2024-02-17:
Working for me with Zigbee2MQTT 1.35.3.1 HA 2024.2.2 (also with 2024.2.1) and Conbee II, on Raspberry pi 4 x64 with HAOS, but not sure if moving to latest HA core has been the fix or not

s3frank, 2024-02-17:
For how long stable?

Best regards,

-FF

miguelpineiroo, 2024-02-18:
Any problem for 3 days

Koenkk, 2024-02-18:
This should be fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) already

psa-anddev, 2024-02-18:
As a fellow developer, I know how bad is to try to estimate anything but, do you know when this fix will be available in the main branch?

MrOmnipotenz, 2024-02-18:
> This should be fixed in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) already

Hi Koenkk,
I am running edge since yesterday, initially want to give it a try.
Didnt work on yesterday, as it didnt work on today.
Deleted and reinstalled it.

AiYoriAoshi, 2024-02-18:
> I'm running HA 2024.2.2, Z2M 1.35.3-1 and ConBee 2 and now also 3. Regardless of the dongle, all Mains powered devices and like half of the battery powered devices are unreachable. When running [...]

I replaced the 1.35.3-1 Version withe Edge on new VM and devices were paired.
I merged the configuration with the production configuration.yaml keeping the device list and and throwing away the database.db and state.json.
Starting to pair with a fresh database, devices are picked up again.

Not sure why it didn't work with the fresh VM and 1.35.3-1 Version.

Koenkk, 2024-02-18:
@psa-anddev 1 March

psa-anddev, 2024-02-18:
Thank you @Koenkk. I look forward to the update. I'll make sure to give an update on this thread of whatever changes.

s3frank, 2024-02-18:
Ditto, until then i am following release channel and stay on 1.35.1

wilotas, 2024-02-19:
Hi guys, 

Just downgraded to 1.35.1 via container. It is still not working

Do we need to do any special action to make it working?

Regards

s3frank, 2024-02-19:
When I use that version and reboot my host it stays stable after that for
days and days.

Best regards,

-FF

closed 2024-02-22 by Koenkk