Koenkk/zigbee2mqtt #21412: Invalid color mode 'hs' received for entity shows in logs for MiBoxer FUT035Z+

spitfire, 2024-02-15:
### What happened?

I have two [MiBoxer FUT035Z+](https://www.zigbee2mqtt.io/devices/FUT035Z+.html) LED controllers connected to plain white LED strips (with just brightness regulation), and my HA core logs is getting spammed with this:
```
2024-02-15 12:39:00.701 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'hs' received for entity light.kuchnia_ledy_prawe
2024-02-15 12:39:01.554 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'hs' received for entity light.kuchnia_ledy_lewe
```

### Zigbee2MQTT version

1.35.3 (from HA Addon)

### Adapter firmware version

20230507

### Adapter

UZG-01 with CC2652P7

### Setup

HAOS, x64 in a VM

spitfire, 2024-08-15:
This is still showing up in logs

spitfire, 2024-12-30:
It is still showing up:
```
Rejestrator: homeassistant.components.mqtt.light.schema_json
Źródło: components/mqtt/light/schema_json.py:365
integracja: MQTT (dokumentacja, Problemy)
Pierwsze zdarzenie: 26 grudnia 2024 22:42:49 (10346 zdarzenia)
Ostatnio zalogowany: 01:04:14

Invalid color mode 'hs' received for entity light.kuchnia_ledy_lewe
Invalid color mode 'hs' received for entity light.wiata_swiatlo_zewnetrzne
Invalid color mode 'hs' received for entity light.swiatla_w_kuchni
Invalid color mode 'hs' received for entity light.kuchnia_ledy
Invalid color mode 'hs' received for entity light.kuchnia_ledy_prawe
```

Xitro01, 2025-02-24:
Same issue with the same device. It solved it for me (temporarily???) by selecting a color temperature in HA.
Probably will return with a restart or update of Zigbee2MQTT, but not sure.

spitfire, 2025-02-24:
> Same issue with the same device. It solved it for me (temporarily???) by selecting a color temperature in HA. The problem will probably return with a restart or update of Zigbee2MQTT, but not sure.

Can you let me know if it returned?

I'm using plain LEDs without CW/WW adjustment, I don't know what would be the impact of changing this setting.

Xitro01, 2025-02-24:
> Can you let me know if it returned?
>
> I'm using plain LEDs without CW/WW adjustment, I don't know what would be the impact of changing this setting.

It is not changing a setting. Just reapplying you color temp in HA itself.

closed 2025-05-04 by github-actions[bot]