Koenkk/zigbee2mqtt #21414: EARU EAKCB-T-M-Z values not settable

erazorv, 2024-02-15:
### What happened?

Hi,
when I try to change values:
- Temperature threshold
- Temperature breaker
- Over current threshold ... and so on.

I get following error message:
 -> publish 'set' 'power_threshold' to '0xe0798dfffee0e56b' failed: 'Error: Expected string or number, got: undefined'

https://www.zigbee2mqtt.io/devices/EAKCB-T-M-Z.html

### Zigbee2MQTT version

1.35.3 

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Setup

Home Assistant OS

finipini, 2024-02-15:
The same thing is happening to me with that device

finipini, 2024-02-15:
> failed: 'Error: Expected string or number, got: undefined'

Update: I was having that error while testing the device with no load..... then i remembered that devices with energy reports often report "non" from power measurements until something is connected to the device. 

Seems like this device does that too even for all the settings it have. I connected a fan with 60watts and all the values are no settable.

erazorv, 2024-02-16:
oh your are right...
with load I can set the values :/

Thanks

closed 2024-02-18 by erazorv