Koenkk/zigbee2mqtt #21419: [Wrong device]: FUT037ZP+ identified as TuYa TS0505B_1

crhbetz, 2024-02-15:
### Link

https://miboxer.com/de/produkt/3-in-1-led-controller-zigbee-3-0-2-4g-output-max-20a

### Model

FUT037ZP+

### Description

Zigbee & RF 3 in 1 LED controller

### Vendor

MiBoxer

### Picture (link)

https://miboxer.com/wp-content/uploads/a1-111.jpg

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0xa4c138d6c4b33027","nwkAddr":52242,"manufId":4417,"manufName":"_TZB210_wxazcmsh","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":29}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":4,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":768,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":61184,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":25,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1},{"cluster":4096,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[{"cluster":33,"type":"endpoint","deviceIeeeAddress":"0x00212effff018011","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1842547789},"lastSeen":1708033771752}

### Notes

Using it in "blue" / RGB-CCT mode. Working perfectly, no issues found.

Pairing and/or using the other modes did not seem to visibly change anything on the z2m side. Always identified as TuYa TS0505B_1 with all controls available in z2m, but having no effect if the controller-side mode did not allow it (for example: RGBW mode = white color temperature won't change).

Let me know if I can help any further; might be slow to respond.

Eoras, 2024-06-01:
Hello, same probleme here !

And i don't know with you @crhbetz but my effects doesn't work, when i try to select one of this : 
[image]

Nothing happens.
Are you doing the same thing as me?

And when I change the option: Color power on behavior to previous it doesn't work, and it always turns back on in "classic" white even if it was turned off in the blue position.

[image]

To fix this Color power on behavior problem I deleted my device, and I reconnected it and I did not touch this option and there it works because in the state there is no longer this parameter .

What bothers me the most is the effects.

Eoras, 2024-06-01:
@Koenkk maybe it's possible to add _TZB210_wxazcmsh to https://www.zigbee2mqtt.io/devices/FUT037Z.html device ?

crhbetz, 2024-06-01:
Hi,

unfortunately I currently can't test. I did a test setup in February to confirm basic function, then made this issue. The house is currently being remodeled and my test setup is not available. The device will be installed some time this month or so.

I think the effects indeed **didn't** work, as you say. Didn't test the color power on behavior.

I hope your PR [#7592](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7592) will solve it!

closed 2024-06-02 by Koenkk

crhbetz, 2024-08-01:
Dear @Eoras - do effects now work for you? The device now identifies as [FUT037Z+](https://www.zigbee2mqtt.io/devices/FUT037Z%2B.html#miboxer-fut037z%252B) for me, but function hasn't changed. z2m version is [1.39.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.39.0) commit: [03269267](https://github.com/Koenkk/zigbee2mqtt/commit/03269267)

Eoras, 2024-08-01:
Effects doesn't work for me too... maybe is necessary to re-open this issue

Koenkk, 2025-12-02:
`FUT037Z+` uses the same code as `TS0505B_1`, it seems this device simplify doesn't support effects.