Koenkk/zigbee2mqtt #21424: [New device support]: MOES ZB-LZD7-RCW-MS

tuxiano, 2024-02-16:
### Link

https://moeshouse.com/products/zigbee-smart-light-zb-lzd7-rcw-en?variant=39331508093009

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0xa4c1380db534ba89","nwkAddr":47000,"manufId":4107,"manufName":"_TZ3210_s9lumfhn","powerSource":"Unknown","modelId":"TS0505B","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,6280,4,5,6,8,768,10,4096,65518],"outClusterList":[25,4096,65518],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_s9lumfhn","powerSource":0,"zclVersion":3,"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"z.1.0"}},"genLevelCtrl":{"attributes":{"61440":0,"currentLevel":254}},"lightingColorCtrl":{"attributes":{"57344":0,"57345":[0,65536113],"61453":1,"colorCapabilities":29,"tuyaRgbMode":0,"colorMode":2,"currentX":10040,"currentY":3116,"currentHue":80,"currentSaturation":254,"colorTemperature":143}},"genOnOff":{"attributes":{"61440":0,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"z.1.0","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1842547789},"lastSeen":1708073311983} {"id":3,"type":"Router","ieeeAddr":"0xa4c13826bb2ce8ff","nwkAddr":37181,"manufId":4107,"manufName":"_TZ3210_s9lumfhn","powerSource":"Unknown","modelId":"TS0505B","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,6280,4,5,6,8,768,10,4096,65518],"outClusterList":[25,4096,65518],"clusters":{"genLevelCtrl":{"attributes":{"61440":0,"currentLevel":254}},"genBasic":{"attributes":{"hwVersion":0}},"lightingColorCtrl":{"attributes":{"57344":0,"57345":[0,65536113],"61453":1,"colorCapabilities":29,"tuyaRgbMode":0,"colorMode":2,"currentX":29111,"currentY":33855,"currentHue":80,"currentSaturation":254,"colorTemperature":143}},"genOnOff":{"attributes":{"61440":0,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"z.1.0","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1842547789},"lastSeen":1708073311384} {"id":4,"type":"Router","ieeeAddr":"0xa4c138ba3776b26b","nwkAddr":1780,"manufId":4107,"manufName":"_TZ3210_s9lumfhn","powerSource":"Unknown","modelId":"TS0505B","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,6280,4,5,6,8,768,10,4096,65518],"outClusterList":[25,4096,65518],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_s9lumfhn","powerSource":0,"zclVersion":3,"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"z.1.0"}},"lightingColorCtrl":{"attributes":{"57344":0,"57345":[0,65536113],"61453":1,"colorCapabilities":29,"tuyaRgbMode":0,"colorMode":2,"currentX":10040,"currentY":3116,"currentHue":80,"currentSaturation":254,"colorTemperature":143}},"genOnOff":{"attributes":{"61440":0,"onOff":0}},"genLevelCtrl":{"attributes":{"61440":0,"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"z.1.0","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1842547789},"lastSeen":1708073322005} {"id":5,"type":"Router","ieeeAddr":"0xa4c138916153ee04","nwkAddr":31263,"manufId":4107,"manufName":"_TZ3210_s9lumfhn","powerSource":"Unknown","modelId":"TS0505B","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,6280,4,5,6,8,768,10,4096,65518],"outClusterList":[25,4096,65518],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_s9lumfhn","powerSource":0,"zclVersion":3,"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"z.1.0"}},"lightingColorCtrl":{"attributes":{"57344":0,"57345":[0,65536113],"61453":1,"colorCapabilities":29,"tuyaRgbMode":0,"colorMode":2,"currentX":10040,"currentY":3116,"currentHue":80,"currentSaturation":254,"colorTemperature":143}},"genLevelCtrl":{"attributes":{"61440":0,"currentLevel":254}},"genOnOff":{"attributes":{"61440":0,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"z.1.0","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1842547789},"lastSeen":1708073323162}

### Comments

Devices get wrongly recognized as Moes ZB-LZD10-RCW:
https://www.zigbee2mqtt.io/devices/ZB-LZD10-RCW.html#moes-zb-lzd10-rcw

According to the debug log the lightingColorCtrl converter is missing.

Debug 2024-02-16 09:54:23No converter available for 'ZB-LZD10-RCW' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,243,10,7,240,72,0],"type":"Buffer"}'
Debug 2024-02-16 09:54:23No converter available for 'ZB-LZD10-RCW' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,244,10,7,240,72,0],"type":"Buffer"}'
Debug 2024-02-16 09:55:08No converter available for 'ZB-LZD10-RCW' with cluster 'lightingColorCtrl' and type 'raw' and data '{"data":[8,153,10,7,240,72,0],"type":"Buffer"}'

Additionally I observed additional issues:
- It is very unreliable set colours etc. to individual bulbs. On the other hand broadcast commands (e.g. setting colours of a group of bulbs works perfectly fine
- Something is wrong with the brightness control, because 1% brightness is still very bright (feels like 20% - 30%)
- No OTA firmware updates
- Breath effect is actually blink effect
- Sync between color mode xy and hs is not working. 

If this helps, I'm totally fine to donate a downlight, because I have almost 40 of these bulb and I would greatly benefit if this bulb would get fully support by z2m.

### External definition

```shell
z2m generates automatically the following definition

const {philipsLight} = require('zigbee-herdsman-converters/lib/philips');

const definition = {
    zigbeeModel: ['TS0505B'],
    model: 'TS0505B',
    vendor: '_TZ3210_s9lumfhn',
    description: 'Automatically generated definition',
    extend: [philipsLight({"colorTemp":{"range":[143,500]},"color":{"modes":["xy","hs"]}})],
    meta: {},
};

module.exports = definition;
```

luca-iodice, 2024-07-15:
Same problem here.
It appears there is also a new version of the light _TZ3210_dwzfzfjc

russbrenner, 2024-09-05:
Ditto here. Did you have any luck finding a resolution?

tuxiano, 2024-10-07:
@Koenkk Hi, I just noticed that you have published a definition for the 10W ([Moes ZB-LZD10-RCW](https://www.zigbee2mqtt.io/devices/ZB-LZD10-RCW.html#moes-zb-lzd10-rcw)) version of device mentioned in this issue. Is it possible to map the definition of the 10Watt version to the 7Watt version, too?

closed 2025-01-07 by github-actions[bot]