Koenkk/zigbee2mqtt #21427: Could not load theme

mmattel, 2024-02-16:
### What happened?

Interestingly, it worked before, with 1.35.3, I get in the browser console:
```
index-a782d339.js:1058 Could not find specified theme
(anonymous) @ index-a782d339.js:1058
```
And the rendering of the web page is broken.
I tried to flush my browsers cache with no effect.
Loading it in a private tab works.
Maybe to make the js file more robust...?

### What did you expect to happen?

Page loading without error = including theme

### How to reproduce it (minimal and precise)

.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20221226

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Docker

sundeveu, 2024-02-17:
I confirm -  on my HA it did the same. You give me a clue what to do: it will be enough to clear all stored in browser cookies and configuration for HA page. I use Firefox and did it on dev tools / data tab clearing all sections. It logged me out but after log-in module works as expected without theme problems. Probably some stored configuration key caused problem.

pganansia, 2024-02-17:
I confirm the same issue.
I am using Zigbee2mqtt under Docker.

pganansia, 2024-02-17:
After clicking on the theme button (Nigth or Day), it works again.

mmattel, 2024-02-18:
> After clicking on the theme button (Nigth or Day), it works again.

Yes, this made it working again.
I propose that an error catch gets implemented if the theme is not found to automate this.

mathmaniac43, 2024-02-18:
Would it be possible as part of app startup to determine if the current theme script file exists, and if not to change it automatically?

jbl200, 2024-02-18:
Thank you! toggling the theme made it work again!
But the Images for #tuya-ts0012 devices remained as just a vertical line. I can live with that but just a note.

drieks, 2024-02-20:
I had the same problem and was able to fix it with the theme button, thanks!

closed 2024-09-18 by github-actions[bot]