Koenkk/zigbee2mqtt #21432: Aqara Vibration sensor DJT11LM reporting vibration but no angles and accelerometer values

Masadido, 2024-02-16:
### What happened?

My Aqara DJT11LM is correctly detecting vibrations but angles and accelerometer info are not updated anymore (always same values reported, whatever the tilt, position etc.).

Could this be related to the new version of Z2M and the version of the sensor (20180130)?

I did reset/reconnect the sensor several times, changed the battery etc.. no luck.

{
    "angle": 114,
    "angle_x": -1,
    "angle_x_absolute": 91,
    "angle_y": 74,
    "angle_y_absolute": 16,
    "angle_z": 16,
    "battery": 100,
    "device_temperature": 29,
    "linkquality": 131,
    "power_outage_count": 113,
    "sensitivity": "high",
    "strength": 9,
    "vibration": false,
    "voltage": 3145,
    "x_axis": -17,
    "y_axis": 1038,
    "z_axis": 296,
    "action": null,
    "device": {
        "applicationVersion": 8,
        "dateCode": "20180130\u0000",
        "friendlyName": "Aqara Vibration Front Door",
        "hardwareVersion": 3,
        "ieeeAddr": "0x00158d0006f24f33",
        "manufacturerID": 4151,
        "manufacturerName": "LUMI",
        "model": "DJT11LM",
        "networkAddress": 23302,
        "powerSource": "Battery",
        "softwareBuildID": "3000-0001",
        "stackVersion": 2,
        "type": "EndDevice",
        "zclVersion": 1
    }
}

### Zigbee2MQTT version

 1.35.3 

### Adapter firmware version

20180130

### Adapter

Aqara DJT11LM

### Setup

Z2M on Docker

Masadido, 2024-02-16:
Extract from log file

debug 2024-02-16 22:07:04: Received Zigbee message from 'Aqara Vibration Front Door', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3145,"10":62555,"152":40,"153":2,"154":[279,67764211],"3":30,"4":5032,"5":114,"6":[0,196608],"8":776},"modelId":"lumi.vibration.aq1"}' from endpoint 1 with groupID 0
debug 2024-02-16 22:07:04: DJT11LM: unknown key 6 with value 0,196608
debug 2024-02-16 22:07:04: DJT11LM: unknown key 153 with value 2
debug 2024-02-16 22:07:04: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}
debug 2024-02-16 22:07:04: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}
info  2024-02-16 22:07:04: MQTT publish: topic 'zigbee2mqtt/Aqara Vibration Front Door', payload '{"action":null,"angle":114,"angle_x":-1,"angle_x_absolute":91,"angle_y":74,"angle_y_absolute":16,"angle_z":16,"battery":100,"device":{"applicationVersion":8,"dateCode":"20180130\u0000","friendlyName":"Aqara Vibration Front Door","hardwareVersion":3,"ieeeAddr":"0x00158d0006f24f33","manufacturerID":4151,"manufacturerName":"LUMI","model":"DJT11LM","networkAddress":23302,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":30,"linkquality":131,"power_outage_count":113,"sensitivity":"medium","strength":9,"vibration":false,"voltage":3145,"x_axis":-17,"y_axis":1038,"z_axis":296}'
debug 2024-02-16 22:07:07: Received Zigbee message from 'Aqara Vibration Front Door', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3145,"10":62555,"152":40,"153":2,"154":[281,67698677],"3":30,"4":5032,"5":114,"6":[0,196608],"8":776},"modelId":"lumi.vibration.aq1"}' from endpoint 1 with groupID 0
debug 2024-02-16 22:07:07: DJT11LM: unknown key 6 with value 0,196608
debug 2024-02-16 22:07:07: DJT11LM: unknown key 153 with value 2
debug 2024-02-16 22:07:07: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}
debug 2024-02-16 22:07:07: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}
info  2024-02-16 22:07:07: MQTT publish: topic 'zigbee2mqtt/Aqara Vibration Front Door', payload '{"action":null,"angle":114,"angle_x":-1,"angle_x_absolute":91,"angle_y":74,"angle_y_absolute":16,"angle_z":16,"battery":100,"device":{"applicationVersion":8,"dateCode":"20180130\u0000","friendlyName":"Aqara Vibration Front Door","hardwareVersion":3,"ieeeAddr":"0x00158d0006f24f33","manufacturerID":4151,"manufacturerName":"LUMI","model":"DJT11LM","networkAddress":23302,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":30,"linkquality":131,"power_outage_count":113,"sensitivity":"medium","strength":9,"vibration":false,"voltage":3145,"x_axis":-17,"y_axis":1038,"z_axis":296}'
debug 2024-02-16 22:07:09: Received Zigbee message from 'Aqara Vibration Front Door', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3145,"10":62555,"152":40,"153":2,"154":[282,67764211],"3":30,"4":5032,"5":114,"6":[0,196608],"8":776},"modelId":"lumi.vibration.aq1"}' from endpoint 1 with groupID 0
debug 2024-02-16 22:07:09: DJT11LM: unknown key 6 with value 0,196608
debug 2024-02-16 22:07:09: DJT11LM: unknown key 153 with value 2
debug 2024-02-16 22:07:09: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}
debug 2024-02-16 22:07:09: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}
info  2024-02-16 22:07:09: MQTT publish: topic 'zigbee2mqtt/Aqara Vibration Front Door', payload '{"action":null,"angle":114,"angle_x":-1,"angle_x_absolute":91,"angle_y":74,"angle_y_absolute":16,"angle_z":16,"battery":100,"device":{"applicationVersion":8,"dateCode":"20180130\u0000","friendlyName":"Aqara Vibration Front Door","hardwareVersion":3,"ieeeAddr":"0x00158d0006f24f33","manufacturerID":4151,"manufacturerName":"LUMI","model":"DJT11LM","networkAddress":23302,"powerSource":"Battery","softwareBuildID":"3000-0001","stackVersion":2,"type":"EndDevice","zclVersion":1},"device_temperature":30,"linkquality":127,"power_outage_count":113,"sensitivity":"medium","strength":9,"vibration":false,"voltage":3145,"x_axis":-17,"y_axis":1038,"z_axis":296}'
debug 202debug 2024-02-16 22:07:04: Received Zigbee message from 'Aqara Vibration Front Door', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3145,"10":62555,"152":40,"153":2,"154":[279,67764211],"3":30,"4":5032,"5":114,"6":[0,196608],"8":776},"modelId":"lumi.vibration.aq1"}' from endpoint 1 with groupID 0
debug 2024-02-16 22:07:04: DJT11LM: unknown key 6 with value 0,196608
debug 2024-02-16 22:07:04: DJT11LM: unknown key 153 with value 2
debug 2024-02-16 22:07:04: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}
debug 2024-02-16 22:07:04: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}
info  2024-02-16 22:07:04: MQTT publish: topic 'zigbee2mqtt/Aqara Vibration Front Door', payload '{"action":null,"angle":114,"angle_x":-1,"angle_x_absolute":91,"angle_y":74,"angle_y_absolute":16,"angle_z":16,"battery":100,"device":{"applicationVersion":8," [... cut ...] 608],"8":776},"modelId":"lumi.vibration.aq1"}' from endpoint 1 with groupID 0
debug 2024-02-16 22:07:11: DJT11LM: unknown key 6 with value 0,196608
debug 2024-02-16 22:07:11: DJT11LM: unknown key 153 with value 2
debug 2024-02-16 22:07:11: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}
debug 2024-02-16 22:07:11: DJT11LM: Processed data into payload {"voltage":3145,"battery":100,"device_temperature":30,"power_outage_count":113}

mrmaximas, 2024-02-16:
I confirm this problem

Masadido, 2024-02-17:
> I confirm this problem

Always good not to be alone :-)

DarwinData, 2024-02-18:
Same problem for me.
Thanks

acaonweb, 2024-02-25:
same problem here, affecting Xiaomi MFKZQ01LM too

lux73, 2024-02-26:
hm, mine DJT11LM works as expected:

[image]

roamingwombat, 2024-03-18:
Mine (DJT11LM with fw 20180130) is experiencing very similar behavior. The one exception is that it seems to work intermittently. My sensor is new, and when I first set it up yesterday, everything seemed to work as expected.

I was going to calibrate the accelerometer today, and noticed that the angle wasn't updating on any axis. Checking the entity trend in Home Assistant showed that it had been "stuck" earlier for about 3 hours, then had shown some movement briefly, but was stuck again. 

I moved the sensor close to the coordinator, thinking maybe a weak signal was to blame. No change. Then I took the battery out and put it back in.

After putting the same battery back in, it started working again... For about 5 minutes. Then it froze again. So, I replaced the battery with a brand new battery... Still frozen.

During this entire time, the vibration worked as expected.

TheEmperor1k, 2024-08-18:
I can confirm this issue. It worked fine for the first day or two, but since is stuck on the angle values.

Nevynnyi, 2024-10-02:
Same problem with two new device, all angle always unknown in MQTT and null in angle. Battery, reconnect not work.
Vibration, Strength, battery, trmp and action works ok.

ChirpyTurnip, 2024-10-02:
Same is happening to me now....two units, both failing to report angles - on one unit I get nothing but this:

```
{
    "last_seen": "2024-10-03T10:41:01+13:00",
    "linkquality": 69
}
```

on the other:

```
{
    "last_seen": "2024-10-03T10:46:38+13:00",
    "linkquality": 54,
    "action": null,
    "angle_x": null,
    "angle_x_absolute": null,
    "angle_y": null,
    "angle_y_absolute": null,
    "angle_z": null,
    "battery": null,
    "device_temperature": null,
    "power_outage_count": null,
    "sensitivity": null,
    "strength": null,
    "vibration": false,
    "voltage": null,
    "x_axis": 0,
    "y_axis": 0,
    "z_axis": 0,
    "angle": 110
}
```

Started happening after the last update. Prior to that very reliable - pretty sure it isn't the device itself....

danise76, 2024-10-02:
I can confirm exactly the same like ChirpyTurnip says. Started to happen with the last Z2M update to v1.40.2-1 yesteday.
It was OK until that version

danise76, 2024-10-02:
> I can confirm exactly the same like ChirpyTurnip says. Started to happen with the last Z2M update to v1.40.2-1 yesteday. It was OK until that version

Just to reconfirm it. I restores Z2M to v1.40.1 from a backup, and then it's working again.  Something broken on 1.40.2

{
    "angle": 14,
    "angle_x": 2,
    "angle_x_absolute": 88,
    "angle_y": -76,
    "angle_y_absolute": 166,
    "angle_z": 14,
    "battery": 0,
    "device_temperature": 29,
    "last_seen": "2024-10-03T01:13:25+02:00",
    "linkquality": 25,
    "power_outage_count": 72,
    "sensitivity": "medium",
    "strength": 65,
    "vibration": false,
    "voltage": 2815,
    "x_axis": 41,
    "y_axis": -1014,
    "z_axis": 259,
    "action": null
}

ChirpyTurnip, 2024-10-02:
Sadly I rolled back to the old version and the problem persisted.....now back on the latest version since it makes no difference. My brother's HA is on he latest version too and his sensor (he has only one) *is* still working correctly....so this might be one of those weird "only some users" things....

I've also noticed that the sensors might stop responding (no update at all), and also that the the reset button on the sensor stops working - you need to pull the battery to bring it back to life...and then it goes back to timing out....it is a very weird problem!

thenoid, 2024-10-02:
RIP having this same issue.

ChirpyTurnip, 2024-10-03:
I followed these [instructions ](https://community.home-assistant.io/t/how-to-downgrade-addons/331223/17) to hack my restore version from 1.40.1-1 back to 1.40.1 since it was already broken in 1.40.1-1 and I had no earlier backup.

In case you don't know: I used the FileBrowser add on to get download/upload access to backups folder, pulled down my Z2M backup tar file (identify this by date created versus your backup history), used 7zip to open the archive and pull out the backup.json file, then used Notepad++ to edit this to set the version number I wanted to restore,. Then you save this file, drag and drop it back into 7zip to update the archive, then upload the modified tar file back to HA, and then restored it. Phew. Too much effort!

It is working again....but obviously I'm not likely to be able to move past this version now.

Nevynnyi, 2024-10-03:
Return to the Docker image of zigbee2mqtt version 1.40.1 - still the same problem, not work :(

cubscholze, 2024-10-03:
Same issue with my Aqara DJT11LM after updating to 1.42.2. I rolled back to 1.40.1 and it's working again.

Nevynnyi, 2024-10-03:
> Same issue with my Aqara DJT11LM after updating to 1.40.2. I rolled back to 1.40.1 and it's working again.

did you reconnect your device after downgrade?

ChristianScholze, 2024-10-03:
Yes, I needed to pair it to get it working.

Nevynnyi, 2024-10-03:
> > > Same issue with my Aqara DJT11LM after updating to 1.40.2. I rolled back to 1.40.1 and it's working again.
> > 
> > 
> > did you reconnect your device after downgrade?
> 
> Yes, I needed to reconnect it to get it working again.

thx, i'll try

k8gg, 2024-10-03:
Can confirm. Something happened between `1.40.1` and `1.40.2`.
After downgrading to 1.40.1, and then reconnect / re-pair the vibration sensor, and the angle_x _y _z would come back.

This is under 1.40.2:
```
{
    "angle": 81,
    "angle_x": null,
    "angle_x_absolute": null,
    "angle_y": null,
    "angle_y_absolute": null,
    "angle_z": null,
    "battery": 100,
    "device_temperature": 30,
    "last_seen": "2024-10-03T10:21:00.678Z",
    "linkquality": 134,
    "power_outage_count": 316,
    "sensitivity": "medium",
    "strength": 2,
    "vibration": false,
    "voltage": 3055,
    "x_axis": 0,
    "y_axis": 0,
    "z_axis": 0,
    "action": null
}
```

Nevynnyi, 2024-10-04:
Maybe need create new issues or bug report with this problem on 1.40.2?

muzo178, 2024-10-04:
> Can confirm. Something happened between `1.40.1` and `1.40.2`. After downgrading to 1.40.1, and then reconnect / re-pair the vibration sensor, and the angle_x _y _z would come back.
>
> This is under [...]

exactly the same behaviour. something happened between `1.40.1` and `1.40.2`.

mrmaximas, 2024-10-04:
Same behavior with latest 1.40.2-dev commit: [b09ae105](https://github.com/Koenkk/zigbee2mqtt/commit/b09ae105)
@Koenkk please pay attention. Thanks!

Claude2666, 2024-10-06:
Same here, 
I upgraded to the latest dev build 1.40.2. ... and my tilt sensor stopped working properly.
I did rollback to the previous release I used : 1.40.1-dev commit: [0540e4fb]  and it started working again.

warren-ja, 2024-10-06:
Add me to the list too.  Exactly the same issue.

Koenkk, 2024-10-06:
Will be fixed inhttps://github.com/Koenkk/zigbee2mqtt/discussions/24185

Nevynnyi, 2024-10-07:
> Will be fixed inhttps://github.com/[/discussions/24185](https://github.com/Koenkk/zigbee2mqtt/discussions/24185)

Thx! ☺️

vkolovos, 2024-10-20:
Dear Koenkk & Nevynnyi. Cloud you please give us an indication on when the new version which fixes the Aqara vibration device issue will be available in production?

jeremysherriff, 2024-10-20:
> Dear Koenkk & Nevynnyi. Cloud you please give us an indication on when the new version which fixes the Aqara vibration device issue will be available in production?

https://github.com/Koenkk/zigbee2mqtt/discussions/24185#discussioncomment-10918895

vkolovos, 2024-10-20:
Thank you. Sorry but I missed that. It's 1st of November.

dwaned, 2024-10-29:
Could it be that the issue also affects other devices such as ThirdReality 3RVS01031Z ? I just installed one and it also does not report any accelerometer values. Will it also be fixed with the release on 1st November?

dwaned, 2024-11-01:
After updating to latest version, the Aqara Vibration sensor now reports all values. The ThirdReality sensor still does not, so probably its a separate issue.

ThierryViens, 2024-11-01:
@dwaned  how did you get the update ? it's not available yet for me.....

dwaned, 2024-11-01:
> @dwaned how did you get the update ? it's not available yet for me.....

Update just showed up for me. Try triggering the check for updates in settings.

warren-ja, 2024-11-01:
I did the update and angles are showing now.  Awesome!

But now the Z2M Web UI will not load, even after multiple restarts.  Anyone else having this issue?

phbasler, 2024-11-01:
I have the update installed but the angles still show as unkown.
Is it necessary to reconnect the sensor?

jeremysherriff, 2024-11-01:
> I have the update installed but the angles still show as unkown. Is it necessary to reconnect the sensor?

Should not be necessary but you do need to force the sending of all data, try clicking the button and then shake it vigorously

vkolovos, 2024-11-01:
Worked as expected after updating to the latest version. Thank you all.
Please note that you have to trigger the sensor to send new stream data.

phbasler, 2024-11-02:
> > I have the update installed but the angles still show as unkown. Is it necessary to reconnect the sensor?
> 
> Should not be necessary but you do need to force the sending of all data, try clicking the button and then shake it vigorously

Tried that, didn't work. But after reconnecting the sensor it works as expected.

jeremysherriff, 2024-12-16:
Keen to hear from others on whether this issue is fully resolved.

My vibration sensor seems to be very unreliable reporting the x/y/z angles and axis values to HA now, but when I set z2m logging to info and filter on the device, the values appear to be recorded in a timely manner and be accurate.  Have changed to a fresh battery but the old one was still above 3000mV so it should have been fine, and the fresh one has only helped for a few days (which was one of the original symptoms reported, despite the code fix being to do with the data payload parsing).

What are other people's experiences?
Also posting this Q in #24185

ThierryViens, 2024-12-16:
Hi @jeremysherriff , on my end it's been very stable since the update was made.. no issue at all, using it with Jeedom via Z2m

closed 2025-02-22 by github-actions[bot]