Koenkk/zigbee2mqtt #21435: Candeo Switch Module C205

chrisandsally, 2024-02-17:
### Link

https://candeo.io/store/zigbee-switch-module-with-or-without-neutral/

### Database entry

(C205)","name":"0xec1bbdfffead2163","sw_version":"2.9.2_r54","via_device":"zigbee2mqtt_bridge_0xe0798dfffe9e862c"},"name":null,"object_id":"0xec1bbdfffead2163","origin":{"name":"Zigbee2MQTT","sw":"1.35.3","url":"https://www.zigbee2mqtt.io"},"payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0xec1bbdfffead2163","unique_id":"0xec1bbdfffead2163_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'

### Comments

I tried to follow instructions but stuck at step 3 - Clone.

The autogenerate process worked!.  Switch capability works. Just want to help added so that it is supported.

### External definition

```shell
const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['C205'],
    model: 'C205',
    vendor: 'Candeo',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false})],
    meta: {},
};

module.exports = definition;
```

chrisandsally, 2024-02-17:
Struggle to add pull request. Now working in Home Assistant.

C205.js

const {onOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Candeo'],
    model: 'C205', // Update this with the real model of the device (written on the device itself or product page)
    vendor: 'Candeo', // Update this with the real vendor of the device (written on the device itself or product page)
    description: 'Candeo Switch Module', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: [onOff({"powerOnBehavior":false})],
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

[image]

Koenkk, 2024-02-18:
Added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-18 by Koenkk