Koenkk/zigbee2mqtt #21439: Schneider Electric     2GANG/DIMMER/2

Highflyer100, 2024-02-17:
### Link

https://merten.de/produkt/meg5126-0300-connected-taster-modul-2fach-system-m.html

### Database entry

*

### Comments

Hi, followed the procedure for adding unsupported devices, but still I get the following error messages in the log:
- No converter available for 'state_3' ("ON")
- No converter available for 'state_4' ("ON")
Any help?

### External definition

```shell
const {deviceEndpoints, onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['2GANG/DIMMER/2'],
    model: '2GANG/DIMMER/2',
    vendor: 'Schneider Electric',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"3":3,"4":4,"21":21,"22":22,"242":242}}), onOff({"powerOnBehavior":false,"endpoints":{"3":3,"4":4}}), identify()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

Highflyer100, 2024-02-18:
const {deviceEndpoints, onOff, identify} = require('zigbee-herdsman-converters/lib/modernExtend');
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['2GANG/DIMMER/2'],
    model: '2GANG/DIMMER/2',
    vendor: 'Schneider Electric',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"3":3,"4":4,"21":21,"22":22,"242":242}}), onOff({"powerOnBehavior":false,"endpoints":{"3":3,"4":4}}), identify()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;

Errors in log:
Zigbee2MQTT:error 2024-02-18 10:30:13: Failed to load external converter file 'dimmer.js' ('2GANG/DIMMER/2' has multiple 'endpoint', this is not allowed)
Zigbee2MQTT:error 2024-02-18 10:30:13: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version

Highflyer100, 2024-02-19:
I made some progress:

With this converter from Nailik, i managed to be able to switch and dimm one of the two lights.
Also found out that endpoint 4 and 21 are for light 1.   3 and 22 for light 2.

This is my converter actually:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['2GANG/DIMMER/2'],
    model: 'MEG5126-0300/MEG5172-0000',
    vendor: 'Schneider Electric',
    description: 'Merten (MEG5172-0000 PlusLink) Dimmer (1fold) insert with Merten (MEG5126-0300) Wiser System M Push Button (2fold)',
    fromZigbee: [
        fz.on_off, 
        fz.brightness, 
        fz.level_config, 
        fz.wiser_lighting_ballast_configuration
        ],
    toZigbee: [
        tz.light_onoff_brightness, 
        tz.level_config, 
        tz.ballast_config, 
        tz.wiser_dimmer_mode
        ],
    exposes: [
        e.light_brightness().withLevelConfig().withEndpoint('l1'),
        e.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        e.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        e.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)')
            ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(3);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
        await reporting.onOff(endpoint1);
        await reporting.brightness(endpoint1);
        const endpoint2 = device.getEndpoint(22);
    },
    endpoint: (device) => {
        return {l1: 3, l2: 22};
    }
};

module.exports = definition;

How can I insert into this converter teh same content for endpoints 4 and 21. I want to duplicate whats already in tegre as my switch is a 2 fold switch with 2 dimmers attached to it.

Rgds,

Highflyer100, 2024-02-19:
Got it working.  2Gang/Dimmer2   MEG5172-0300 with MEG5126-0300.

Both lights can be switched and dimmed.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['2GANG/DIMMER/2'],
    model: 'MEG5126-0300/MEG5172-0000',
    vendor: 'Schneider Electric',
    description: 'Merten (MEG5172-0000 PlusLink) Dimmer (1fold) insert with Merten (MEG5126-0300) Wiser System M Push Button (2fold)',
    fromZigbee: [
        fz.on_off, 
        fz.brightness, 
        fz.level_config, 
        fz.wiser_lighting_ballast_configuration
        ],
    toZigbee: [
        tz.light_onoff_brightness, 
        tz.level_config, 
        tz.ballast_config, 
        tz.wiser_dimmer_mode
        ],
    exposes: [
        e.light_brightness().withLevelConfig().withEndpoint('l1'),
        e.light_brightness().withLevelConfig().withEndpoint('l2'),
        e.numeric('ballast_minimum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the minimum light output of the ballast'),
        e.numeric('ballast_maximum_level', ea.ALL).withValueMin(1).withValueMax(254)
            .withDescription('Specifies the maximum light output of the ballast'),
        e.enum('dimmer_mode', ea.ALL, ['auto', 'rc', 'rl', 'rl_led'])
            .withDescription('Sets dimming mode to autodetect or fixed RC/RL/RL_LED mode (max load is reduced in RL_LED)')
            ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(4);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
        await reporting.onOff(endpoint1);
        await reporting.brightness(endpoint1);      
        const endpoint2 = device.getEndpoint(3);
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl', 'lightingBallastCfg']);
        await reporting.onOff(endpoint1);
        await reporting.brightness(endpoint1);      
    },
    endpoint: (device) => {
        return {l1: 4, l2: 3};
    }
};

module.exports = definition;

Highflyer100, 2024-02-19:
Help for a pull request appreciated! :-)

Koenkk, 2024-02-20:
>  No feedback to HA, when switched on via wall mounted physical switches.

Does the configure succeed? You can force it via the z2m frontend -> device -> yellow refresh icon.

Highflyer100, 2024-02-21:
Forcing it does not work either. But the same problems exits with the HUE bulbs with the HA integration. Even with a wrong state in HA, the automation still works, that's the most important thing.

Apart this, the converter for the 2Gang/Dimmer/2   MEG 5126-0300  with MEG5172-0000 insert works very well. ON/OFF and Dimming, Start Level Brightness ---- all good.

I tried to pull a request, but it didn't work out. I need to look for a tutorial.  But my converter could be a good base for all 2 switch Schneider Electric  devices.

Nailik, 2024-04-12:
@Highflyer100 by now the 2Gang/Dimmer/2 MEG 5126-0300 with MEG5172-0000 is supported by z2mqtt maybe you could remove your custom converted and try it out. 

however this "problem": And when switched ON, it always goes to full brightness. is still the same (when using harware button). my idea is to change Ballast maximum level at a specific time in order for it to not go full brightness - but Ballast maximum level seems to have no effect.

Highflyer100, 2024-04-14:
@Nailik I deleted my custom converter and tried yours. 

Seems to work fine for me. 

When switched ON, it goes to full brightness. It doesn't matter if I am using HA or Hardware. I would love to have the original feature, like it was in Wiser Room App in bluetooth mode, where you were able to determine that the startup level should correspond to the level when switsched OFF.

What I did however, and it works fine for me as I use the corresponding lights mostly during evening time, I wrote under 
- Dev console  - Endpoint 3 - Cluster:LevelCtrl - Attribute OnLevel to 60     which corresponds to 25%.

Same for endpoint 4

This worked for HA and Hardware to have the lights started with only 25% brightness. (Attribute Number between 0-255). So I just need very minor adjustements to set the right brightness.

Nailik, 2024-04-14:
@Highflyer100 
with https://github.com/Koenkk/zigbee-herdsman-converters/pull/7395#issuecomment-2053560669 the levelCtrl is in current dev build of z2mqtt. 
However it currently can't be changed by UI because the save button is missing (at least for me) see https://github.com/Koenkk/zigbee2mqtt/issues/22195#issuecomment-2052416708

I (will) use NodeRed to update it, you just need to send `{"write":{"cluster":"genLevelCtrl","options":{},"payload":{"onLevel":254}}}` onto `zigbee2mqtt/light_id/endpoint_id/set` where `light_id` is the name of your light and `endpoint_id` either 3 or 4 (left and right)

Highflyer100, 2024-04-14:
I can actually change it in the DEV Console,   endpoint 3  cluster: levelctrl  atrribute: Onlevel  . Put a number in between 1-255 and click on write.

Highflyer100, 2024-04-14:
[image]

closed 2024-11-11 by github-actions[bot]

Nailik, 2024-12-30:
@Highflyer100 have you been able to get ballast level to work? For me it seems it doesn't change anything for the 2-Gang-Dimmer but works for the 1-Gang-Dimmer. 
(Currently one light blinks when brightness is too high and therefore i wanted to reduce it but max_level is not suppored)

Highflyer100, 2024-12-30:
@Nailik  Ballast Level is not working for me in the EXPOSES. But as L and R side perfectly working, incl. brightness at startup and dimmer function, I stopped working on it. When brightness at max, no flickering. 
You can still change Ballast in DEV Console by finding the right Attribute (For me Endpoint 3 or 4 (depending on L or R); cluster lightingBallastCfg; then read or write new attribute)

Nailik, 2024-12-30:
@Highflyer100 could you check if `lightingBallastCfg` actually changes the max brightnes of your lights? for me there is absolutely no difference even when using the dev console

Highflyer100, 2025-01-04:
@Nailik Well, actually when I increase the max ballast level, the lights get brighter. But since I am using LED bulbs, I've read that LED bulbs work out there ballast differently. I am no electrician, but I assume that these setting are only interesting for regular bulbs.