Koenkk/zigbee2mqtt #21440: [New device support]: TS0601 thermostat zigbee manufacturer _TZE200_znlqjmih

paoloman, 2024-02-17:
### Link

none

### Database entry

{"id":11,"type":"EndDevice","ieeeAddr":"0x0ceff6fffe739031","nwkAddr":61844,"manufId":4098,"manufName":"_TZE200_znlqjmih","powerSource":"Battery","modelId":"TS0601 ","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"mod elId":"TS0601","manufacturerName":"_TZE200_znlqjmih","powerSource":3,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"con figuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":170818 3712797,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the instruction on the support page...

zigbee2mqtt fail to start

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_znlqjmih',
        },
    ],
    model: 'TS0601_thermostat',
    vendor: 'TuYa',
    description: 'Radiator valve with thermostat',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [ // here you should put all functionality that your device exposes
            e.child_lock(), e.window_detection(),
            e.binary('window_open', ea.STATE, true, false).withDescription('Window open?'),
            e.battery_low(), e.valve_detection(), e.position(),
            e.climate().withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET,
                    'Mode of this device, in the `heat` mode the TS0601 will remain continuously heating, i.e. it does not regulate ' +
                    'to the desired temperature. If you want TRV to properly regulate the temperature you need to use mode `auto` ' +
                    'instead setting the desired temperature.')
                .withLocalTemperatureCalibration(-9, 9, 0.5, ea.STATE_SET)
                .withPreset(['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco', 'away'])
                .withRunningState(['idle', 'heat'], ea.STATE),
            e.auto_lock(), e.away_mode(), e.away_preset_days(), e.boost_time(), e.comfort_temperature(), e.eco_temperature(), e.force(),
            e.max_temperature().withValueMin(16).withValueMax(70), e.min_temperature(), e.away_preset_temperature(),
            e.week(),
            e.text('workdays_schedule', ea.STATE_SET)
                .withDescription('Workdays schedule, 6 entries max, example: "00:20/5°C 01:20/5°C 6:59/15°C 18:00/5°C 20:00/5°C 23:30/5°C"'),
            e.text('holidays_schedule', ea.STATE_SET)
                .withDescription('Holidays schedule, 6 entries max, example: "00:20/5°C 01:20/5°C 6:59/15°C 18:00/5°C 20:00/5°C 23:30/5°C"'),
        ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },
};

module.exports = definition;
```

CH-Tefi, 2024-02-18:
i have the same thermostat _TZE200_znlqjmih

paoloman, 2024-02-18:
Does It work?
From my side, it result as unsupported and I can ready only linkquality.

agak79, 2024-02-18:
Same problem here. Funny think is I bought 2 TRV valves from the same vendor, one is detected as _TZE200_cpmgn2cf and fully supported and the other one as _TZE200_znlqjmih and not supported but externally are the same, even the manual and the box are the same.

paoloman, 2024-02-18:
Same situation on my side

Koenkk, 2024-02-18:
Added the `_TZE200_znlqjmih`!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-18 by Koenkk

paoloman, 2024-02-18:
thank you Koenkk for your hard work.
Is the dev branch already available?

[image]

thanks

paoloman, 2024-02-19:
does anybody tried dev branch?
from my side, still get same error message

> pi@raspberrypi:/opt/zigbee2mqtt $ git checkout dev
> error: pathspec 'dev' did not match any file(s) known to git

thank you

agak79, 2024-02-19:
I tried and it's working

paoloman, 2024-02-19:
I get that error on raspberry and the devices still result as unsupported in the zigbee2mqtt front-end

paoloman, 2024-02-19:
@agak79, did you follow the procedure at [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) ?

agak79, 2024-02-19:
> @agak79, did you follow the procedure at [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) ?

The procedure on home assistant is pretty simple, basically installing a different add-on, the edge version, so probably different procedure than I'm your case.

paoloman, 2024-02-19:
@Koenkk could you please help on switching procedure to dev branch?
thank you

agak79, 2024-02-19:
> @Koenkk could you please help on switching procedure to dev branch? thank you

Hi

I'm far from an expert, but I didn't use any command as in your example.

I just downloaded the edge version of the add on, disabled the standard version before configure the edge version, start the edge version and everything was working. I think both version use the same configuration.yaml so it's everything installed but you can have both running

paoloman, 2024-02-19:
I'm using zigbee2mqtt with Node-Red, so I need to manually switch to dev branch using the procedure linked from Koenkk... Unfortunally, something goes wrong but I cannot undesrtand how to fix

paoloman, 2024-02-23:
@Koenkk , could you please help me? the procedure to switch to the development branch seems not work...

[image]

I apply the procedure but the devices are still "unsupported"
I suppose the problem is in the lines I highligthed

thank you

Koenkk, 2024-02-24:
Note sure why that happens, anyway, they will be supported in the 1 March release

paoloman, 2024-02-24:
Thank you for your feedback and your hard work on zigbee2mqtt!

paolo