Koenkk/zigbee2mqtt #21442: Problem with color on Eglo 99099

msaphoto06, 2024-02-17:
### What happened?

Good morning,

Is it possible to correct the remote control support?
This does not bring up the colors.
I use a converter which works but it does not give me the action_groups associated with color and refresh.
THANKS

const awoxRemoteHelper = {
    convertToColorName: (buffer) => {
        const commonForColors = buffer[0] === 17 && buffer[2] === 48 && buffer[3] === 0 && buffer[5] === 8 && buffer[6] === 0;

        if (commonForColors && buffer[4] === 254) {
            return 'red';
        } else if (commonForColors && buffer[4] === 85) {
            return 'green';
        } else if (commonForColors && buffer[4] === 169) {
            return 'blue';
        }
        return null;
    },
    isRefresh: (buffer) => {
        return buffer[0] === 17 && buffer[2] === 16 && buffer[3] === 1 && buffer[4] === 1;
    },
};

const fzLocal = {
    colors: {
        cluster: 'lightingColorCtrl',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            const color = awoxRemoteHelper.convertToColorName(msg.data);
            if (color != null) {
                return {
                    action: color,
                };
            }
        },
    },

### Zigbee2MQTT version

Last

### Adapter firmware version

Last

### Adapter

Sonoff

### Setup

Dockers

Koenkk, 2024-02-18:
You don't need to use a converter, this should work out-of-the-box with 1.35.3

msaphoto06, 2024-02-18:
Good morning,

I am in 35.3
I deleted the converter in z2m
I restarted, deleted the remote and added it.
I can see the color raised, but still no GroupId unlike the other functions.

THANKS

Action_Group OK
`Debug 2024-02-18 18:16:18Received Zigbee message from '0xa4c138946a4d2d2b', type 'commandStepWithOnOff', cluster 'genLevelCtrl', data '{"stepmode":0,"stepsize":18,"transtime":2}' from endpoint 1 with groupID 32778

Info 2024-02-18 18:16:18MQTT publish: topic 'zigbee2mqtt/0xa4c138946a4d2d2b', payload '{"action":"brightness_step_up","action_group":32778,"action_step_size":18,"action_transition_time":0.02,"linkquality":80}'`

Action_Group No OK
`Debug 2024-02-18 18:17:11Received Zigbee message from '0xa4c138946a4d2d2b', type 'raw', cluster 'lightingColorCtrl', data '{"data":[17,240,48,0,254,8,0],"type":"Buffer"}' from endpoint 1 with groupID 32778

Info 2024-02-18 18:17:11MQTT publish: topic 'zigbee2mqtt/0xa4c138946a4d2d2b', payload '{"action":"red","linkquality":87}'`

Koenkk, 2024-02-20:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-20 by Koenkk

msaphoto06, 2024-02-22:
Hello,

It's not OK. 
GroupId OK but No color info
[image]

Koenkk, 2024-02-22:
We didn't map this color, what color should it be?

msaphoto06, 2024-02-22:
Hello
https://github.com/Koenkk/zigbee2mqtt/issues/14786

Red is 254
Blue is 69
Green is 84

Koenkk, 2024-02-24:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

msaphoto06, 2024-02-24:
I am Sorry
Blue is 169 not 69.
Typo
:(

msaphoto06, 2024-02-24:
I tested the dev. It's ok for 2 colors out of 3 with action group. (remains 169 ;))
I see in the debug and the color goes back to home assistant.
however. My associated bulb group does not change color (without HA automation)
Is there a converter to create for this? Like
if color red -> x:... y:...
If yes, how?
THANKS
Also don't know how to bring up the "actionGroup" value in HA

msaphoto06, 2024-02-25:
Thanks for the fix.
The first question is not about HA but Z2M.
The color is fine in debug with its groupId but does not change the color of the group (z2m) unlike brighness or on/off....

Koenkk, 2024-02-25:
That's because this is a custom manufacturer specific command, other bulbs don't understand it.

msaphoto06, 2024-03-02:
It S possible to expose attribut " action_group" in sensor to home assitant?
Only action and link are exposed

msaphoto06, 2024-03-10:
?

Koenkk, 2024-03-10:
To respond to `action_group` see https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#via-mqtt

msaphoto06, 2024-03-11:
thank you, but I don't know what to put in value template