Koenkk/zigbee2mqtt #21445: [New device support]: MTG075-ZB-RL - TuYa Zigbee Human Presence Sensor - alternate Manufacturer ID

jeffgoh, 2024-02-18:
### Link

https://www.aliexpress.com/item/1005005909374871.html

### Database entry

{"id":18,"type":"Router","ieeeAddr":"0x70ac08fffef62f11","nwkAddr":49012,"manufId":4098,"manufName":"_TZE200_mp902om5","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_mp902om5","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1708148800903}

### Comments

This is the 5.8G, ceiling/wall surface mount version. On the box it's marked as MTG075-ZB-RL, but the fingerprint is just slightly different. Needed to add '_TZE200_mp902om5' to the fingerprint for MTGHope this helps somebody else buying the same item:

```
diff -r1.1 node_modules/zigbee-herdsman-converters/device/tuya.js
7013c7013
<         fingerprint: tuya.fingerprint('TS0601', ['_TZE204_sbyx0lm6', '_TZE204_clrdrnya', '_TZE204_dtzziy1e', '_TZE204_iaeejhvf', '_TZE204_mtoaryre']),
---
>         fingerprint: tuya.fingerprint('TS0601', ['_TZE204_sbyx0lm6', '_TZE204_clrdrnya', '_TZE204_dtzziy1e', '_TZE204_iaeejhvf', '_TZE204_mtoaryre', '_TZE200_mp902om5' ]),

```

### External definition

```shell
it was just easier to add the fingerprint
```

Koenkk, 2024-02-18:
Added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-18 by Koenkk

EcarDK, 2024-03-13:
Hey @jeffgoh  and @Koenkk 
I just bought this sensor you link and i can't get it into my zigbee2mqtt. Is there anything I can do about it?

I don't understand what you mean by fingerprint :)

AlexMKX, 2024-04-03:
One more sensor fingerprint _TZE200_sbyx0lm6