Koenkk/zigbee2mqtt #21446: configure problem in DEV version

arteck, 2024-02-18:
### What happened?

Failed to configure 'button.4er.kino', attempt 1 (Error: Write 0x04cf8cdf3c79400c/1 manuSpecificLumi({"mode":1}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 46400 - 1 - 9 - 64704 - 4 after 10000ms) at Object.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:502:47) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.write (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:390:28) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:2195:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))

its this button 
https://www.zigbee2mqtt.io/devices/WXCJKG13LM.html#aqara-wxcjkg13lm
and this button also
https://www.zigbee2mqtt.io/devices/WXCJKG12LM.html#aqara-wxcjkg12lm

### Zigbee2MQTT version

1.35.3 DEV

### Adapter firmware version

20230922

### Adapter

sonoff -P

### Setup

zigbee2mqtt in LXC

sjorge, 2024-02-18:
You might need to wake up the device before z2m tries to reconfigure it. It's probably sleeping.

Koenkk, 2024-02-18:
@sjorge currently z2m will already reconfigure when the device sends a message, maybe we should just make the logging a bit more friendly?

sjorge, 2024-02-18:
> @sjorge currently z2m will already reconfigure when the device sends a message, maybe we should just make the logging a bit more friendly?

Perhaps, I think the main problem is that they all just have send _immediate_ set now (as they are all missing genPollCtrl I think), hence all the timeout, no mac or no route we see with all the Aqara/Xiaomi/Maji stuff as the rename to the proper Lumi manifacturer is triggering reconfigures on all of them.

From experience the TVOC sensor does not complete a configure/reconfigure (even if triggered manually when the device is awake) without the quirk, somewhere mid-configure it would hit a timeout. With the quirk it works.

Koenkk, 2024-02-18:
> From experience the TVOC sensor does not complete a configure/reconfigure (even if triggered manually when the device is awake) without the quirk, somewhere mid-configure it would hit a timeout. 

But then the configure will take very long, right? (and block other configures)

Maybe we can speed up the configure for this device? I'm a bit hesitant to changing it for all as previously this worked fine.

sjorge, 2024-02-18:
I’m pretty sure it is partially related to the whole sendWhen, that it now hits the timeouts sometimes. Not a super big deal as you can just manually keep it awake during paring. (But either way the quirk it should just work, and be a lot faster from my experience as as the next step is done pretty quick because the device replies to the bind/configure to confirm it so z2m just works it’s way through it and not depend on well times button pushes)This is a bigger problem now because to refactor to lumi is now probably triggering a mass reconfigure for all those devices, and they are most-likely sleeping, which means we hit the timeouts. Which in turn causes a lot of delays AND failures, resulting in z2m trying again.I think adding the quirk would actually improve this, since it should be first in the extend list.- z2m triggers a reconfigure at startup- quirk changes to behavior from immediate to only try when the device is active.- worst case we timeout and we try again later, best case the device responds and we finish.It’s going to be painful either way, due to timeouts or is waiting. But in the later scenario we can guarantee it will actually succeed and we don’t wast time trying multiple times. (Which will likely all hit timeouts)~ sjorge

Koenkk, 2024-02-18:
I see, then I propose to:

- Add the quirk
- Don't configure on startup for devices with this quirk (which is done [here](https://github.com/Koenkk/zigbee2mqtt/blob/fe0742a628fb782692f9679367e2ab1b11139dd6/lib/extension/configure.ts#L68))

sjorge, 2024-02-18:
Sounds good, we should key this of the database entry the quirk adds though.We don’t want to configure battery devices with genPollCtrl on startup either.I’ll look at this somewhere this week, hopefully.~ sjorge

sjorge, 2024-02-25:
Oops I just had time to look at this but looks like you already fixed it. I was gonna check for the `checkinInterval` property in the database was present and then skip. But checking if it's a router probably also works equally well.

As then even battery devices without genPollCtl would immediately be excluded. i'll look at adding the quirk to the lumi devices then. Been rather busy lately unfortunately.

rccoleman, 2024-03-01:
I'm still getting what I think are the same messages for all my battery-operated Aqara motion sensors, even with the latest dev (1.36.0-dev commit: [38ed953](https://github.com/Koenkk/zigbee2mqtt/commit/38ed953)):

```
Error 2024-03-01 14:00:28Failed to configure 'DNH Motion Sensor', attempt 2 (Error: Read 0x54ef4410005ca518/1 genPowerCfg(["batteryVoltage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 22147 - 1 - 34 - 1 - 1 after 10000ms) at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:499:47) at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/app/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:466:28) at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/lumi.ts:1351:13) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25) at Configure.configure (/app/lib/extension/configure.ts:122:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

I get the same message over and over for all my Aqara motion sensors, such that whenever I open the UI, my screen is flooded with them.

closed 2024-09-06 by arteck