Koenkk/zigbee2mqtt #21449: Samotech SM323 v1 - problem when Z2M tries configuring

james-fry, 2024-02-18:
### What happened?

I have ~10 Samotech SM323 v1 dimmers.
Z2M periodically tries to configure them but this always fails with an exception Error: Device 0xf4b3b1fffef93cf2 has no input cluster haElectricalMeasurement
This is correct - SM323 v1 dimmers do not have any electrical measurement capability. This was added in SM323 v2.

I suspect the problem is that the herdsman converter (samotech,ts) does not differentiate between v1 and v2 of the device.

    {
        zigbeeModel: ['SM323'],
        fingerprint: [{modelID: 'Dimmer-Switch-ZB3.0', manufacturerName: 'Samotech'}, {modelID: 'HK_DIM_A', manufacturerName: 'Samotech'}],
        model: 'SM323',
        vendor: 'Samotech',
        description: 'Zigbee retrofit dimmer 250W',
        **extend: [light({configureReporting: true}), electricityMeter()],**
    },

### What did you expect to happen?

Configuration of SM323 v1 to complete without errors

### How to reproduce it (minimal and precise)

This happens for all my Samotech SM323 v1 dimmers.

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

20230507

### Adapter

LAUNCHXL-CC1352P Evaluation board

### Setup

Addon on HAOS VM x86-64

### Debug log

Not debug. WIll update with debug when this issue manifests with debug enabled.

`Zigbee2MQTT:info  2024-02-18 11:24:00: Configuring 'Extension Downlights'
Zigbee2MQTT:info  2024-02-18 11:24:00: MQTT publish: topic 'zigbee2mqtt_cc1352p/Extension Downlights', payload '{"brightness":254,"current":null,"device":{"applicationVersion":0,"dateCode":"20230318","friendlyName":"Extension Downlights","hardwareVersion":0,"ieeeAddr":"0xf4b3b1fffef86133","manufacturerID":4107,"manufacturerName":"Samotech","model":"SM323","networkAddress":24532,"softwareBuildID":"
Zigbee2MQTT:info  2024-02-18 11:24:00: MQTT publish: topic 'zigbee2mqtt_cc1352p/Extension Downlights', payload '{"brightness":254,"current":null,"device":{"applicationVersion":0,"dateCode":"20230318","friendlyName":"Extension Downlights","hardwareVersion":0,"ieeeAddr":"0xf4b3b1fffef86133","manufacturerID":4107,"manufacturerName":"Samotech","model":"SM323","networkAddress":24532,"softwareBuildID":"
Zigbee2MQTT:info  2024-02-18 11:24:00: MQTT publish: topic 'zigbee2mqtt_cc1352p/Extension Downlights', payload '{"brightness":254,"current":null,"device":{"applicationVersion":0,"dateCode":"20230318","friendlyName":"Extension Downlights","hardwareVersion":0,"ieeeAddr":"0xf4b3b1fffef86133","manufacturerID":4107,"manufacturerName":"Samotech","model":"SM323","networkAddress":24532,"softwareBuildID":"
Zigbee2MQTT:info  2024-02-18 11:24:00: MQTT publish: topic 'zigbee2mqtt_cc1352p/Extension Downlights', payload '{"brightness":254,"current":null,"device":{"applicationVersion":0,"dateCode":"20230318","friendlyName":"Extension Downlights","hardwareVersion":0,"ieeeAddr":"0xf4b3b1fffef86133","manufacturerID":4107,"manufacturerName":"Samotech","model":"SM323","networkAddress":24532,"softwareBuildID":"
Zigbee2MQTT:info  2024-02-18 11:24:00: MQTT publish: topic 'zigbee2mqtt_cc1352p/Extension Downlights', payload '{"brightness":254,"current":null,"device":{"applicationVersion":0,"dateCode":"20230318","friendlyName":"Extension Downlights","hardwareVersion":0,"ieeeAddr":"0xf4b3b1fffef86133","manufacturerID":4107,"manufacturerName":"Samotech","model":"SM323","networkAddress":24532,"softwareBuildID":"
Zigbee2MQTT:error 2024-02-18 11:24:00: Failed to configure 'Extension Downlights', attempt 3 (Error: Device 0xf4b3b1fffef86133 has no input cluster haElectricalMeasurement
    at getEndpointsWithInputCluster (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:19:15)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:36)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:31)
    at Configure.configure (/app/lib/extension/configure.ts:120:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

Koenkk, 2024-02-18:
Can you provide the data/database.db entry of your device?

Note to self; [database.db entry for device supporting it](https://github.com/Koenkk/zigbee2mqtt/issues/19599#issuecomment-1854247671)

james-fry, 2024-02-18:
> Can you provide the data/database.db entry of your device?
> 
> Note to self; [database.db entry for device supporting it](https://github.com/Koenkk/zigbee2mqtt/issues/19599#issuecomment-1854247671)

Sure - this is what I see:

`{"id":70,"type":"Router","ieeeAddr":"0x0c4314fffe235d16","nwkAddr":17764,"manufId":4714,"manufName":"Samotech","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"Samotech","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09"}},"genLevelCtrl":{"attributes":{"currentLevel":87}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca604d0","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca604d0","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1660633525,"reporting":1},"lastSeen":1708257697033}
{"id":71,"type":"Router","ieeeAddr":"0x0c4314fffe1e4ddd","nwkAddr":19713,"manufId":4714,"manufName":"Samotech","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"Samotech","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09"}},"genLevelCtrl":{"attributes":{"currentLevel":246}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca604d0","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca604d0","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1660633525,"reporting":1},"lastSeen":1708275349783}
{"id":72,"type":"Router","ieeeAddr":"0x0c4314fffe1f9e01","nwkAddr":54527,"manufId":4714,"manufName":"Samotech","powerSource":"Mains (single phase)","modelId":"Dimmer-Switch-ZB3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Dimmer-Switch-ZB3.0","manufacturerName":"Samotech","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09"}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca604d0","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca604d0","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20200701","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1660633525,"reporting":1},"lastSeen":1708257696863}
{"id":78,"type":"Router","ieeeAddr":"0xf4b3b1fffef93cc6","nwkAddr":3261,"manufId":4107,"manufName":"Samotech","modelId":"SM323","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SM323","manufacturerName":"Samotech","powerSource":18,"zclVersion":8,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20230318","swBuildId":"1.00"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca604d0","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b001ca604d0","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"b [... cut ...] Change":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20230318","swBuildId":"1.00","zclVersion":8,"interviewCompleted":true,"meta":{"configured":-1660633525,"reporting":1},"lastSeen":1708274329241}
`

Koenkk, 2024-02-20:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-20 by Koenkk

james-fry, 2024-02-20:
> Fixed!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Fab. Big thanks, Koen!

james-fry, 2024-03-02:
> Fixed!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

I can see from the release notes that the fix forr this issue is supposed to be included in 1.36.0.
However I am still seeing the errors even with 1.36.0 (HA addon) being installed:

```
Zigbee2MQTT:error 2024-03-02 19:27:40: Failed to configure 'Garden Wall Lights', attempt 2 (Error: Device 0x0c4314fffe235d16 has no input cluster haElectricalMeasurement
    at getEndpointsWithInputCluster (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:21:15)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:36)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:31)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
```

I looked at the code change and it looks like even though you have separated V1 from V2, the electrical measurerments is still there in the final row of code:

    {
        // v1 doesn't support electricity measurements
        // https://github.com/Koenkk/zigbee2mqtt/issues/21449
        fingerprint: [{manufacturerName: 'Samotech', modelID: 'Dimmer-Switch-ZB3.0'}],
        model: 'SM323_v1',
        vendor: 'Samotech',
        description: 'Zigbee retrofit dimmer 250W',
        extend: [light({configureReporting: true}), **electricityMeter()**],
    },

Koenkk, 2024-03-02:
My bad, fixed it now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.