Koenkk/zigbee2mqtt #21450: [Feature request]: LiXee ZLinky_TIC add expose in Tempo

Descagir, 2024-02-18:
### Is your feature request related to a problem? Please describe

I would like to monitor in HA the following expose (status_register_tempo_jour) :
[image]
I have the following setup :
[image]

### Describe the solution you'd like

I am guessing that this is the following expose :
[image]
[image]
But it is not enable in Tarif Standard Tempo. If I add "DEMAIN" in TIC command Whitelist, report of DEMAIN doestn't work.

### Describe alternatives you've considered

nope.

### Additional context

Any help is welcome.

mildis, 2024-02-24:
According to reference doc [Enedis-NOI-CPT_54E](https://www.enedis.fr/media/2035/download), section 6.2.3.14, 'Registre de statuts', tomorrow_color is only set for *legacy mode*.  
For all others mode including 'standard mode', this value is set to 0, "not announced".

Descagir, 2024-02-24:
Thank you for your time and answer. 
Nevertheless, as you can see in my first post and the following picture, the "status_register_tempo_jour : BLANC" is working and the value is in the log.
[image]
Is it possible to add this register, in the expose section ?
It's interesting to monitor the color of the day. In function, (ex: if it's ROUGE (RED)), we can turn off the switches.

mildis, 2024-02-24:
You should get the current color in the `LTARF` exposed value.  
It matches the `current_price` key in the json.  

Status register provides tempo color (`tempo_jour`) and HP/HC (`tarif_dist` and `tarif_four`) separately.

Descagir, 2024-02-24:
Indeed.
[image]
I didn't guess that. Thank you.
Is it possible to only have the tempo color : "BLEU", "BLANC" et "ROUGE" ?
Thank you in advance.

mildis, 2024-02-24:
Unfortunately no.  
This is a value sent by the provider, so different providers could name their tariff differently.  
Register status is decoded per Enedis specs until another provider arise.

Descagir, 2024-02-24:
Ok.
In HA, is it possible to extract only the second term of the expose (so, the color) ? And put the second term in a new variable ? 
In Python, it's ok (=> str.split()[1]). But in HA..
Thank you for your time.

mildis, 2024-02-24:
Sorry, I don't know HA (I'm using Jeedom :) )
Look for action/methods like 'contains' or 'endWith' to match the terms.

closed 2024-09-22 by github-actions[bot]