Koenkk/zigbee2mqtt #21452: Issue with nested topics after 8a9477f

mundschenk-at, 2024-02-18:
### What happened?

I have set up my topics by room, with `some_room` being a group containing all the lights in that room and `some_room/light1`, `some_room/light2` ... being the topics for the individual devices and `some_room/groupA` being a group. I was previously on `dev` but some commits back, before `1.35.3`. 

Yesterday, I updated to the current `dev` (5766bf4) and initially didn't note anything out of the ordinary. However, when I later tried to use a Hue dimmer switch in another room (set to toggle various groups and lamps individually), I was puzzled that trying to toggle a single light toggled the entire room. Further investigation showed that only the top-level group for a room could still be controlled.

Looking at the newish commits, I am assuming this is related to changes to endpoint matching. Rolling back to `dev` commit 5766bf4 fixed the issue for me.

### What did you expect to happen?

I should be able to control separate devices and groups separately like before.

### Zigbee2MQTT version

dev

### Adapter firmware version

20231112

### Adapter

Sonoff P

### Setup

Plain in LXC container on NUC

Koenkk, 2024-02-18:
Looking at the [changes of 18.33.2](https://github.com/Koenkk/zigbee-herdsman-converters/commits/v18.33.2/) I cannot explain it, are you sure that commit fixes it?

mundschenk-at, 2024-02-18:
I haven't dissected the issue, I just switched back to when the latest stable version was split off (this is my production network).

mundschenk-at, 2024-02-18:
@Koenkk Sorry for the mixup. I've updated the commits I tried in the report. I suspect the issue is the result of https://github.com/Koenkk/zigbee2mqtt/pull/21214 or the changes deriving from it, but I have not investigated this deeper at this point.

Koenkk, 2024-02-19:
I expect so, @grafalex82 could you take a look at this?

grafalex82, 2024-02-20:
@Koenkk Sure, looking into this. Can it be related to [group state handling](https://www.zigbee2mqtt.io/guide/usage/groups.html#state-changes)?

@mundschenk-at I am trying to re-create your set up. Perhaps I could not get it properly from your description. Here is what I did:
- Created 2 lights named `room/light1` and `room/light2`
- Created a group named `room/group`
- I do not have a dimmer, so I took a regular 2-gang switch, and bound its buttons to lights 1 and 2

From what I see, sending state change to the group can on and off both lights. Also using 2-gang switch I can on/off lights individually, and other light in the group does not change its state.

Could you please provide more details, and preferrably Z2M debug logs?

grafalex82, 2024-02-20:
FYI: [fixed a minor bug](https://github.com/Koenkk/zigbee2mqtt/pull/21493) related to endpoint name parsing. Not sure this is related to this problem, but just in case.

mundschenk-at, 2024-02-20:
> * Created 2 lights named `room/light1` and `room/light2`
> * Created a group named `room/group`
> * I do not have a dimmer, so I took a regular 2-gang switch, and bound its buttons to lights 1 and 2

I don't have a minimal test case, but the scenario is a bit different:

- Light 1 named `room/light1`
- Light 2 named `room/light2`
- Light 3 named `room/light3`
- Group 1 named `room/group1` (containing Light 1 and Light 2)
- Group 2 named simply `room` (containing all three lights)

The dimmer switch is named `room/dimmer`, but it is not that important in the scenario because it is only bound to the coordinator and all interactions happen over MQTT. I am assuming the issue happens because the topic for Group 2 is also the prefix for the other devices.

This is not tested, but a minimal case to reproduce would probably be:

- Light 1 `room/light1`
- Light 2 `room/light2`
- Group `room`

grafalex82, 2024-02-20:
@mundschenk-at Thank you for the description. It does make sense.

This indeed relates to my recent changes. However, this is not a regression - this rather an expected change in how names are parsed. The problem is that the mqtt API is ambiguous, and the question is which limitations we set on the naming to avoid this ambiguouty. 

Here is what happening. When it sees a name like `room/group1` it can't say in advance whether this is device named `room` with endpoint `group1`, or a device/group named 'room/group1`. So it has to guess the first option, and then try the second. In your case the `room` matches first. 

Perhaps parsing a group name would not require separation of the name and endpoint name. But since mqtt topic for `set` operation is the same for groups and devices we can't say in advance whether we match device name + endpoint, or a group name.

I can change the order of matching, so that `room/group1` is matched first, and `room` + `group1` next. This will fix your issue, but break someone'w case with endpoint named `group1` in the device `room`. 

I can also add a heuristic, like if it sees a `room/group1` and `room` is a group, then it definitely can't be a device+endpoint name case.

In the meantime I would suggest renaming outer group `room` to something like `room/all` and it will solve your issue.

> Sounds very much like this could be related, yes.

Please test. However the fix is related to a different case with `long/names/with/slashes`.

mundschenk-at, 2024-02-20:
Regarding the just merged PR: I read through it again and I agree, that's not it after all.

> Perhaps parsing a group name would not require separation of the name and endpoint name. But since mqtt topic for set operation is the same for groups and devices we can't say in advance whether we match device name + endpoint, or a group name.

That would only fix my use case, but not more general one as someone else could have named a device "room" (probably not literally with a room name, but a singular topic that is the prefix for another topic.

However, I'm not quite sure how things break because what I actually see is that a command to one of the lights triggers the group (e.g. `room/light1/set` affects `room`). I can see how partial matching might occur, but surely there must be some there is no real ambiguity for the full topic name? (The lights don't actually start with `light`, so there should be no overlap with endpoint names.) Or maybe I am missing something here?

Since the docs encourage hierarchical structuring using the `friendly_name` and it follows the restrictions laid out (no reserved endpoint names, no bare numbers), I do think this a regression. At the very least it needs to be documented as a breaking change, if it is indeed unavoidable.

grafalex82, 2024-02-21:
Hopefully [this PR](https://github.com/Koenkk/zigbee2mqtt/pull/21521) will fix the issue.

grafalex82, 2024-02-23:
@mundschenk-at The fix has been merged to dev. Could you please check?

mundschenk-at, 2024-02-23:
Just updated, looking good 👍

closed 2024-02-23 by mundschenk-at