Koenkk/zigbee2mqtt #21453: Zigbee2Mqtt stops without any error message

saucekorn69, 2024-02-18:
### What happened?

Hello,

I run Zigbee2Mqtt into a Debian 11 VM on my NAS.

I followed these instructions: https://www.zigbee2mqtt.io/guide/installation/01_linux.html#configuring

I mounted a Popp usb stick which is on USB0.

Extract of my configuration.yaml

      homeassistant: false
      frontend: true
      mqtt:
        base_topic: zigbee2mqtt
        server: mqtt://localhost
      serial:
        port: /dev/ttyUSB0
        adapter: ezsp
      advanced:
        log_level: debug
        network_key: XXX
       
        homeassistant_legacy_entity_attributes: false
        legacy_api: false
        legacy_availability_payload: false
      device_options:
        legacy: false

Here is what I have when I start the Zigbee2Mqtt (npm start)

> zigbee2mqtt@1.35.3 start
> node index.js

  Zigbee2MQTT:debug 2024-02-18 14:50:35: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
  Zigbee2MQTT:info  2024-02-18 14:50:35: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-02-18.14-50-35' filename: log.txt
  Zigbee2MQTT:info  2024-02-18 14:50:35: Starting Zigbee2MQTT version 1.35.3 (commit #fe0742a)
  Zigbee2MQTT:info  2024-02-18 14:50:35: Starting zigbee-herdsman (0.33.8)
  Zigbee2MQTT:debug 2024-02-18 14:50:35: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[233,2,2,60,58,100,186,33],"networkKey":"HIDDEN","panID":35696},"serialPort":{"adapter":"ezsp","path":"/dev/ttyUSB0"}}'
  Zigbee2MQTT:info  2024-02-18 14:50:38: zigbee-herdsman started (resumed)
  Zigbee2MQTT:info  2024-02-18 14:50:38: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
  Zigbee2MQTT:debug 2024-02-18 14:50:38: Zigbee network parameters: {"channel":11,"extendedPanID":233,"panID":35696}
  Zigbee2MQTT:info  2024-02-18 14:50:38: Currently 0 devices are joined:
  Zigbee2MQTT:info  2024-02-18 14:50:38: Zigbee: disabling joining new devices.
  Zigbee2MQTT:info  2024-02-18 14:50:38: Connecting to MQTT server at mqtt://localhost
  Zigbee2MQTT:debug 2024-02-18 14:50:38: Using MQTT anonymous login
  Zigbee2MQTT:error 2024-02-18 14:50:38: MQTT error:
  Zigbee2MQTT:error 2024-02-18 14:50:38: MQTT failed to connect, exiting...
  Zigbee2MQTT:info  2024-02-18 14:50:38: Stopping zigbee-herdsman...
  Zigbee2MQTT:info  2024-02-18 14:50:39: Stopped zigbee-herdsman

Content of the log file:

debug 2024-02-18 14:50:35: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
info  2024-02-18 14:50:35: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-02-18.14-50->
info  2024-02-18 14:50:35: Starting Zigbee2MQTT version 1.35.3 (commit #fe0742a)
info  2024-02-18 14:50:35: Starting zigbee-herdsman (0.33.8)
debug 2024-02-18 14:50:35: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":nu>
info  2024-02-18 14:50:38: zigbee-herdsman started (resumed)
info  2024-02-18 14:50:38: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorr>
debug 2024-02-18 14:50:38: Zigbee network parameters: {"channel":11,"extendedPanID":233,"panID":35696}
info  2024-02-18 14:50:38: Currently 0 devices are joined:
info  2024-02-18 14:50:38: Zigbee: disabling joining new devices.
info  2024-02-18 14:50:38: Connecting to MQTT server at mqtt://localhost
debug 2024-02-18 14:50:38: Using MQTT anonymous login
error 2024-02-18 14:50:38: MQTT error:
error 2024-02-18 14:50:38: MQTT failed to connect, exiting...
info  2024-02-18 14:50:38: Stopping zigbee-herdsman...
info  2024-02-18 14:50:39: Stopped zigbee-herdsman

What should I do to understand what is the problem ?

I am not an experienced user in Linux.

Thank you !

Nicolas.

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211210

### Adapter

POPP Zigbee Stick

### Setup

Debian 11 VM on NAS Synology

### Debug log

debug 2024-02-18 14:50:35: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
info  2024-02-18 14:50:35: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-02-18.14-50->
info  2024-02-18 14:50:35: Starting Zigbee2MQTT version 1.35.3 (commit #fe0742a)
info  2024-02-18 14:50:35: Starting zigbee-herdsman (0.33.8)
debug 2024-02-18 14:50:35: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":nu>
info  2024-02-18 14:50:38: zigbee-herdsman started (resumed)
info  2024-02-18 14:50:38: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorr>
debug 2024-02-18 14:50:38: Zigbee network parameters: {"channel":11,"extendedPanID":233,"panID":35696}
info  2024-02-18 14:50:38: Currently 0 devices are joined:
info  2024-02-18 14:50:38: Zigbee: disabling joining new devices.
info  2024-02-18 14:50:38: Connecting to MQTT server at mqtt://localhost
debug 2024-02-18 14:50:38: Using MQTT anonymous login
error 2024-02-18 14:50:38: MQTT error:
error 2024-02-18 14:50:38: MQTT failed to connect, exiting...
info  2024-02-18 14:50:38: Stopping zigbee-herdsman...
info  2024-02-18 14:50:39: Stopped zigbee-herdsman

Chartreusito, 2024-02-18:
> `info 2024-02-18 14:50:38: Connecting to MQTT server at mqtt://localhost
debug 2024-02-18 14:50:38: Using MQTT anonymous login
error 2024-02-18 14:50:38: MQTT error:
error 2024-02-18 14:50:38: MQTT failed to connect, exiting...`

I would advise you to check your mqtt configuration. Take another pass on the docs for this part and it should be better.

saucekorn69, 2024-02-23:
Problem solved, configuration.yaml wast not pointing to the proper broker IP.