Koenkk/zigbee2mqtt #21457: [New device support]: SCHNEIDER ELECTRIC MUR36014

BruceLIOU, 2024-02-18:
### Link

https://www.se.com/fr/fr/product/MUR36014/mureva-evlink-prise-courant-2p+t-renforc%C3%A9e-pour-ve-ip66-ik08-gris/?%3Frange=226303288-prises-de-recharge-mureva-evlink&selectedNodeId=109455364809

### Database entry

{"id":30,"type":"Router","ieeeAddr":"0x70ac08fffe03fc81","nwkAddr":36618,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Mains (single phase)","modelId":"EVSCKT/OUTLET/1","epList":[6,242],"endpoints":{"6":{"profId":260,"epId":6,"devId":9,"inClusterList":[0,3,4,5,6,1794,1800,2820,2821,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"EVSCKT/OUTLET/1","manufacturerName":"Schneider Electric","powerSource":1,"zclVersion":3,"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.010.000 R"}},"genOnOff":{"attributes":{"onOff":1}},"haElectricalMeasurement":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff06f0e7","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":6,"hwVersion":1,"dateCode":"","swBuildId":"002.010.000 R","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1708273491823}

### Comments

Hi,
This equipment is recognized under ZHA but not under Z2MQTT.
I tried to follow the procedure for its integration but without success.
I also cloned the zigbee2mqtt.io repo in order to make a pull request for the image of this team but again without success.
Thank you in advance for its integration.
This is a reinforced socket for mainly charging electric vehicles.
It allows you to turn the socket off and on as well as monitor the electricity consumption on this socket.

### External definition

```shell
const {onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['EVSCKT/OUTLET/1'],
    model: 'EVSCKT/OUTLET/1',
    vendor: 'Schneider Electric',
    description: 'Reinforced power socket',
    extend: [onOff({"powerOnBehavior":false}), electricityMeter()],
    meta: {},
};

module.exports = definition;
```

Koenkk, 2024-02-19:
with the external definition, does the device work?

BruceLIOU, 2024-02-20:
Hi , the device work but energy is not send
{
    "current": 0,
    "linkquality": 255,
    "power": 0,
    "state": "OFF",
    "voltage": 240,
    "energy": null
}

Koenkk, 2024-02-22:
Does the configure succeed? You can do it via the z2m frontend -> device -> yellow refresh button

BruceLIOU, 2024-02-22:
When i refresh this device i got this error
`Failed to configure 'Prise VE', attempt 5 (Error: Read 0x70ac08fffe03fc81/6 haElectricalMeasurement(["acPowerDivisor","acPowerMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28) at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:469:22) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:222:25) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25) at Configure.configure (/app/lib/extension/configure.ts:120:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:55:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

Koenkk, 2024-02-24:
Can you try again after replacing `electricityMeter()` with `electricityMeter({current: {divisor: 1000}, voltage: {divisor: 1}, power: {divisor: 1}, energy: {divisor: 100}}),`. The divisors are probably wrong, e.g. power might be multiplied by 10 or 100 so update the converter accordingly and reconfigure.

BruceLIOU, 2024-03-06:
Hi, energy don't have value...
[image]

Koenkk, 2024-03-07:
What is you read the `currentSummDelivered`?

[image]

BruceLIOU, 2024-03-08:
This return an error

Koenkk, 2024-03-08:
What's the error?

BruceLIOU, 2024-03-08:
[image]
No return for node 6 and this for 242

Koenkk, 2024-03-09:
What if you read the `genBasic` cluster `zclVersion`

BruceLIOU, 2024-03-09:
Nothing return for this...

Koenkk, 2024-03-10:
When pairing with Home Assistant, does it show the `energy`?

BruceLIOU, 2024-03-10:
No energy is always empty (no value)

Koenkk, 2024-03-10:
So doesn't seem to be supported, try with:

```
electricityMeter({cluster: 'electrical',  current: {divisor: 1000}, voltage: {divisor: 1}, power: {divisor: 1}}),
```

Play a bit with the divisor numbers, because what I see from your screenshot the values reported are too high.

BruceLIOU, 2024-03-11:
After restart Z2M, i have this into log
`Zigbee2MQTT:error 2024-03-11 08:18:29: Failed to configure 'Prise VE', attempt 1 (Error: ConfigureReporting 0x70ac08fffe03fc81/6 haElectricalMeasurement([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"activePower"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":50,"attribute":"rmsCurrent"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"rmsVoltage"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:672:22)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:57:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:235:21)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-03-11 08:18:32: Failed to configure 'Prise VE', attempt 2 (Error: ConfigureReporting 0x70ac08fffe03fc81/6 haElectricalMeasurement([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"activePower"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":50,"attribute":"rmsCurrent"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"rmsVoltage"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:672:22)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:57:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:235:21)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))
Zigbee2MQTT:error 2024-03-11 08:18:37: Failed to configure 'Prise VE', attempt 3 (Error: ConfigureReporting 0x70ac08fffe03fc81/6 haElectricalMeasurement([{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"activePower"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":50,"attribute":"rmsCurrent"},{"minimumReportInterval":10,"maximumReportInterval":65000,"reportableChange":5,"attribute":"rmsVoltage"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:672:22)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:57:13)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:235:21)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)
    at Configure.configure (/app/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

Koenkk, 2024-03-11:
Looks that the `power` is also not supported, does it work when pairing with HA?

BruceLIOU, 2024-03-13:
I come from Jeedom (French solution) with their zigbee plugin and everything was working. From memory everything also went back to ZHA. It's strange

BruceLIOU, 2024-03-16:
Hi,
I reset the proposed script to default and reconfigured this device and here are the error messages received in the logs:
[image]

Pepeanuts, 2024-03-22:
Hi, try to contact @LGO44 on https://community.home-assistant.io/t/drayton-wiser-home-assistant-integration/80965/3550.
It's a french active developpers on Wiser HUB integration for HA, maybe he could help you with this product and get logs from Wiser HUB.

vincemonn, 2024-04-02:
Hi,
Same issue...
@BruceLIOU have you found a solution ? I'm also on Jeedom.

Thanks

BruceLIOU, 2024-04-03:
Hi @vincemonn,
No for moments...

Micka41, 2024-05-06:
> What if you read the `genBasic` cluster `zclVersion`

Info 2024-05-06 18:20:01Read result of 'genBasic': {"zclVersion":3}

Micka41, 2024-05-06:
> What if you read the `genBasic` cluster `zclVersion`

Read result of 'seMetering': {"currentSummDelivered":[0,0],"status":0,"meteringDeviceType":0,"extendedStatus":"0x0000000000000000","unitOfMeasure":0,"multiplier":1,"divisor":1000000,"summaFormatting":75,"instantaneousDemand":0}
Read result of 'haElectricalMeasurement': {"measurementType":8,"rmsVoltage":238,"rmsCurrent":0}

BruceLIOU, 2024-05-07:
@Micka41 do you use generated_external_definition ?

Micka41, 2024-05-07:
> @Micka41 do you use generated_external_definition ?
No, I did the tests “raw”: I added the plug and interrogated all the “seMetering” and “haElectricalMeasurement” parameters by plugging it into my coffee machine (the plug isn't fixed yet and I don't have the electric car yet).

BruceLIOU, 2024-05-07:
@Micka41 do tou have this conf into Link
[image]

Micka41, 2024-05-07:
My config :
Expose :
[image]
Lien :
[image]
Rapports :
[image]

The amperage must be divided by 1000, but I don't know how to do that.

Micka41, 2024-05-07:
Manually queries the “summaFormatting” and “currentSummDelivered” attributes in the seMetering cluster once. This should then update itself.

sky-future, 2024-06-22:
@BruceLIOU Did you fix it ? With zigbee2mqtt ?

BruceLIOU, 2024-07-16:
Hi, i have same problem...
@Micka41 nothing when call this properties
@sky-future no this problem is on zigbee2mqtt

maquidoux, 2024-12-16:
Hi @BruceLIOU ,
Have you successfully configured the device and retrieved the information in zigbee2mqtt? 
I can't make the initial pairing and I'd like to know what configuration to change / add and in which file?
Thanks a lot!

adrien-parasote, 2025-01-03:
Hello Team,

Did someone find the solution ? I'm running the same issue without any clue what to do next to fix it :'(

PS : Happy new year :)

maquidoux, 2025-01-03:
Hi @adrien-parasote and happy new year too!
Everything works for me. I reset the MUREVA socket and added a Zigbee router (NOUS A1Z socket) close to the MUREVA. My bridge was too far away I think. 
On the other hand, I also had the problem of measuring with too high  multiplier. After modifying the socket's .js file, everything is OK.

adrien-parasote, 2025-01-04:
Hi,

ok that’s strange i’ve a zigbee router at arround 7m…
I’ll try next week to re-do the pairing sitting with my conbee3 near to the device

keep you in touch

adrien-parasote, 2025-02-19:
Hello team, 

I'm back to give feedbacks (sorry it takes long time for me because of personal issues :'( )
First of all : I do the pairing again with my conbee3 at 50cm of the device and even if the first load didn't work the second automatic try work fine \o/

Then I face same issues as described : I was able to play with the switch button but no data uploaded to HA. So I follow theses steps : 

1. enable `genBasic`, `haElectricalMeasurement` & `seMetering`
2. go to `expose` tab and click on reload button for `Current` then `Voltage` then `Power`
At that stage I got the following message 

```
error 2025-02-19: Publish 'get' 'power' to 'Mureva EVlink' failed: 'Error: ZCL command 0x70ac08fffe040510/6 haElectricalMeasurement.read(["activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

Nevertheless, data are still coming to HA as you can see bellow

<details>
<summary>Show screen</summary>

[image]

</details>

Pain points that need to be taken care of : 
- Power & Energy : need to be divided by 1000
- Current : need to be divided by 100

So what I need in the config is the following

```
electricityMeter({cluster: 'electrical',  current: {divisor: 100}, voltage: {divisor: 1}, power: {divisor: 1000}})
```

I understand I need to play with the configuration but, and sorry for the stupid question, how to do that ?

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/schneider_electric.ts

adrien-parasote, 2025-02-19:
Hello (again),

After playing with the dev console with those parameters: 
   - cluster: haElectricalMeasurement
   - attribute: 
      - acCurrentDivision
      - rmsCurrent
      - rmsVoltage

I got the following answer:
```
zhc:tz: Read result of 'haElectricalMeasurement': {"acCurrentDivisor":1000,"rmsCurrent":9848,"rmsVoltage":221}
```
And by 🪄magic🪄, data exposed are now good for those fields...

Also:
- there is no way as per the doc [here](https://www.zigbee2mqtt.io/devices/MUR36014.html) (and many personal tries) to set division values.
- `Power` sensor cannot be obtained via UI. When I tried to refresh, I got : 
```
z2m: Publish 'get' 'power' to 'Mureva_EVlink' failed: 'Error: ZCL command 0x70ac08fffe040510/6 haElectricalMeasurement.read(["activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
```

_Note:  a basic calcul W = V * A allow me to think `Power` data progress, seems to be better now_

- As `Energy` is sum of `Power`, if previous data are right those one also.

I will monitor next few days the data and be back to you

rissson, 2025-03-29:
Heya, found this issue while setting up a MUR36014 and encountering weird values for Power, Current and Energy:

[image]

@adrien-parasote, what do you mean by "playing with the dev console with those parameters"? It seems I have just about the same values as you:
```json
{"acCurrentDivisor":1000,"rmsCurrent":23,"rmsVoltage":233}
```

adrien-parasote, 2025-03-30:
Hi @rissson I requested data as you did with the dev console and so the current value was back on track in terms of magnitude (assuming you don't consume 23A 😬)

In all cases, and as you can see in my last screenshot, power report was still too high...
230*10 = 2 300W not 2 300kW. For that part as the plug doesn't allow me to have a correction, I build a sensor template to correct it

closed 2025-06-07 by github-actions[bot]

schauveau, 2025-09-18:
I have a working external converter with correct units.

```
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

// power   in seMetering::instantaneousDemand
// energy  in seMetering::currentSummDelivered
// current in haElectricalMeasurement::rmsCurrent
// voltage in haElectricalMeasurement::rmsVoltage

export default {
    zigbeeModel: ['EVSCKT/OUTLET/1'],
    model: 'EVSCKT/OUTLET/1',
    vendor: 'Schneider Electric',
    description: 'Mureva EvLink',
    extend: [
      m.onOff({"powerOnBehavior":false}),
      m.electricityMeter(
       {
         cluster: "metering",        
         energy:  {multiplier:1 , divisor: 1000000},
         power:   {multiplier:1 , divisor: 1000000}
       }
      ),     
      m.electricityMeter(
       {
         voltage: {multiplier:1 , divisor: 1},
         current: {multiplier:1 , divisor: 1000},
         energy:  false,
         power:   false,
       }
      ),
    ],
    meta: {},
};
```

Notice how I use `electricityMeter` twice. First to define `energy` and `power` in cluster`seMetering` and then to define `voltage` and `current` in cluster`haElectricalMeasurement`.  

I believe that the problem was that there it is also possible to get power values from the `haElectricalMeasurement` cluster. 

Also, `seMetering` only use one multiplier and one divisor register for both energy and power so the values passed to `electicityMeter` must be strictly identical. Since power was incorrectly configured in `haElectricalMeasurement`, the scaling was wrong for both energy and power. 

Reminder: after updating the external converter, restart Z2M and make sure that you 
reconfigure the device in the Devices page otherwise the new settings may not be correctly applied.

schauveau, 2025-09-19:
I will try.

schauveau, 2025-09-19:
My latest converter with a configurable powerOnBehavior. 

I also noticed in the code that electricity meter accepts a `cluster` argument for `power`. 
This is probably the intended way to solve the power issue.

```js
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

// Known registers
//   power  = seMetering::instantaneousDemand
//   energy = seMetering::currentSummDelivered
//   current = haElectricalMeasurement::rmsCurrent
//   voltage = haElectricalMeasurement::rmsVoltage

// The following registers are readable and may contain useful information
// such as an error flag when overheating (TESTS NEEDED) 
//  - seMetering::status
//  - seMetering::extendedStatus
//  - haDiagnostic::schneiderMeterStatus
//  - haDiagnostic::schneiderDiagnosticRegister1
// 

export default {
    zigbeeModel: ['EVSCKT/OUTLET/1'],
    model: 'EVSCKT/OUTLET/1',
    vendor: 'Schneider Electric',
    description: 'Mureva EvLink',
    extend: [
      m.onOff(),
      m.electricityMeter(
       {
         energy:  {multiplier:1 , divisor: 1000000},
         power:   {multiplier:1 , divisor: 1000000, cluster: "metering"},
         voltage: {multiplier:1 , divisor: 1},
         current: {multiplier:1 , divisor: 1000},
       }
      ),
    ],
    meta: {},
};

```

schauveau, 2025-09-19:
After some experiments, here is the converter I am going to use in my pull request

```js
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['EVSCKT/OUTLET/1'],
    model: 'EVSCKT/OUTLET/1',
    vendor: 'Schneider Electric',
    description: 'Mureva EvLink',
    extend: [
      m.onOff(),
      m.electricityMeter(
       {
           voltage: {multiplier: 1, divisor: 1},
           power: {cluster:"metering"}
       }
      ),
    ],
    meta: {},
};
```

After a reset of the device, I noticed that the multipliers and divisors are correctly set by default: 
- for current: the device sets `acCurrentMultiplier=1` and `acCurrentDivisor=1000` in cluster `haElectricalMeasurement`.
- for power and energy: the device sets `multiplier=1` and `divisor=1000000` in cluster `seMetering`.

However, the attributes `acVoltageMultiplier` and `acVoltageDivisor` are missing in `haElectricalMeasurement` which causes an error that breaks most of the configuration.  This is why both values MUST be explicitly specified for voltage.

The argument `cluster::metering` is also needed to get the power from cluster`seMetering` instead of the default `haElectricalMeasurement`. 

Note: A device reconfiguration (the orange buttons found in the Devices page) or a full device reset & repairing may be required to set all multipliers or divisors to the proper value. That can also be done manually in the Dev Console using the values given above.

schauveau, 2025-09-19:
Cluster 1800 = 0x0708 is not implemented by Z2M. It is documented in the ZCL (Zigbee Cluster Library) as **Device Management for Smart Energy Network**. It is does Supplier Control, Tenancy Control and Password Control.  It could be that the device lock operates using a default password that is only known by the Wiser Home app. If so, that feature may be difficult to replicate in Z2M.

adrien-parasote, 2025-10-04:
Hey, it's working finally fine ! Lot of thank @schauveau !