Koenkk/zigbee2mqtt #21458: TuYa TS0044 messed up in 1.35.3-1

TomaamoT7, 2024-02-18:
### What happened?

Changelog for 1.35.3-1 says that it was addressing issue with TuYa TS0042:
#21196 Fix no actions for TuYa TS0042 (@Koenkk)
Unfortunatelly it messed up TuYa TS0044 devices.

After the upgrade to 1.35.3-1 the TuYa TS0044 devices stopped working. They also show incorrect device picture (now they look like remote with buttons instead of 4-way switch).
Everything is working ok after rollback to  previous version (1.35.2-1). 

Zigbee2MQTT
Installed version
1.35.2-1
Latest version
1.35.3-1

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

0x26660700

### Adapter

Conbee II

### Setup

Add-on on Home Assistant running directly on Raspberry Pi4

Koenkk, 2024-02-19:
Can you provide the data/database.db entry of your device?

TomaamoT7, 2024-02-20:
Sure. I have 7 of them bought from different sellers in different time. They have minimal mechanical differences but they are all reported the same way (same manufacturer and model id). Here's one example (see at the bottom) I extracted it directly as text from database.db file.

One additional info: in order to avoid missed clicks and high battery drain I applied (months ago) the following fix:
https://community.home-assistant.io/t/tuya-ts0044-first-action-ignored/475940/2
I can confirm that with the fix there are no missed clicks and I don't need to replace batteries every 2-3 months in them.
Anyway most likely it has nothing to do with the current issue.

Second comment: looks like the zigbee2mqtt.io site is wrong now (it used to be ok).
Both the remote (that I don't have) and the switch are shown as the same device.
https://www.zigbee2mqtt.io/devices/TS0044.html   -> this should be the switch (that was working fine, but is messed up right now, picture in the atachement
[image]
)
https://www.zigbee2mqtt.io/devices/TS0044_1.html  ->this is most likely the remote (I don't have it and I can't confirm)
Both share the same (or similar name) and that may be the issue??

{"id":4,"type":"EndDevice","ieeeAddr":"0xcc86ecfffe168bd0","nwkAddr":63031,"manufId":4098,"manufName":"_TZ3000_wkai4ga5","powerSource":"Battery","modelId":"TS0044","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,1,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"uO�,\u0019\u0000\u0000\u0000\u0000\u0007\u0000\u0000\u0000\u0000\u0011","65506":31,"65508":1,"65534":0,"modelId":"TS0044","manufacturerName":"_TZ3000_wkai4ga5","powerSource":3,"zclVersion":3,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff07ba7d","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":0,"inClusterList":[1,6],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-302029049},"lastSeen":1708464576726}

Koenkk, 2024-02-22:
I've fixed the detection part!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

brinzlee, 2024-03-05:
I have installed the updated version 1.36 which is supposed to address this issue and it still shows the wrong icons as described above. Restoring to 1.35.2 corrects this discrepancy

Koenkk, 2024-03-05:
can you provide a screenshot of the about screen with 1.35.2?

brinzlee, 2024-03-05:
@Koenkk 
[image]
[image]

Koenkk, 2024-03-05:
Fixed the picture, may take some time for the cache to refresh.

brinzlee, 2024-03-05:
@Koenkk Thanks I refreshed the cache and the first one is fixed but the second one is still wrong icon. Its a different model number but the same switch visually  
[image]

Koenkk, 2024-03-06:
fixed!

Assuming this issue can be closed now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-03-06 by Koenkk

TomaamoT7, 2024-03-28:
@Koenkk: The pictures are still wrong. They are switched now.
https://www.zigbee2mqtt.io/devices/TS0044.html -> this should be the switch picture (square, 4 buttons)
[image]

https://www.zigbee2mqtt.io/devices/TS0044_1.html -> this should be the remote picture (rectangle remote like)
[image]

It's only about the pictures, functionality is correct (at least for the devices that I have.

Koenkk, 2024-03-28:
fixed!

brinzlee, 2024-04-02:
@koenkk. I am very confused how these updates work. And how the icons update. Is this something that happens when the addon is updated or something that is cloud updated. When I updated the app from 1.35.2 to 1.36 the icons were incorrect on
_TZ3000_uaa99arv and _TZ3000_wkai4ga5. You mentioned the cache would take some time to update and after a while _TZ3000_wkai4ga5 proceeded to the correct icon but _TZ3000_uaa99arv never did....Now I have updated to 1.36.1 and both icons have reverted back to the incorrect picture. Is this an update issue or still the cache misbehaving ?

TomaamoT7, 2024-04-02:
For me the fix that was applied on 28 March by Koenkk corrected the pictures issue. Now they are displayed correctly.
All of my switches are _TZ3000_wkai4ga5, I don't have any other.

To the best of my knowledge it's just the cloud data update, so no need to download/install anything... 
I was also quite confused about how it works. For me the pictures were switched from 5th to 28th March.

TomaamoT7, 2024-04-04:
@Koenkk: the story continues...
I updated to zigbee2mqtt_1.36.0-1 and the images are wrong again.

This is what I have now:
[image]

_TZ3000_wkai4ga5 should be 4 button SWITCH.
https://www.zigbee2mqtt.io/devices/TS0044.html

Koenkk, 2024-04-04:
@TomaamoT7 I don't understand, in https://github.com/Koenkk/zigbee2mqtt/issues/21458#issuecomment-2024960636 you mentioned that `TS0044_1` (which is the device you have) should use this picture:

[image]

brinzlee, 2024-04-04:
I think there is a lot of confusion now in this thread. My pictures that I referred to are now showing the the 4 button rectangular remote control when they should both refer to the 4 button square panel with green leds.

aoaite, 2024-04-07:
My TS044 (the square one) stopped working after update. It is now incorrectly recognized as TS0044_1(the long one).

Koenkk, 2024-04-08:
@brinzlee for each manufacturer name (starting with _TZ) can you indicate if it should be the long remote or the square 4 button remote picture?

brinzlee, 2024-04-08:
@Koenkk 

_TZ3000_wkai4ga5
_TZ3000_uaa99arv

**these should both be the square 4 button remotes with the green LEDs.**

Using the current version 1.36.1-1 of the Home Assistant Addon they are being shown as the the long remote which is incorrect

Thanks for your kind help

Koenkk, 2024-04-08:
@brinzlee done!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.