Koenkk/zigbee2mqtt #21463: ZigBee2MQTT service continuesly restarting

its-Zeke, 2024-02-19:
### What happened?

My zigBee2mqtt service is starting successfully. In the log we can find the 'ZigBee2MQTT started' and it also runs as expected.
However, after around 1,5 minute the service daemon is logging: 'zigbee2mqtt.service: start operation timed out. Terminating.'
After that it restarts, the service runs for another 1,5 minute and is failing again. 
[image]

When i manually start the zigbee by 'npm start', or 'node index.js', it works fine and keeps running. Thus, i think the daemon doesn't get the reply, that the service started successfully somehow ?

My zigbee2mqtt.service config is the same as in the install documentation, except for the user that im running it (also tried 'root'):
[image]

### What did you expect to happen?

I enable the zigbee2mqtt.service in the systemd, it successfully starts and keeps running.

### How to reproduce it (minimal and precise)

I just start the systemd service and it keeps looping the starting -> failing due to timeout -> restarting thing.

### Zigbee2MQTT version

1.35.3 commit: fe0742a

### Adapter firmware version

20210708

### Adapter

Sonoff ZBDongle-P

### Setup

Running it plain on a Pi 4B

its-Zeke, 2024-02-19:
This is what the service status looks like while starting, even though the logging says it already started:
[image]

It remains in 'activating (start)'..

Koenkk, 2024-02-19:
Is this fixed after installing the following package?

```
sudo apt-get install -y  libsystemd-dev
```

staleand, 2024-02-19:
I have the same issue running 1.35.3-dev commit: [5766bf47] (ran into the issue on the current stable version as well, that's why I updated to the dev commit). Z2M runs for about ~80sec and then restarts.

If I start it manually with "node index.js" it works without crashing.

its-Zeke, 2024-02-20:
Hey @Koenkk, thanks for the reply!
Unfortunately it doesn't fix the issue:
[image]

This time also i did not enable the service, just tryed to start it as a service and theres not a single reply, thus it runs into timeout.

FlxSteiner, 2024-02-21:
I had the same issue when updating from an older z2m version where my node version was below 18. I updated node but forgot to run "npm ci". I assume the package 'sd-notify' was missing causing the service to never get the ready message and restarting after the default of 90s.

staleand, 2024-02-21:
Thank you @FlxSteiner! 

I just ran "npm ci" and it's been running for 5+ minutes now without restart. 
Seems like you just solved an headache for me 🥇  

Cheers 👍

its-Zeke, 2024-02-21:
Good call @FlxSteiner, the package was missing in my directory (checked with 'npm ls'), somehow tagged as unmet optional dependancy ?
[image]

I installed the package and now the service receives the start response and remains running!

Thank you.

FlxSteiner, 2024-02-21:
This raises the question if a migration guide of sorts would be needed or a more detailed description of the update process (at https://www.zigbee2mqtt.io/guide/installation/01_linux.html#for-later-update-zigbee2mqtt-to-the-latest-version). 

As well as probably a mention of the sd-notify package requirement to use the service (at https://www.zigbee2mqtt.io/guide/installation/01_linux.html#optional-running-as-a-daemon-with-systemctl). @Koenkk

Koenkk, 2024-02-22:
@chrthi can you take a look at this?

neuhausj, 2024-02-27:
Just quick feedback: I also ran into the same issue and installing sd-notify + libsystemd-dev + reboot + npm ci fixed it.
Thanks for the pointers ;)

lduesing, 2024-02-29:
On a minimal Debian install there is no make and no g++ installed, thus the install of optional sd-notify fails silently. Please add both to the libsystemd-dev dependency.

Koenkk, 2024-02-29:
@lduesing what do you mean with "both"? is `sd-notify` an apt package, getting this on ubuntu:

```
ubuntu@j3455:~$ sudo apt-get install sd-notify
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package sd-notify
```

lduesing, 2024-02-29:
```
 /opt/zigbee2mqtt  master *1
❯ sudo apt install make g++ libsystemd-dev
❯ npm ci
```
sd-notify is an node package, which is installed by package.json:
```
  "optionalDependencies": {
    "sd-notify": "^2.8.0"
  }
```
I meant to add "make" and "g++" to the lower part of the debian install dokument. (just look for libsystemd-dev, it is there twice, first with "make" and "g++" and the second without.

anatoliyrnd, 2024-03-02:
> sudo apt-get install -y  libsystemd-dev
+npm ci  
fixed it.
Thanks  ;)

Koenkk, 2024-03-02:
Updated! Assuming this can be closed now.

closed 2024-03-02 by Koenkk

NiPu87, 2024-03-10:
I have zigbee2mqtt running on a Raspberry Pi 3B and encountered the same issue.

Unfortunately, installing libsystemd-dev ("libsystemd-dev is already the newest version (247.3-7+deb11u4)"), installing sd-notify (2.8.0), rebooting and running `npm ci` did not fix the issue.

Is there anything else I should do or check?

Please find below the result of `npm ls`:
zigbee2mqtt@1.36.0 /opt/zigbee2mqtt
├── @babel/core@7.23.9
├── @babel/plugin-proposal-decorators@7.23.9
├── @babel/preset-env@7.23.9
├── @babel/preset-typescript@7.23.3
├── @types/debounce@1.2.4
├── @types/finalhandler@1.2.3
├── @types/humanize-duration@3.27.4
├── @types/jest@29.5.12
├── @types/js-yaml@4.0.9
├── @types/node@20.11.20
├── @types/object-assign-deep@0.4.3
├── @types/readable-stream@4.0.10
├── @types/ws@8.5.10
├── @typescript-eslint/eslint-plugin@7.0.2
├── @typescript-eslint/parser@7.0.2
├── ajv@8.12.0
├── babel-jest@29.7.0
├── bind-decorator@1.0.11
├── connect-gzip-static@3.0.1
├── debounce@2.0.0
├── deep-object-diff@1.1.9
├── eslint-config-google@0.14.0
├── eslint-plugin-jest@27.9.0
├── eslint@8.57.0
├── fast-deep-equal@3.1.3
├── finalhandler@1.2.0
├── git-last-commit@1.0.1
├── humanize-duration@3.31.0
├── jest@29.7.0
├── js-yaml@4.1.0
├── json-stable-stringify-without-jsonify@1.0.1
├── jszip@3.10.1
├── mkdir-recursive@0.4.0
├── moment@2.30.1
├── mqtt@5.3.5
├── object-assign-deep@0.4.0
├── rimraf@5.0.5
├── sd-notify@2.8.0
├── semver@7.6.0
├── source-map-support@0.5.21
├── tmp@0.2.1
├── typescript@5.3.3
├── uri-js@4.4.1
├── winston-syslog@2.7.0
├── winston-transport@4.7.0
├── winston@3.11.0
├── ws@8.16.0
├── zigbee-herdsman-converters@18.42.0 overridden
├── zigbee-herdsman@0.35.1 overridden
└── zigbee2mqtt-frontend@0.6.159

If further information on the config would be helpful, please let me know.

Any help is highly appreciated.

w4tson, 2024-03-11:
@NiPu87 I had same issue with Raspberry Pi 3B. Tried a bunch of things described on this Issue to no avail. Then I finally started putting logging statements in the code itself in the `lib/controller.ts` file. I was interested to know whether sdNotify was getting instantiated. It was. Nothing seemed to work and after about 10+ attempts It did. I'm baffled. 

At one point I erroneously manually added the NOTIFY_SOCKET env var in my systemd config. I think the last thing I did before it worked was remove this however it makes no sense because I didn't have it in there previously and all I added were a bunch of dumb logging statements in `controller.ts`

Hope any of that helps

```
 apt list -a libsystemd-dev
Listing... Done
libsystemd-dev/stable,now 252.17-1~deb12u1+rpi1 armhf [installed]
```

```
npm ls 
zigbee2mqtt@1.36.0 /opt/zigbee2mqtt
├── @babel/core@7.23.9
├── @babel/plugin-proposal-decorators@7.23.9
├── @babel/preset-env@7.23.9
├── @babel/preset-typescript@7.23.3
├── @types/debounce@1.2.4
├── @types/finalhandler@1.2.3
├── @types/humanize-duration@3.27.4
├── @types/jest@29.5.12
├── @types/js-yaml@4.0.9
├── @types/node@20.11.20
├── @types/object-assign-deep@0.4.3
├── @types/readable-stream@4.0.10
├── @types/ws@8.5.10
├── @typescript-eslint/eslint-plugin@7.0.2
├── @typescript-eslint/parser@7.0.2
├── ajv@8.12.0
├── babel-jest@29.7.0
├── bind-decorator@1.0.11
├── connect-gzip-static@3.0.1
├── debounce@2.0.0
├── deep-object-diff@1.1.9
├── eslint-config-google@0.14.0
├── eslint-plugin-jest@27.9.0
├── eslint@8.57.0
├── fast-deep-equal@3.1.3
├── finalhandler@1.2.0
├── git-last-commit@1.0.1
├── humanize-duration@3.31.0
├── jest@29.7.0
├── js-yaml@4.1.0
├── json-stable-stringify-without-jsonify@1.0.1
├── jszip@3.10.1
├── mkdir-recursive@0.4.0
├── moment@2.30.1
├── mqtt@5.3.5
├── object-assign-deep@0.4.0
├── rimraf@5.0.5
├── sd-notify@2.8.0
├── semver@7.6.0
├── source-map-support@0.5.21
├── tmp@0.2.1
├── typescript@5.3.3
├── uri-js@4.4.1
├── winston-syslog@2.7.0
├── winston-transport@4.7.0
├── winston@3.11.0
├── ws@8.16.0
├── zigbee-herdsman-converters@18.42.0 overridden
├── zigbee-herdsman@0.35.1 overridden
└── zigbee2mqtt-frontend@0.6.159
 
```

```
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment=NODE_ENV=production
Type=notify
ExecStart=/usr/bin/node index.js
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
# Or use StandardOutput=null if you don't want Zigbee2MQTT messages filling syslog, for more options see systemd.exec(5)
StandardError=inherit
WatchdogSec=10s
Restart=always
RestartSec=10s
User=foo

[Install]
WantedBy=multi-user.target
```

```
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

NiPu87, 2024-03-12:
Thanks, @w4tson! I will go through this and try to figure out, how to fix it.

ChoBoMon, 2024-03-25:
Did you fix the problem @NiPu87? I have exactly the same problem

NiPu87, 2024-04-04:
> @NiPu87 @ChoBoMon any progress on this? I have the same issue :-(

Sorry, no progress. Tried it on different versions of Raspberry Pi OS (11, 12), with different versions of node (18, 19, 20), but it didn't work.

Next thing I'll do is to check it on a Raspi 4 that I have available and see if that works.

ChoBoMon, 2024-04-07:
I removed these two lines below from the zigbee2mqtt.service systemctl config and now its working.  
```
Type=notify
WatchdogSec=10s
```
Not sure what i am missing out because i removed those lines, but atleast it doesn't boot every 2min.

jsponz, 2024-04-12:
@ChoBoMon any effect after making the changes?

shudashov, 2024-04-16:
Exactly the same problem... Ubuntu 22.04 fresh install. aply all recomendations - no effect.

chrthi, 2024-04-16:
Strange. I don't see anything wrong here, except that if you still have `ExecStart=/usr/bin/npm start` in your service file, you should put `ExecStart=/usr/bin/node index.js` or `NotifyAccess=all`.
npm start creates some child processes and systemd only accepts notifications from the service process itself by default.

shudashov, 2024-04-17:
@chrthi ,  thanks for the answer. my service file is:

`[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment=NODE_ENV=production
Type=notify
ExecStart=/snap/bin/node index.js
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
# Or use StandardOutput=null if you don't want Zigbee2MQTT messages filling sys>
StandardError=inherit
WatchdogSec=10s
Restart=always
RestartSec=10s
User=z2m

[Install]
WantedBy=multi-user.target`

what do you mean with `or NotifyAccess=all` ?
Thanks a lot

mjeveritt, 2024-10-25:
I had this problem too, but prompted by comments above, the solution is to make sure that the libsystemd-dev package is installed such that `sd-notify` is pulled into the node package dependencies.

Then, I did `npm rebuild` and `npm run build` before restarting zigbee2mqtt, and everything has been fine since.

It is not _necessary_ to remove the `Type=` and `Watchdog=` lines if you do this.

Make sure that if you're Not using a pi, that you amend the `User=` line to match an appropriate system user in the service file though!

tolgacanatan, 2024-11-20:
Same issue here. I changed the `StandartOutput` to `null` **and it's working well now**.  :tada:   Here is my latest .service file:

```
(zigbee2mqtt) pi@raspberrypi:/opt/zigbee2mqtt $ cat /etc/systemd/system/zigbee2mqtt.service
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Type=notify
Environment=NODE_PATH=/opt/zigbee2mqtt/lib/node_modules
Environment=NPM_CONFIG_PREFIX=/opt/zigbee2mqtt
Environment=npm_config_prefix=/opt/zigbee2mqtt
Environment=NODE_ENV=production
ExecStart=/opt/zigbee2mqtt/bin/node index.js
WorkingDirectory=/opt/zigbee2mqtt
#StandardOutput=inherit
StandardOutput=null
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

service status (It was stucking at 'activating' before. It's 'active' now.

```
(zigbee2mqtt) pi@raspberrypi:/opt/zigbee2mqtt $ sudo service zigbee2mqtt status
● zigbee2mqtt.service - zigbee2mqtt
     Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; preset: enabled)
     Active: active (running) since Wed 2024-11-20 14:40:44 +03; 4min 26s ago
   Main PID: 4227 (node)
      Tasks: 11 (limit: 9259)
        CPU: 3.492s
     CGroup: /system.slice/zigbee2mqtt.service
             └─4227 /opt/zigbee2mqtt/bin/node index.js

Nov 20 14:40:40 raspberrypi systemd[1]: Starting zigbee2mqtt.service - zigbee2mqtt...
Nov 20 14:40:44 raspberrypi systemd[1]: Started zigbee2mqtt.service - zigbee2mqtt.
```

I also installed `libsystemd-dev` system package while trying to solve the issue but it didn't work without changing `StandardOutput` to `null`. I'm not sure `libsystemd-dev` is required to solve the issue.

```
(zigbee2mqtt) pi@raspberrypi:/opt/zigbee2mqtt $  apt list -a libsystemd-dev
Listing... Done
libsystemd-dev/stable,now 252.31-1~deb12u1 arm64 [installed]

libsystemd-dev/stable 252.31-1~deb12u1 armhf
```

```
(zigbee2mqtt) pi@raspberrypi:/opt/zigbee2mqtt $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

Goodbye "became unavailable" :)

Thank you for the discussions and suggestions above! :metal:

hmeine, 2025-01-12:
I ran into the same problem, also on an older Raspberry Pi (running raspbian / buster). I upgraded zigbee2mqtt via "git pull", eventually removing the node_modules and dist folders, and following the official instructions on using pnpm install + build.  Repeating this (after also installing libsystemd-dev), I noticed the following two lines in the `pnpm i` output:
```
node_modules/.pnpm/unix-dgram@2.0.6/node_modules/unix-dgram: Running install script, failed in 6.5s (skipped as optional)
node_modules/.pnpm/sd-notify@2.8.0/node_modules/sd-notify: Running install script, failed in 6.5s (skipped as optional)
```
(I also tried changing StandardOutput, without expecting it to fix my problem, and it did not.)
I am not very familiar with systemd(-notify) and service configs, but I wonder if it has to do with the older system environment on older raspis? I also noticed this stdout line from z2m:
```
Starting Zigbee2MQTT without watchdog.
```
Does that mean that I should follow @ChoBoMon's service file modifications? After all, z2m does not seem to restart itself, but systemd seems to cause it.

bourboif, 2025-01-29:
I also have the same problem, unable to install sd-notify always unmet dependency ... (npm v20; rpi3 debian 11). I'm going to deactivate notification in zigbee2mqtt service file

tbnobody, 2025-02-03:
I fixed it on my side.
When executing the command `pnpm i --frozen-lockfile`
I got the following output at the end:
```
Ignored build scripts: @serialport/bindings-cpp, esbuild, sd-notify, unix-dgram. Run "pnpm approve-builds" to pick which dependencies should be allowed to run scripts.
```

I executed `pnpm approve-builds`, selected all items and built it........ after a `systemctl restart zigbee2mqtt` it just worked

hmeine, 2025-02-08:
> I ran into the same problem, also on an older Raspberry Pi (running raspbian / buster). I upgraded zigbee2mqtt via "git pull", eventually removing the node_modules and dist folders, and following [...]

I believe I found out what caused these failures, and it was indeed the older system environment. Raspbian buster has Python 3.7.3, which apparently is not supported anymore by pnpm. The failing install scripts are using `python3` and the `:=` syntax (if you ask me, I think it's a bad idea to rely on recent python features as part of build scripts – it would not be not too difficult to be more compatible IMHO).

hmeine, 2025-02-14:
After my two previous posts, I upgraded Raspbian from buster to the latest distro (bookworm), which indeed fixed the problem that `sd-notify` would not install correctly because some installation script(s?) require a too recent Python version.

However, systemd *still does not get notified properly* and keeps restarting z2m due to a "timeout".

With "fixed the problem" I meant that the result of `pnpm i --frozen-lockfile` looks perfect, all scripts ran successfully, sd-notify is listed among other successfully installed devDependencies, and the output of `npm ls` looks perfect, too. I then set the log_level to "debug" and could even see `z2m: sd-notify loaded` in the logs.  From looking at the code, setting `Z2M_WATCHDOG` affects the sd-notify usage, but does not seem to be a prerequisite for the `this.sdNotify?.ready();` call which I assume should be the one telling systemd that z2m is running.

I am using the unmodified service file from https://www.zigbee2mqtt.io/guide/installation/01_linux.html#optional-running-as-a-daemon-with-systemctl which directly starts /usr/bin/node and not npm.  I tried `NotifyAccess=all` as well, but it was no different from `main` (which is implied with `Type=notify`).

What else could be the problem?  (I am running z2m as "pi" user, but that seems to be a usual setup.)  I had to install the system from scratch and this is now a very clean installation following the official instructions, with no other software installed or running on the system.

To be honest – I have spent an unreasonable amount of time on debugging this from a home automation perspective (for which `Type=exec` would absolutely suffice in my case). ;-) This is just my hacker genes trying to understand everything and not accepting defeat.