Koenkk/zigbee2mqtt #21467: [New device support]: LYWSD03MMC-z

TheGroundZero, 2024-02-19:
### Link

https://www.zigbee2mqtt.io/devices/LYWSD03MMC.html

### Database entry

{   "id": 25,   "type": "EndDevice",   "ieeeAddr": "0xa4c138b32f2fcff4",   "nwkAddr": 47596,   "manufId": 4417,   "manufName": "Xiaomi",   "powerSource": "Battery",   "modelId": "LYWSD03MMC-z",   "epList": [     1   ],   "endpoints": {     "1": {       "profId": 260,       "epId": 1,       "devId": 770,       "inClusterList": [         0,         1,         3,         32,         1026,         1029       ],       "outClusterList": [         25,         516       ],       "clusters": {         "genBasic": {           "attributes": {             "modelId": "LYWSD03MMC-z",             "manufacturerName": "Xiaomi",             "powerSource": 3,             "zclVersion": 3,             "appVersion": 0,             "stackVersion": 2,             "hwVersion": 20,             "dateCode": "20240211",             "swBuildId": "3001-0120"           }         },         "genPollCtrl": {           "attributes": {             "checkinInterval": 14400           }         },         "genPowerCfg": {           "attributes": {             "batteryPercentageRemaining": 200,             "batteryVoltage": 31           }         },         "msTemperatureMeasurement": {           "attributes": {             "measuredValue": 2361           }         },         "msRelativeHumidity": {           "attributes": {             "measuredValue": 5412           }         }       },       "binds": [         {           "cluster": 32,           "type": "endpoint",           "deviceIeeeAddress": "0x00124b0024c2cfef",           "endpointID": 1         },         {           "cluster": 1,           "type": "endpoint",           "deviceIeeeAddress": "0x00124b0024c2cfef",           "endpointID": 1         },         {           "cluster": 1026,           "type": "endpoint",           "deviceIeeeAddress": "0x00124b0024c2cfef",           "endpointID": 1         },         {           "cluster": 1029,           "type": "endpoint",           "deviceIeeeAddress": "0x00124b0024c2cfef",           "endpointID": 1         }       ],       "configuredReportings": [         {           "cluster": 1,           "attrId": 33,           "minRepIntval": 3600,           "maxRepIntval": 65000,           "repChange": 10,           "manufacturerCode": null         },         {           "cluster": 1026,           "attrId": 0,           "minRepIntval": 10,           "maxRepIntval": 3600,           "repChange": 100,           "manufacturerCode": null         },         {           "cluster": 1029,           "attrId": 0,           "minRepIntval": 10,           "maxRepIntval": 3600,           "repChange": 100,           "manufacturerCode": null         }       ],       "meta": {}     }   }

### Comments

I have flashed two LYWSD03MMC in the past without issue. But those have show a ZigbeeModel of "LYWSD03MMC" while this one shows "LYWSD03MMC-z"

Flashed with @pvvx's flasher (https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)
IIRC the firmware version was `Z03MMC_0120.bin`.

Previous flashes were with `0119` but have since been upgrade via Z2M.
This was also the first flash from a device out-of-the-box with the Mi firmware, while the previous devices were flashed with PVVX's BLE firmware prior to flashing to the Zigbee firmware.

[image]
[image]

### External definition

```shell
# Generated by DevTools, unaltered
const {batteryPercentage, temperature, humidity} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['LYWSD03MMC-z'],
    model: 'LYWSD03MMC-z',
    vendor: 'Xiaomi',
    description: 'Automatically generated definition',
    extend: [batteryPercentage(), temperature({"endpoints":["1"]}), humidity({"endpoints":["1"]})],
    meta: {},
};

module.exports = definition;
```

pvvx, 2024-02-19:
https://github.com/pvvx/ZigbeeTLc/issues/58#issuecomment-1944377995
https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/lywsd03mmc-z.js

TheGroundZero, 2024-02-19:
So I downloaded [lywsd03mmc-z.js](https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/lywsd03mmc-z.js) to My Z2M config folder, then added it as an External Converter, and removed and re-joined my device.
That seems to have done it for that sensor.

It does appear my 2 other, previously functional, sensors are not throwing errors.
I'll remove those and re-join them as well and see if that fixes things.

BartCardan, 2024-02-21:
Just because I am curious: what is changed in the latest release that we need to add a external converter?

TheGroundZero, 2024-02-22:
> Just because I am curious: what is changed in the latest release that we need to add a external converter?

From what I understood, Z2M picks the device config based on the Zigbee Model.
The config for the "LYWSD03MMC" Zigbee Model maps to the config for the devbis firmware. The config for the ZigbeeTLc firmware is a bit different, so you'd run into issues with missing or nonfunctional sensors etc.
So to differentiate devices flashed with ZigbeeTLc, they now identify as "LYWSD03MMC-z", but no config has been mapped to this Device Model in Z2M yet.
I suppose the External Converter is a temporary fix until the device is officially supported by Z2M.

Milinator, 2024-02-24:
> I added the device with lywsd03mmc-z.js and it is basically working, but it's reporting only (exactly) every hour. That's why I want to revert back to BLE and then reflash the zigbee firmware with a given interval. Problem is: I can't go back to BLE, I'm following this blog post: https://smarthomescene.com/guides/how-to-setup-zigbee2mqtt-local-ota-index/
> 
> Have done everything as described, but there is no update available in Z2M. Maybe that's also caused by the "z" at the end? Any help'd be appreciated, I don't really want to buy a UART dongle...

Fixed with this update file
https://github.com/devbis/z03mmc/issues/118#issuecomment-1942371299

dasEtwasmitPunkten, 2024-03-02:
@Koenkk I saw this device wasn't included in the last release. Is there a plan to integrate it in one of the next versions?
thanks in advance

Koenkk, 2024-03-02:
@smarts4home since you created https://github.com/Koenkk/zigbee-herdsman-converters/pull/7070 before, do you still want to add this device?

mrmaximas, 2024-03-03:
@pvvx not possible to set some parameters via z2m web ui, like Comfort temperature min/max etc. 
how to do it?
Publish 'set' 'comfort_temperature_max' to 'Test Sensor' failed: 'Error: Write 0xa4c1383bc0835aec/1 hvacUserInterfaceCfg({"259":{"value":2300,"type":41}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')'

kafisc1, 2024-03-10:
This external converter (and also without any external convert) stopped working for my LYWSD03MMC-z since upgrading z2m with the following error:
Exception while calling fromZigbee converter: Definition 'LYWSD03MMC-z' has not endpoint defined}

Is anyone facing the same issue?

txitxo0, 2024-03-11:
> This external converter (and also without any external convert) stopped working for my LYWSD03MMC-z since upgrading z2m with the following error: Exception while calling fromZigbee converter: Definition 'LYWSD03MMC-z' has not endpoint defined}
> 
> Is anyone facing the same issue?

Yep, here experimenting the same behaviour

kafisc1, 2024-03-11:
> Yep, here experimenting the same behaviour

Switched from dev to stable branch for now.
The error does not occur here.

txitxo0, 2024-03-11:
I am using the zigbee2mqtt home asistant addon and, as far as I know it is using the stable branch as well

txitxo0, 2024-03-12:
> Switched from dev to stable branch for now. The error does not occur here.

I am using the zigbee2mqtt home asistant addon and, as far as I know it is using the stable branch as well. How did you do it?

kafisc1, 2024-03-13:
1. I added the external converter from the buttom
2. I removed the HASS-Addon (make copy of addon config from the section socat ) "Zigbee2MQTT-dev" and installed Zigbee2MQTT (enter prior saved config into socat in the addon settings)
3. I rebooted the whole machine and it was working again

External converter:

`// Autor: Markus Bodengriller https://github.com/Bodengriller
// requires zigbee2mqtt v1.34+
// external converter for Xiaomi LYWSD03MMC https://github.com/pvvx/ZigbeeTLc
// based on external converter for devbis-Firmware
// https://raw.githubusercontent.com/devbis/z03mmc/master/converters/lywsd03mmc.js

const {
    batteryPercentage,
    temperature,
    humidity,
    enumLookup,
    binary,
    numeric,
    quirkAddEndpointCluster,
} = require('zigbee-herdsman-converters/lib/modernExtend');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');

const dataType = {
    boolean: 0x10,
    uint8: 0x20,
    uint16: 0x21,
    int8: 0x28,
    int16: 0x29,
    enum8: 0x30,
};

const definition = {
    zigbeeModel: ['LYWSD03MMC-z'],
    model: 'LYWSD03MMC',
    vendor: 'Xiaomi',
    description: 'Temperature & humidity sensor with custom firmware',
    extend: [
        quirkAddEndpointCluster({
            endpointID: 1,
            outputClusters: [],
            inputClusters: [
                'genPowerCfg',
                'msTemperatureMeasurement',
                'msRelativeHumidity',
                'hvacUserInterfaceCfg',
            ],
        }),
        batteryPercentage(),
        temperature({reporting: {min: 10, max: 300, change: 10}}),
        humidity({reporting: {min: 10, max: 300, change: 50}}),
        enumLookup({
            name: 'temperature_display_mode',
            lookup: {'celsius': 0, 'fahrenheit': 1},
            cluster: 'hvacUserInterfaceCfg',
            attribute: {ID: 0x0000, type: dataType.enum8},
            description: 'The units of the temperature displayed on the device screen.',
        }),
        binary({
            name: 'show_smiley',
            valueOn: ['SHOW', 0],
            valueOff: ['HIDE', 1],
            cluster: 'hvacUserInterfaceCfg',
            attribute: {ID: 0x0002, type: dataType.enum8},
            description: 'Whether to show a smiley on the device screen.',
        }),
        numeric({
            name: 'temperature_calibration',
            unit: 'ºC',
            cluster: 'hvacUserInterfaceCfg',
            attribute: {ID: 0x0100, type: dataType.int8},
            valueMin: -12.7,
            valueMax: 12.7,
            valueStep: 0.1,
            scale: 10,
            description: 'The temperature calibration, in 0.1° steps.',
        }),
        numeric({
            name: 'humidity_calibration',
            unit: '%',
            cluster: 'hvacUserInterfaceCfg',
            attribute: {ID: 0x0101, type: dataType.int8},
            valueMin: -12.7,
            valueMax: 12.7,
            valueStep: 0.1,
            scale: 10,
            description: 'The humidity offset is set in 0.1 % steps.',
        }),
        numeric({
            name: 'comfort_temperature_min',
            unit: 'ºC',
            cluster: 'hvacUserInterfaceCfg',
            attribute: {ID: 0x0102, type: dataType.int8},
            valueMin: -127.0,
            valueMax: 127.0,
            description: 'Comfort parameters/Temperature minimum, in 1°C steps.',
        }),
        numeric({
            name: 'comfort_temperature_max',
            unit: 'ºC',
            cluster: 'hvacUserInterfaceCfg',
            attribute: {ID: 0x0103, type: dataType.int8},
            valueMin: -127.0,
            valueMax: 127.0,
            description: 'Comfort parameters/Temperature maximum, in 1°C steps.',
        }),
        numeric({
            name: 'comfort_humidity_min',
            unit: '%',
            cluster: 'hvacUserInterfaceCfg',
            attribute: {ID: 0x0104, type: dataType.uint8},
            valueMin: 0.0,
            valueMax: 100.0,
            description: 'Comfort parameters/Humidity minimum, in 1% steps.',
        }),
        numeric({
            name: 'comfort_humidity_max',
            unit: '%',
            cluster: 'hvacUserInterfaceCfg',
            attribute: {ID: 0x0105, type: dataType.uint8},
            valueMin: 0.0,
            valueMax: 100.0,
            description: 'Comfort parameters/Humidity maximum, in 1% steps.',
        }),
    ],
    ota: ota.zigbeeOTA,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const bindClusters = ['msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg'];
        await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
        await reporting.temperature(endpoint, {min: 10, max: 300, change: 10});
        await reporting.humidity(endpoint, {min: 10, max: 300, change: 50});
        await reporting.batteryPercentageRemaining(endpoint);
        try {
            await endpoint.read('hvacThermostat', [0x0010, 0x0011, 0x0102, 0x0103, 0x0104, 0x0105]);
            await endpoint.read('msTemperatureMeasurement', [0x0010]);
            await endpoint.read('msRelativeHumidity', [0x0010]);
        } catch (e) {
            /* backward compatibility */
        }
    },
};

module.exports = definition;`

kafisc1, 2024-03-14:
> Thanks for this LYWSD03MMC-z support ! I hope it will make it to the official version.
> 
> One small issue on my side : the batterie reports 100% all the time, any clue ? should I reapply firmware?

The battery report is working for me. Seems like the devices are staying a long time on 100% and are only reporting on greater changes.
[image]

sabamimi, 2024-03-14:
Thanks @kafisc1 for your feedback

txitxo0, 2024-03-14:
I have just added this external converter from the settings menu and reboot all the host, but I am still experimenting issues. The device is not supported and the temp and humidity are not being exposed.

I am using the ZigBee regular addon in HA, not the edge version

TheGroundZero, 2024-03-14:
> I have just added this external converter from the settings menu and reboot all the host, but I am still experimenting issues. The device is not supported and the temp and humidity are not being exposed.
> 
> I am using the ZigBee regular addon in HA, not the edge version

You need to remove the device from Z2M and re-pair it.
https://sequr.be/blog/2024/02/flash-xiaomi-lywsd03mmc-ble-sensor-with-zigbee-firmware/#lywsd03mmc-z

pvvx, 2024-03-14:
> The battery report is working for me. Seems like the devices are staying a long time on 100% and are only reporting on greater changes.

The initial, ultra-short period of operation of the new battery is cut off. These are voltages greater than 3.0V.
Otherwise, in a few days you will have 80% and so 80% of the working time.

[image]

fhempy, 2024-04-01:
@koe

> @smarts4home since you created [Koenkk/zigbee-herdsman-converters#7070](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7070) before, do you still want to add this device?

Hi,
I thought that the code from this external converter (https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/lywsd03mmc-z.js) is going to be added to the April release. Any further plans for the next release, or is there something missing?

pvvx, 2024-04-01:
Standard Zigbee 3.0 does not require special support.
Converters are required solely to support non-standard functions.
As a result, if zigbee2mqtt supports Zigbee 3.0, then the thermometers should work without any converters.

TeHashX, 2024-04-03:
https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/lywsd03mmc-z.js is not working anymore after updating to Z2M to 1.36.1-1, I get Unsupported device. I tried to delete and add again, same thing. I even added a new device, still unsupported.
Downgraded to 1.36.0-1 and all good.

dasEtwasmitPunkten, 2024-04-03:
> https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/lywsd03mmc-z.js is not working anymore after updating to Z2M to 1.36.1-1, I get Unsupported device. I tried to delete and add again, same thing. I even added a new device, still unsupported. Downgraded to 1.36.0-1 and all good.
> 
> FW 3001-0120 20240226

I am experiencing the same behavior. Fortunately the temperature and humidity sensors are still working. The custom functions comfort temperature and humidity limits are gone.

TeHashX, 2024-04-03:
I updated the file and it's working now, was a small change about 20 hours ago, I'm not sure if this was the culprit https://github.com/pvvx/ZigbeeTLc/commit/6d8e14b7bd2173350448420e8f5aa4b124c34fb4

korbinio, 2024-04-05:
> @Koenkk
> 
> > @smarts4home since you created [Koenkk/zigbee-herdsman-converters#7070](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7070) before, do you still want to add this device?
> 
> Hi, I thought that the code from this external converter (https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/lywsd03mmc-z.js) is going to be added to the April release. Any further plans for the next release, or is there something missing?

@Koenkk: +1 from me for adding this to the next release :)

Koenkk, 2024-04-06:
Can someone make a pull request for this?

xylorin, 2024-04-13:
Any chance to receive a standard support for this device in future?

baloo2, 2024-04-17:
+1 to add support - thank a lot

pvvx, 2024-04-18:
Version 0.1.2.1 includes the ability to change the name of the device. :)

baloo2, 2024-04-18:
Thanks! Shoud the update be available online in Z2M or I have to update it manually (my_index.json)?

BWFCGUIXT, 2024-04-18:
Zigbee2MQTT version [1.36.1] 
commit: [ffc2ff1]
Coordinator type: EZSP v8
Coordinator revision: 6.10.3.0 build 297

Now getting the following error when trying to update:
```

Error 2024-04-18 13:13:54Failed to check if update available for 'Temperature/Humidity Sensor Living room' (The value of "offset" is out of range. It must be >= 0 and <= 98266. Received 98268)

Debug 2024-04-18 13:13:54RangeError: The value of "offset" is out of range. It must be >= 0 and <= 98266. Received 98268 at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:88:9) at Buffer.readUInt16LE (node:internal/buffer:245:5) at parseSubElement (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:149:26) at Object.parseImage (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/common.ts:192:29) at fillImageInfo (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:29:26) at /app/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:51:109 at Array.map (<anonymous>) at getIndex (/app/node_modules/zigbee-herdsman-converters/src/lib/ota/zigbeeOTA.ts:51:45) at processTicksAndRejections (node:internal/process/task_queues:95:5)
```

Probably completely unrelated but maybe someone has the solution for the following error:

`error 2024-04-18 13:18:43: Entity 'homeassistant/sensor' is unknown`

pvvx, 2024-04-18:
LYWSD03MMC-z FW ver 0.1.2.1
```
zha_toolkit_version: v1.1.10
zigpy_version: 0.63.4
zigpy_rf_version: 0.38.1
ieee_org: update.xiaomi_lywsd03mmc_z_firmware
ieee: a4:c1:38:0b:5e:ed:85:b7
command: attr_read
command_data: null
start_time: "2024-04-18T13:27:14.572671+00:00"
errors: []
params:
  endpoint_id: 1
  cluster_id: 516
  attr_id: 256
  dir: 0
  tries: 1
  expect_reply: true
  args: []
  read_before_write: true
  read_after_write: true
attr_type: "0x29"
write_is_equal: false
result_read:
  - "256": 0
  - {}
success: true
```

TeHashX, 2024-04-18:
> LYWSD03MMC-z FW ver 0.1.2.1
> 
> ZHA: LYWSD03MMC-z by Xiaomi , Firmware: 0x01213001
> 
> ```
> zha_toolkit_version: v1.1.10
> zigpy_version: 0.63.4
> zigpy_rf_version: 0.38.1
> ieee_org: update.xiaomi_lywsd03mmc_z_firmware
> ieee: a4:c1:38:0b:5e:ed:85:b7
> command: attr_read
> command_data: null
> start_time: "2024-04-18T13:27:14.572671+00:00"
> errors: []
> params:
>   endpoint_id: 1
>   cluster_id: 516
>   attr_id: 256
>   dir: 0
>   tries: 1
>   expect_reply: true
>   args: []
>   read_before_write: true
>   read_after_write: true
> attr_type: "0x29"
> write_is_equal: false
> result_read:
>   - "256": 0
>   - {}
> success: true
> ```
> 
> attr_type: "0x29" - Signed 16bit int
> 
> In future firmware versions, to eliminate dependencies from Z2M glitches, it is planned to install device settings via BLE from the Google Chrome web browser. The firmware will support both BLE and Zigbee. Switching to BLE will be temporary, for LYWSD03MMC for ~80 seconds to connect after turning the battery off and on.

This is the right OTA file?
https://github.com/pvvx/ZigbeeTLc/blob/master/bin/devbis/1141-0203-79993001-Z03MMC_v0121.zigbee

robvanoostenrijk, 2024-04-29:
The support for LYWSD03MMC-z (and other pvvx ZigbeeTLc devices) is covered in https://github.com/Koenkk/zigbee-herdsman-converters/pull/7457

xylorin, 2024-05-06:
just wanted to thank you all, today i checked my HA and the device is now correctly working!

z-master42, 2024-05-17:
Es muss im Log der Eintrag connecting bridge ... auftauchen. Eine IP-Adresse wird da nicht stehen.

closed 2024-12-30 by github-actions[bot]