Koenkk/zigbee2mqtt #21469: NodON SIN-4-FP-21 Mode

Cityloue15, 2024-02-19:
### What happened?

Hello, following the integration of the SIN-4-FP-21 module from NodON, I do not have modes such as comfort, eco etc...

### What did you expect to happen?

I was expecting to see the actions/information mode

### Zigbee2MQTT version

 1.33.2

### Adapter firmware version

26780700

### Adapter

ConBee2

### Setup

 Debian GNU/Linux 10 (buster) 64bits (aarch64)

courville, 2024-02-21:
Perhaps could be considered as duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/21071

MichaelMure, 2024-04-03:
Possibly related: https://github.com/Koenkk/zigbee2mqtt/issues/22068

closed 2024-11-01 by github-actions[bot]