Koenkk/zigbee2mqtt #21475: Aqara S1 Air Sensor (KQJCMB11LM) - PM2.5 reported 0 but value available

chrisbucher, 2024-02-19:
### What happened?

The Aqara S1 Air quaility sensor always reports PM25 as 0 under "Exposes" and HA entity, but the value is reported correctly under Dev console Endpoint 1 - cluster pm25Measurement - attribute measuredValue. Changing the calibration in "Settings (specific)" also changes the exposed / HA entity value by the calibration value:  PM25 calibtration = 5 -> exposed value also 5.

[image]

[image]

`Read result of 'pm25Measurement': {"measuredValue":7}`

### Zigbee2MQTT version

1.35.2 commit: unknown

### Adapter firmware version

20220219

### Adapter

Sonoff ZigBee 3.0 USB Dongle-E

### Setup

Add-on on HomeAssistant on RPI4

mild-delirium, 2024-02-25:
I have one of those. The device reports (and shows on its display) the correct value in ug/m3, but it looks like [z2m internally divides the value by 1000](https://github.com/Koenkk/zigbee-herdsman-converters/commit/1903b682aad1a64050f9ccbcfd507e86139e7401), converting it into mg/m3, so most of the time it's zero. Looks like an easy to fix bug.

Koenkk, 2024-02-25:
@avdept could you check this?

avdept, 2024-02-26:
> @avdept could you check this?

Sure, I'll take a look!

avdept, 2024-02-26:
Ok I did Quick Look and seems like it's not the convertor but something else. For some reason value that sends to MQTT is "stuck" with some specific number(0.0008 in my case) while on screen reading is different.
The reason I did /1000 because device reports value in Gramms while we want to have uG which is divided by 1000. I matched this value by looking at the display reading.

So @Koenkk any ideas why value sent to MQTT might stuck?

mild-delirium, 2024-02-26:
@avdept 
> The reason I did /1000 because device reports value in Gramms while we want to have uG which is divided by 1000. I matched this value by looking at the display reading.

If you needed to divide by 1000 to get micrograms, the original reported value would have to be in nanograms, not grams.

Anyway, I just read the 'measuredValue' in z2m dev console (screenshot below) and it says 14, which exactly matches the value on the device's screen.  This results in the pm25 value = measuredValue/1000 = 0.014, but I suppose later it gets converted to an integer so pm25 value is zero (moreover, sometimes when I get spikes in pm25 and the device shows on the screen values over 1000, pm25 reported by z2m becomes 1). It still looks like the culprit is the division by 1000 .

[image]

Koenkk, 2024-02-26:
@mild-delirium what is the unit on the screen? `µg/m³`?

mild-delirium, 2024-02-27:
@Koenkk yes on the screen it says `µg/m³`. Usually it's around 10 µg/m³ here. With some creative hand waving I can force it to show much higher values (have to do that in order for the pm25 value to jump into the middle of the screen because it's "too high" so that it displays the unit of measurement, otherwise it's just a number at the top of the screen) and pm25Measurement:measuredValue in the log always matched the value on the screen.

A couple of snippets from the log grepped for pm25.

**Normal value, 0012 on the screen:**
_Zigbee2MQTT:debug 2024-02-27 05:17:22: Received Zigbee message from 'air_quality_living_room', type 'attributeReport', cluster '**pm25Measurement**', data '{**"measuredValue":12**}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-02-27 05:17:22: MQTT publish: topic 'zigbee2mqtt/air_quality_living_room', payload '{"co2":539,"humidity":40.95,"last_seen":"2024-02-27T05:17:22+03:00","linkquality":171,**"pm25":0**,"temperature":24.83,"update":{"installed_version":26,"latest_version":26,"state":"idle"},"update_available":null}'_

**"Forced" higher value, never occur naturally, the screen shows 4513 µg/m³:**
_Zigbee2MQTT:debug 2024-02-27 05:47:06: Received Zigbee message from 'air_quality_living_room', type 'attributeReport', cluster '**pm25Measurement**', data '**{"measuredValue":4513}'** from endpoint 1 with groupID 0
Zigbee2MQTT:info  2024-02-27 05:47:06: MQTT publish: topic 'zigbee2mqtt/air_quality_living_room', payload '{"co2":595,"humidity":42.17,"last_seen":"2024-02-27T05:47:06+03:00","linkquality":160,**"pm25":5**,"temperature":24.72,"update":{"installed_version":26,"latest_version":26,"state":"idle"},"update_available":null}'_

avdept, 2024-02-27:
Ok, I think that division by 1000 is a problem. If I remember correctly I could've been confused by prefix 000 on Lcd and thought those were fraction parts. I'll remove that division by 1000 and it should be good

mild-delirium, 2024-02-27:
Heh, I recall that those zeroes in front of the value confused me as well.

Thank you!

Koenkk, 2024-02-27:
Pushed the fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-02-27 by Koenkk