Koenkk/zigbee2mqtt #21476: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}' resukt in restart of service

borjeM, 2024-02-19:
### What happened?

Approx every 90s zigbee2mqtt reports in log
 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
And service is restarted

### What did you expect to happen?

Servie should run without restart

### How to reproduce it (minimal and precise)

info  2024-02-19 19:10:41: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2024-02-19 19:10:41: Disconnecting from MQTT server
info  2024-02-19 19:10:41: Stopping zigbee-herdsman...
info  2024-02-19 19:10:41: Stopped zigbee-herdsman
info  2024-02-19 19:10:41: Stopped Zigbee2MQTT

Next log file show
info  2024-02-19 19:10:55: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2024-02-19.19-10-55' filename: log.txt
info  2024-02-19 19:10:55: Starting Zigbee2MQTT version 1.35.3 (commit #fe0742a6)
info  2024-02-19 19:10:55: Starting zigbee-herdsman (0.33.8)
info  2024-02-19 19:11:01: zigbee-herdsman started (resumed)
info  2024-02-19 19:11:01: Coordinator firmware version: '{"meta":{"maintrel":"3 ","majorrel":"6","minorrel":"10","product":8,"revision":"6.10.3.0 build 297"},"type":"EZSP v8"}'
info  2024-02-19 19:11:01: Currently 45 devices are joined:

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff

### Setup

zigbee2mqtt running in rpi 4 with sd card

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14335737/log.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14335740/log.txt)

var42, 2024-02-20:
Same for me 
zigbee2mqtt + mosquitto on rpi 4 with sd card

var42, 2024-02-21:
#21463 https://github.com/Koenkk/zigbee2mqtt/issues/21463#issuecomment-1956434161

closed 2024-09-19 by github-actions[bot]