Koenkk/zigbee2mqtt #21477: Expected switch device class (in Home Assistant)

Thecrafterja, 2024-02-19:
### What happened?

Probably since the update to home assistant 2024.2.2, window open of my Bosch thermostat is not available anymore. Developers at HA think it is related to zigbee2mqtt.

### How to reproduce it (minimal and precise)

Use a Bosch thermostat II, window open option won‘t be available in HA

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

20210708

### Adapter

Sonoff Zigbee Dongle-P

### Setup

Add-on in home assistant os,

SteffenKr, 2024-02-26:
This issue looks like a duplicate of this: #21208
According to a [comment](https://github.com/Koenkk/zigbee2mqtt/issues/21208#issuecomment-1937911133) it should be fixed in the next release (1.36)

NicoDreamzZ, 2024-03-02:
It is fixed with the release 1.36 :)

Thecrafterja, 2024-03-02:
Thanks for your responses. I updated my zigbee2mqtt, and it is working again :)

closed 2024-03-02 by Thecrafterja