Koenkk/zigbee2mqtt #21495: OSRAM AC02517 Smart+ Switch Mini is always marked unavailable in HA even if online in Z2M

silberjan, 2024-02-20:
### What happened?

I have one OSRAM AC02517 that works perfectly in my setup for some time now. Recent i wanted to add a new one and everything seemed to work fine. It could connect to z2m, it was properly interviewed and bound. In Z2M it shows almost exactly the same as the old switch. Hoever in HA the new switch shows as unavailable and cannot be used to create automations. When listening to the MQTT broker, i can see that actions are still dispached though. When i turn off the availability feature in Z2M completely, the switch works in HA. When i turn it on again the switch shows as unavailable again in HA. I would like to use the availability feature in general though so turning it off is not really a solution.
I investigated a little further and i could make 2 observations:
1. The new remote as a slightly different serial number (old: AC02517 | x2619 new:  AC02517 | x1719)
2. When listening to the mqtt messages there is one small difference: the old one provides `"battery_low":false` in the payload of the message while the new one does not. (both provide the `"battery"` property)

So my theory is that HA marks the switch as unavailable because the `battery_low` property is missing. Please correct me if this is not correct.

I would love to do a PR to fix this but im a little lost where too look exactly in the codebase. 

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

0x26690700

### Adapter

ConBee2/RaspBee2

### Setup

Add-on Home assistant on raspi 3

### Debug log

Old switch (AC02517 | x2619) that works and shows as available:  
```
Debug 2024-02-19 21:55:35Received Zigbee message from 'ledvance-switch-1', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":10}' from endpoint 3 with groupID null
Info 2024-02-19 21:55:35MQTT publish: topic 'zigbee2mqtt/ledvance-switch-1', payload '{"action":"circle_click","battery":80,"battery_low":false,"linkquality":255,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2900}'
Info 2024-02-19 21:55:35MQTT publish: topic 'zigbee2mqtt/ledvance-switch-1', payload '{"action":"","battery":80,"battery_low":false,"linkquality":255,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":2900}'
Info 2024-02-19 21:55:35MQTT publish: topic 'zigbee2mqtt/ledvance-switch-1/action', payload 'circle_click'
Debug 2024-02-19 21:55:36Received Zigbee message from 'ledvance-switch-1', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":370,"transtime":10}' from endpoint 3 with groupID null
```
New switch (AC02517 | x1719) that only works when availability is off:  
```
Debug 2024-02-19 21:55:54Received Zigbee message from 'ledvance-switch-2', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":10}' from endpoint 3 with groupID null
Info 2024-02-19 21:55:54MQTT publish: topic 'zigbee2mqtt/ledvance-switch-2', payload '{"action":"circle_click","battery":100,"linkquality":207,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3100}'
Info 2024-02-19 21:55:54MQTT publish: topic 'zigbee2mqtt/ledvance-switch-2', payload '{"action":"","battery":100,"linkquality":207,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":3100}'
Info 2024-02-19 21:55:54MQTT publish: topic 'zigbee2mqtt/ledvance-switch-2/action', payload 'circle_click'
Debug 2024-02-19 21:55:55Received Zigbee message from 'ledvance-switch-2', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":370,"transtime":10}' from endpoint 3 with groupID null
```

silberjan, 2024-02-23:
okay so i had it again with different remotes and the issue is now narrowed: 

it works when i first restart HA, wait until everything is up and only then restart Z2M (not from HA but from the Z2M GUI). My theory is now that HA somehow misses the availability message somehow. If anyone has an idea how to configure everything so the restart procedure is not required pls let me know.

OlafKocanda, 2024-03-07:
This here actually saved my problems with unavailability of devices. Just disabling "availability" in settings and now all devices work again for >9 hours.

closed 2024-10-04 by github-actions[bot]