Koenkk/zigbee2mqtt #21499: [New device support]: Tuya Blind Motor TS0601 _TZE200_libht6ua

bitdivision, 2024-02-20:
### Link

https://www.aliexpress.com/item/1005006468996253.html

### Database entry

`{"id":26,"type":"EndDevice","ieeeAddr":"0x90395efffe2d1465","nwkAddr":23737,"manufId":4098,"manufName":"_TZE200_libht6ua","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�mg-g","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_libht6ua","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1708438615204}`

### Comments

I found a similar device: https://github.com/Koenkk/zigbee2mqtt/issues/17436

For pairing I didn't find a canonical way to do it, but holding one of the direction buttons until the LED starts flashing slowly and then pressing the power button 5 times seemed to do it.

I copied the external definition from [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/17436#issuecomment-1600901453). This worked apart from set_upper_limit, where the buttons were reversed. I have reversed these in the attached external definition file. Then setting limits is a case of manually moving the blind the bottom, attaching the motor and clicking start (as described in the referenced comment).

I don't have a TuYa gateway, so haven't checked what the manufacturer functionality is. I also don't know whether opening mode is necessary or supported on this device.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_libht6ua' },],
    model: 'TS0601',
    vendor: 'TuYa',
    description: 'Roller Blinds Motor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    exposes: [
        exposes.text('work_state', ea.STATE),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.battery(),
        exposes.enum('opening_mode', ea.STATE_SET, ['tilt', 'lift']).withDescription('Opening mode'),
        exposes.enum('motor_direction', ea.STATE_SET, ['left', 'right']).withDescription('Motor side'),
        exposes.enum('set_upper_limit', ea.STATE_SET, ['start', 'stop']).withDescription('Learning'),
        exposes.enum('factory_reset', ea.STATE_SET, ['SET']).withDescription('Remove limits'),
    ],
    whiteLabel: [
        tuya.whitelabel('Tuya', 'TS0601', 'Cover', ['_TZE200_libht6ua']),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.raw],
            [4, 'opening_mode', tuya.valueConverterBasic.lookup({'tilt': tuya.enum(0), 'lift': tuya.enum(1)})],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'motor_direction', tuya.valueConverterBasic.lookup({'left': tuya.enum(0), 'right': tuya.enum(1)})],
            [102, 'set_upper_limit', tuya.valueConverterBasic.lookup({'start': tuya.enum(0), 'stop': tuya.enum(1)})],
            [107, 'factory_reset', tuya.valueConverter.setLimit],
        ],
    },
};

module.exports = definition;
```
```

bitdivision, 2024-02-24:
There are a few issues here. work_state appears to be incorrectly mapped, and the position is reversed (0 is open and 100 is closed). Using invert_cover doesn't seem to help with this. I'll investigate and update the converter.

tsm, 2024-03-20:
Hi, I don't know if it's the same code, but I added the adapter manually from this repository:
https://github.com/javilopezalarcon/Blind-motor-_TZE200_libht6ua-z2m/blob/main/_TZE200_libht6ua.js

and for my needs it is sufficient.

In fact, it gives incorrect information about the status (OPEN to CLOSE and CLOSE to OPEN in relation to the remote control attached to the roller shutter motor), while in Domoticz there is a switch to reverse the behavior and then it works as it should.

Some important information that is not in the manual and is difficult to discover:
1) Zigbee pairing takes place after pressing the reset button several times (5 times?) in a row,
2) By default there is a very short interval from "fully open" to "fully closed", to calibrate a longer interval you should:
- close the blinds and put the belt on the matching motor reel (note: it is best to test this beforehand with the blinds half closed and using the default settings),
- hold down the "pause" button and the arrow button at the same time for a few seconds,
- the roller blind will start to rise, when it reaches the maximum opening level, quickly press "Stop" -> the setting will be saved and the device will know the span.

Jacksoldano, 2024-04-20:
I have a variant of this part TS0601 _TZE200_libht6ua" that pairs by short pressing the "power" button two times in quick succession,

To set the fully closed / open positions is a little different as well,
  1.  Set the device in the “closed" position 
  2. Then hold the "up" and "down" arrow buttons for a few seconds, the indicator LED (Blue) began blinking
  3. Press the "up" arrow (The motor slowly opened the blinds)
  4. When in position press the "stop/pause" button

At this point the motor was configured,

Thanks for the comments in this chain, for anyone having issues getting Z2M to detect this device, take a look at this post https://github.com/Koenkk/zigbee2mqtt/issues/22083 I had to remove from the follwing from the "javilopezalarcon/Blind-motor-_TZE200_libht6ua-z2m" config

    Removed
           const extend = require('zigbee-herdsman-converters/lib/extend');

   Replaced with following at top as per "[MrEcosse](https://github.com/MrEcosse) comment
           const {} = require('zigbee-herdsman-converters/lib/modernExtend

and everything worked perfectly!

CorvetteCole, 2024-07-22:
If you need the direction to be reversed, press the down arrow instead of the up arrow in @Jacksoldano directions. Make sure you start in the closed position same way tho

tsm, 2024-07-23:
Thank you, next time I will try this approach

closed 2025-01-07 by github-actions[bot]