Koenkk/zigbee2mqtt #21501: Z2M crashes >= 1/day with unknown cause, how best to debug?

scottrhoyt, 2024-02-20:
### What happened?

For the last ~4 months, Z2M has been crashing regularly. I am running the docker container on a Synology NAS connecting over ethernet to a CC2652P coordinator (Z-Bee-Duo), and I am alerted by email every time the container restarts unexpectedly. I have looked in the web-based logs tool for any errors or warnings, and I don't see any. I have the debug level set to `Info`. My Z2M instance was running for about 2 years in largely the same environment without this issue. What is the best way to gather more information to debug this crash? Thanks.

### What did you expect to happen?

Z2M runs without crashing.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20221226

### Adapter

Z-Bee-Duo CC2652P

### Setup

Docker on Synology

### Debug log

[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/14348564/log.zip)

mydliar, 2024-04-10:
I might have a similar problem - on 1st April I upgraded from what could be zigbee2mqtt v1.29(ish) to v1.36 and then few days later to v1.36.1. Ever since then my zigbee2mqtt container (runs under Container Manager on Synology NAS) hangs or crashes almost every night.

There is no way to tell what exactly happens, only that the MQTT feed stops until I restart the container (other 9 MQTT data feeds run just fine without ay interruption).  The times when it hangs vary but are generally are between 1:30am and 3am. I do have some nightly backup jobs scheduled around those times, but a) it never used to be a problem; b) the container doesn't hang every night whereas the backups do run every night; and c) I can't reproduce it if I start the backup manually e.g. during the day.

The Zigbee2MQTT logs don't show anything suspicious except the usual zigbee chatter (logging level set to debug).

Any ideas where to look for other clues regarding to what's going on with the Zigbee2MQTT porcess?

Zigbee2MQTT version
[1.36.1](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.36.1) commit: [ffc2ff1](https://github.com/Koenkk/zigbee2mqtt/commit/ffc2ff1)
Coordinator type
EZSP v8
Coordinator revision
6.7.8.0 build 373
Coordinator IEEE Address
0x588e81fffecb5794
Frontend version
0.6.161
zigbee_herdsman_converters_version
19.11.2
zigbee_herdsman_version
0.40.3

mydliar, 2024-04-11:
> I might have a similar problem - on 1st April I upgraded from what could be zigbee2mqtt v1.29(ish) to v1.36 and then few days later to v1.36.1. Ever since then my zigbee2mqtt container (runs under [...]

now logged as #22178

Koenkk, 2024-04-11:
You might want to try out the new [ember](https://github.com/Koenkk/zigbee2mqtt/discussions/21462) driver.

closed 2024-11-09 by github-actions[bot]