Koenkk/zigbee2mqtt #21511: [New device support]: IKEA Rodret E2201 (W2345)

treb33, 2024-02-21:
### Link

https://www.ikea.com/nl/nl/p/rodret-draadloze-dimmer-afstandsbediening-smart-wit-20528128/

### Database entry

{"id":11,"type":"EndDevice","ieeeAddr":"0x8c6fb9fffebabfe3","nwkAddr":10957,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"RODRET Dimmer","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,32,4096,64636],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genPollCtrl":{"attributes":{"checkinInterval":172800}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0025791350","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":113,"hwVersion":1,"dateCode":"20221114","swBuildId":"1.0.47","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1708502388484,"defaultSendRequestWhen":"active","checkinInterval":43200}

### Comments

Device gets the right Model in Z2M, RODRET Dimmer, but is Unsupported.
No option for external definition in Dev Console.
It is the same RODRET that is already supported (https://www.zigbee2mqtt.io/devices/E2201.html) but i think a new model W2345 (?)

### External definition

```shell
Can not generate a external definition, option is not available
```

treb33, 2024-02-21:
Zigbee2MQTT: version 1.30.4-dev commit: [a404f9f]
Coordinator type: zStack3x0
Coordinator revision: 20221226
Coordinator IEEE Address: 0x00124b0025791350
Frontend version: 0.6.129

chris-1243, 2024-02-21:
You should try the latest version of Z2M. You are on 1.30.4 and the latest is 1.35.3.

Lots of changes made and my rodret is perfectly working (firmware 20221114, version 1.0.47).

treb33, 2024-02-21:
Ah stupid...thank you!

closed 2024-02-21 by treb33