Koenkk/zigbee2mqtt #21523: Sonoff Dongle E (EZSP) randomly crashes.

supermattze, 2024-02-21:
### What happened?

I run Home Assistant OS on a RPI 3B+ with a Sonoff Zigbee 3.0 Dongle E with latest firmware (not the multiprotocoll, the dedicated zigbee protocoll).

Randomly the Zigbee2MQTT stops. This is somewhere inbetween 1 minute and 2 days.
I already ordered an original Power Supply for the Pi3B which prevents the HAOS voltage warning.
I also swapped to a brand new SD Card with a fresh installation and backup restore.

Log files shows a end sequence starting with "Adapter disconnected":
`error 2024-02-21 18:37:34: Adapter disconnected, stopping
info  2024-02-21 18:37:34: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
info  2024-02-21 18:37:34: Disconnecting from MQTT server
info  2024-02-21 18:37:34: Stopping zigbee-herdsman...
error 2024-02-21 18:37:34: Failed to stop Zigbee2MQTT`

After enabling herdsman debug Z2M quits right after starting with the following sequence:
`2024-02-21T20:07:52.354Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2024-02-21T20:07:52.354Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2024-02-21T20:07:55.435Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2024-02-21T20:07:55.438Z zigbee-herdsman:adapter:ezsp:uart --> DATA (1,1,0): 0200010500
2024-02-21T20:07:55.442Z zigbee-herdsman:adapter:ezsp:uart --> [7d314021a9512a9a0d7e]
2024-02-21T20:07:55.443Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (2)
Zigbee2MQTT:error 2024-02-21 21:07:55: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2024-02-21 21:07:55: Failed to start zigbee
Zigbee2MQTT:error 2024-02-21 21:07:55: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2024-02-21 21:07:55: Exiting...
Zigbee2MQTT:error 2024-02-21 21:07:55: Error: Failure send version:{"type":"Buffer","data":[1,0,1,0,0,12]}
    at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:646:23
    at Queue.execute (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)
    at Ezsp.version (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:456:13)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:167:9)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:60:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)`

The complete logs are attached below.

The Adapter is on a 0.5m USB Extension.

### What did you expect to happen?

I expect Zigbee2MQTT to run continuously without crashing.

### How to reproduce it (minimal and precise)

wait 1minute to 2 days. It will happen.
Enable herdsman debug to accelerate.

### Zigbee2MQTT version

1.35.3-1

### Adapter firmware version

Zigbee (EZSP) 7.3.1.0 build 176

### Adapter

Zigbee 3.0 USB Dongle Plus (E) from SONOFF

### Setup

Home Assistant OS on Pi 3B+ with new SD and Original Pi 3 Power Supply

### Debug log

[z2m-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14365186/z2m-log.txt)

[herdsman-debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/14365180/herdsman-debug.txt)

kizio1000, 2024-02-22:
Had the same and many other problems over last five months, bottom line is that I got enough of dongle E and couple days ago I have switched to Sonoff dongle P and like with touch of magic wand all works fine and perpetual problems with zigbee network are gone. I should have done it months before instead having this struggle. Just leaving this comment for consideration for those who are trying to get Sonoff E working time after time ;-)

Koenkk, 2024-02-22:
Please try the new [driver](https://github.com/Koenkk/zigbee2mqtt/discussions/21462)

closed 2024-02-22 by Koenkk