Koenkk/zigbee2mqtt #21524: Crash report - Opt-in - Issue Tracker

tech2mar, 2024-02-21:
### Is your feature request related to a problem? Please describe

We all know that we can do things better, Koen is doing such a great job.  This only works if the community also does a good job, we actually have problems related to firmware - compilation - devices... We don't know if it's a device that's causing problems and if there's a company that makes shitty devices we should name and shame them. We have a wife at home who might kill us if the network goes down and the lights go out (again). I know we are not the same, but we need a new approach to some issues.

My idea is that we could opt in to some kind of (automated) information sending like - device goes offline after x time without any user interaction, we should be able to track that kind of behaviour.

Maybe I'm watching too many films ;) I know there is a lot of information that needs to be filtered and someone else needs to interpret that information so we can do a better job.

I'm sure we can do something. 

### Describe the solution you'd like

Collection of bad behaviour of devices that are causing routing problems and end devices that are sleeping differently than the default.

### Describe alternatives you've considered

The alternatives are more complicated and take more time. 

### Additional context

No information

Koenkk, 2024-02-22:
> device goes offline after x time without any user interaction, we should be able to track that kind of behaviour.

Although the idea sounds nice, what would you want to do with this info? You need a lot of context and probably sniffs to determine what happened, right?

tech2mar, 2024-02-22:
Indeed not a simple thing, but creating some sniffing possibility to z2m?. I'm way to far from network engendering just asking questions and maybe asking for something impossible to do.

tech2mar, 2024-02-23:
I know the firmware is different, I came from the Conbee world (using a Wireshark modified version. And a Conbee with a different fw for sniffing.... ...it is nothing that could be done overnight.
My take on this is that maybe we can collect more information about z2m behaviour...
Maybe we could send out logs automatically?
Something the community can do.

closed 2024-12-31 by github-actions[bot]