Koenkk/zigbee2mqtt #21529: Dashboard not showing devices using modernExtend

MFornander, 2024-02-22:
### What happened?

I've been extending the device definition for my TS0201 TuYa sensors to allow setting the displayed temperature unit on the LCD.  I got it working but I wanted to simplify it and just use the new modernExtend since it does a better job of configuring the device.  I got that working too but I lost all the devices using my modernExtend definition on the dashboard.

* I have tried with both main and dev branch.
* Using the plain generated definition results in exceptions preventing device from working at all.
* Removing the endpoint arguments results in a working device but missing on Dashboard.
* I don't think this is a frontend issue since it's all working when using standard expose definition.

### What did you expect to happen?

The dashboard showing all devices, including devices using modernExtern interface.

### How to reproduce it (minimal and precise)

1. Start Z2M and confirm devices showing up under Dashboard.

2. Generate external definition under Dev Console
3. Add generated.js under /data and copy contents (attached below as generated.js)
4. Start Z2M and see exceptions:
Zigbee2MQTT:error 2024-02-22 09:56:15: Exception while calling fromZigbee converter: Definition 'TS0201' has not endpoint defined}

5. Modify generated.js to skip endpoints (attached below as generated_fixed.js)
6. Modify configuration.yaml to use generated.js
7. Start Z2M and confirm device using generated definition under About tab
8. Switch to Dashboard and realize devices using generic_fixed.js are not showing up (everything else is fine)

### Zigbee2MQTT version

1.35.3-dev commit: 22d33360

### Adapter firmware version

20230507

### Adapter

SMLIGHT SLZB-06 over Ethernet

### Setup

Z2M plain to debug (DEBUG=zigbee-herdsman:adapter:zStack:znp* npm start(

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14376731/log.txt)

---
generated.js:
---
const {batteryPercentage, temperature, humidity} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0201'],
    model: 'TS0201',
    vendor: '_TZ3210_ncw88jfq',
    description: 'Automatically generated definition',
    extend: [batteryPercentage(), temperature({"endpoints":["1"]}), humidity({"endpoints":["1"]})],
    meta: {},
};

module.exports = definition;

---
generated_fixed.js
---
const {batteryPercentage, temperature, humidity} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0201'],
    model: 'TS0201',
    vendor: '_TZ3210_ncw88jfq',
    description: 'Automatically generated definition (fixed)',
    extend: [batteryPercentage(), temperature(), humidity()],
    meta: {},
};

module.exports = definition;

grebenyuk51, 2024-04-23:
same issue

shown on dashboard with old exposes
[image]
[image]

not shown on dashboard using modernExtend
[image]
[image]

Interesting thing! When action received, card appears for several milliseconds and disappears
[image]
as you can see, there is no temperature on the card

grebenyuk51, 2024-04-24:
@Koenkk, help us please. What we are doing wrong?

Koenkk, 2024-04-25:
Created a PR to fix this: https://github.com/nurikk/zigbee2mqtt-frontend/pull/2004

Koenkk, 2024-04-28:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-04-28 by Koenkk

grebenyuk51, 2024-04-29:
thanks, everything is working as expected . Except one think. Now I can see all information from exposes tab. Is it a bug or feature?
[image]

Koenkk, 2024-04-30:
Pushed another fix!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

grebenyuk51, 2024-05-01:
yeah, everything looks perfect. thanks a lot!
[image]