Koenkk/zigbee2mqtt #21531: [New device support]: Gledopto PRO GL-C-201P LED Controller

niklasfink, 2024-02-22:
### Link

https://aliexpress.com/item/1005006217216495.html

### Database entry

doesn't pair

### Comments

The 5-in-1 PRO product combines different Gledopto LED Controllers into one. Unfortunately, it doesn't pair with Z2M. I tried it with two different ZigBee sticks, one of them being the Sonoff ZBDongle-E 3.0. When I click on permit join and restart the device, reset it, try any pairing button combination, nothing happens.

When using ZHA, it pairs immediately. And depending on the mode you select on the device, it pairs as a different controller type. (Dimmer only gives Brightness/OnOff, CCT gives white temp control, RGB gives color control via ZigBee and so on)

It would be great if Z2M could have the same smooth pairing with the Gledopto GL-C-201P as ZHA.

[image]

### External definition

```shell
-
```

jstemberg, 2024-03-09:
Pairing also does not work for me

rp2019433, 2024-03-25:
Hi.
Just bought GLEDOPTO **GL-C-201P** LED Controller and I was able to pair it with Z2M, but not with the correct model. Actually, Z2M identifies **GL-C-201P** as GL-C-008P.

But there a big problem:  **GL-C-201P** model has one a push button interface to attach an external switch, so on/off can be also controller by this external switch. Since Z2M does not recognize the correct model, when external switch is used, on/off states are not passed to Z2M, then the real state is not reflected to Home Assistant. I mean, if you turn on/off the controller using external switch, Z2M does not know it.

[image]

It would be good with Z2M could take a look at it.

rp2019433, 2024-03-30:
Hi @niklasfink.

> When using ZHA, it pairs immediately. And depending on the mode you select on the device, it pairs as a different controller type. (Dimmer only gives Brightness/OnOff, CCT gives white temp control, RGB gives color control via ZigBee and so on)

Pairing GL-C-201P with ZHA, are you able have state (on/off) synchronized with Home Assistant when using external push switch?

Thanks.

dodevskid, 2024-04-23:
I purchased eight of these controllers with the intention of integrating them into my home renovation project as indirect CCT lighting. I was amazed with the option to use pushbutton for local on/off and dimming. However, upon testing, I discovered they were recognized as GL-C-008P in zigbee2mqtt, with limited controls to on/off functionality only.

I would appreciate any support in fully integrating these controllers to their intended functionality. I attached screenshots for reference.
[image]
[image]
[image]

dodevskid, 2024-04-26:
> the real state is not reflected to Home Assistant. I mean, if you turn on/off the controller using external switch, Z2M does not

any edit or update on this? I'm having the same problem

dodevskid, 2024-04-26:
> Hi. Just bought GLEDOPTO **GL-C-201P** LED Controller and I was able to pair it with Z2M, but not with the correct model. Actually, Z2M identifies **GL-C-201P** as GL-C-008P.
> 
> But there is a big problem: **GL-C-201P** model has a push button interface to attach an external switch, so on/off can be also controlled by this external switch. Since Z2M does not recognize the correct controller model, when external switch is used, on/off states are not passed to Z2M, then the real state is not reflected to Home Assistant. I mean, if you turn on/off the controller using external switch, Z2M does not know it.
> 
> ![image](https://private-user-images.githubusercontent.com/57510238/316659092-0cc7d146-a399-4d04-819c-073c4cced38c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQxNjUyMTQsIm5iZiI6MTcxNDE2NDkxNCwicGF0aCI6Ii81NzUxMDIzOC8zMTY2NTkwOTItMGNjN2QxNDYtYTM5OS00ZDA0LTgxOWMtMDczYzRjY2VkMzhjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI2VDIwNTUxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRjZjVjMzhjMDk3ZmRlYWI5NTU4ZDJiNmNhNjc2NjZmZTliMDYxZWZhYjVhMTQxNjI0NDM2ZjllZDU3ZmJkYTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8aINeWz_NT3-D_CMuxxP1-XuxK44TYSQO3CCdqtDIXo)
> 
> Would appreciate if Z2M team take a look at it.
> 
> Thank you.

Any update on this? I'm having the same problem. Both eight of this for my renovation project.

Koenkk, 2024-04-28:
Does it work when configuring reporting for `onOff`?

[image]

dodevskid, 2024-04-28:
Yes, thank you a lot.

Oggymator, 2024-06-23:
its a 5in1 Controller, so in will not be found as a GL-C-201P.
It will use the same FW and ID like the normal ones C-007P, C-008P, etc. It depends on the mode you set, dimmer, CCT, RGB, RGBW or RGB-CCT.
Important, first set the mode you need, after that start pairing!

rp2019433, 2024-06-26:
I have been testing the GL-C-201P (current in the GL-C-006P mode) device for a few weeks now. One thing I noticed is the inconsistence in brigthness value when external button is used.

If controller is turned off through Home Assistant and then later turned on using external push button, light brightness on startup is always set to 2%, what was not the last brightness value. It might be caused by the behavior below.

When controller is turned on/off using external push button:

ON
`[2024-06-26 09:31:48] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LED Home Sala', payload '{"action":"","brightness":253,"color_mode":"color_temp","color_temp":500,"color_temp_startup":65535,"last_seen":"2024-06-26T09:31:48-03:00","linkquality":248,"state":"ON","update":{"installed_version":620834817,"latest_version":620834817,"state":"idle"},"update_available":null}'
`
OFF - **here the brightness value is NOT zero, as expected**
`[2024-06-26 09:32:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LED Home Sala', payload '{"action":"","brightness":253,"color_mode":"color_temp","color_temp":500,"color_temp_startup":65535,"last_seen":"2024-06-26T09:32:02-03:00","linkquality":248,"state":"OFF","update":{"installed_version":620834817,"latest_version":620834817,"state":"idle"},"update_available":null}'`

When controller is turned on/off through Home Assistant:

ON
`[2024-06-26 09:32:36] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LED Home Sala', payload '{"brightness":253,"color_mode":"color_temp","color_temp":500,"color_temp_startup":65535,"last_seen":"2024-06-26T09:32:36-03:00","linkquality":255,"state":"ON","update":{"installed_version":620834817,"latest_version":620834817,"state":"idle"},"update_available":null}'`

OFF - **here the brightness value is ZERO, so the current brightness is not kept when controller is turned ON later** 
`[2024-06-26 09:33:17] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/LED Home Sala', payload '{"brightness":0,"color_mode":"color_temp","color_temp":500,"color_temp_startup":65535,"last_seen":"2024-06-26T09:33:17-03:00","linkquality":252,"state":"OFF","update":{"installed_version":620834817,"latest_version":620834817,"state":"idle"},"update_available":null}'	`

Maybe a soution could be to create a Brightness value startup, as Color temp startup, so user can set "previous" to brightness value as well.

[image]

antonrez89, 2024-07-05:
@rp2019433 does you solve the problem? I have the same :(

rp2019433, 2024-07-05:
> @rp2019433 does you solve the problem? I have the same :(

Hi. I've done some changes (above). Still testing the results.

[image]

antonrez89, 2024-07-05:
@rp2019433 a have made the same, but it doesnt have effect. now only brightness level publish in mqtt.
I made fix with automation. but it is very ugly...need better solution

`- id: '1720187164195'
  alias: test push dim
  description: ''
  trigger:
  - platform: device
    domain: mqtt
    device_id: cb2a76f26dea6c0b44e2e677c2809422
    type: action
    subtype: 'on'
  condition:
    - condition: numeric_state
      entity_id: light.gledopto_dimmer_2_20a
      attribute: brightness
      below: 3
  action:
    - device_id: cb2a76f26dea6c0b44e2e677c2809422
      domain: light
      brightness_pct: 100
      type: turn_on
      entity_id: 6f68e6830ad8a38c72e1c9aa524e3096
  mode: single`

dodevskid, 2024-07-31:
I'm using the Adaptive Lighting integration from HACS, and it works perfectly when controlling the lights through the Home Assistant app. However, I'm encountering an issue with physical switches. When I press the physical button, the light state changes to the last known state rather than following the adaptive lighting settings.

For example, if I turn off the lights through the app at night when the brightness is at 2%, and then the next day I press the physical switch, the light turns on at 2% brightness instead of adjusting to the current adaptive lighting settings.

Is there any walkaround for this problem?

rp2019433, 2024-08-11:
> I can't even get it to pair with my HA via Z2M :-/ Tried to reset the thing several times but nothing happens when I try to pair.
> 
> Any Ideas? Some secret magic spell I have to say? ^^

Hi. Device must pair with Z2M without any problem. I have several of them working. Check basic things like software versions running (Z2M, Dongle, HA, etc) and your logs to understand what is going on.

Michael-Hennemann, 2024-08-11:
Z2M on HA -> both latest.  Dongle is a Zonoff Zigbee 3.0 stick.. the older one. (TI chip I think)  
Haven't done any firmware update of the coordinator because until now: never touch a running system 😅 

The log file says basically nothing. There is not a single entry about the new device. Maybe it's DoA ?

The pairing process should be pretty straight forward?
 
1) set Z2M to allow join
2) power on led controller

Did I miss something? 

cheers
Michael

Oggymator, 2024-08-11:
I tried is last week and it works.
But there are something wrong by pairing and mode switching, most time it is wrong mode that is connected. Will explain later.

I use latests Z2M with ZigStar dongle, no Silicon Chip

rp2019433, 2024-08-12:
> Z2M on HA -> both latest. Dongle is a Zonoff Zigbee 3.0 stick.. the older one. (TI chip I think) Haven't done any firmware update of the coordinator because until now: never touch a running system [...]

Hi. This is the procedure to pair the device:
[image]

Michael-Hennemann, 2024-08-14:
Hi rp2019433, 
thanks for your message, but I think the image you sent me is not an procedure it is merely a description of what each button does. I scanned the manual I got with my zigbee controller. (please see below) There are two kinds of paring: Gateway Paring and Touch Link Pairing Mode. I tried both, reset the led controller probably a dozen times but no luck.... 

[image]

Oggymator, 2024-08-14:
> Hi rp2019433, thanks for your message, but I think the image you sent me is not an procedure it is merely a description of what each button does. I scanned the manual I got with my zigbee controller. (please see below) There are two kinds of paring: Gateway Paring and Touch Link Pairing Mode. I tried both, reset the led controller probably a dozen times but no luck....
> 
> ![grafik](https://private-user-images.githubusercontent.com/45975814/357968908-b021a8ef-72b1-4206-afeb-81382710acf3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM2NjMyMjMsIm5iZiI6MTcyMzY2MjkyMywicGF0aCI6Ii80NTk3NTgxNC8zNTc5Njg5MDgtYjAyMWE4ZWYtNzJiMS00MjA2LWFmZWItODEzODI3MTBhY2YzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODE0VDE5MTUyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgzYTUzMzhlYTEwM2ZhZTRkYjNkYmJmYjcwNmYzNGYwOGQxYjMxMDQ0Mzg2ZTViNjc4MTE1ZTc4NmE3NGI0MDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Ttz0Bdtd893WZBE72C21bguR6Mb8AA9PG5EvkkNeieg)

It’s easy.
Start searching for new devices, after that push the reset button for over 5 seconds till the led stripes flash.
Now the software will find it.

Michael-Hennemann, 2024-08-14:
> 
> It’s easy. Start searching for new devices, after that push the reset button for over 5 seconds till the led stripes flash. Now the software will find it.
> 

That is what I've tried like 15 times so far - no luck. Nothing happens. No log entry, no error message in Z2M, just nothing. 
Maybe the led controller was DoA and I will just send it back to Mr. Bezos....

niklasfink, 2024-08-14:
> That is what I've tried like 15 times so far - no luck. Nothing happens. No log entry, no error message in Z2M, just nothing. Maybe the led controller was DoA and I will just send it back to Mr. [...]

I doubt that it’s a defective unit. I bought 6 of them and they all behave the same. After fiddling around with them for an hour I got an unreliable connection for one, after I have change the mode to RGBW and then it announced itself as a different Gledopto RGBW device.
But never got it to work again afterwards with Z2M.

With ZHA it works, as written in my thread opening at the top.

Michael-Hennemann, 2024-08-15:
Got myself a GL-C-008P. Works as expected. 
Either the GL-C-201P that I got was defective or Z2M support for the device is buggy, I guess.

Oggymator, 2024-08-16:
Here is a video how to connect.

https://github.com/user-attachments/assets/09d8bd02-275b-4afe-92e0-7e9a459bc7f2

Michael-Hennemann, 2024-08-16:
> Here is a video how to connect.
> 2cb1e49f-f632-471c-84ad-4bd4f736fe48.mov

Thanks for the video!  That is exactly what I tried several times. 
I guess my unit must have been defective...

closed 2025-01-07 by github-actions[bot]

IvanNIvanov, 2025-01-09:
The same issue for me. I just received one GL-C-201P device and cannot connect it. I am not sure but I think that the light indicator should turn on when I plug in the power. In my case the light is off even if the power is plugged in. I tried different power supplies with no success. Z2M cannot find it. Even Tuya Zigbee Gateway cannot see the device. So I suspect this one is defective and cannot be powered on.

mminehanNZ, 2025-02-02:
I can confirm that a GL-C-203P paired in CCT mode is identified as a GL-C-006P. You have to set the working mode on the controller before pairing.

Then you need to set reporting for OnOff, LevelCtrl and LColorCtrl in Zigbee2MQTT to get reporting of the relevant changes made when using the controller's "Push" button.

kszere, 2025-07-19:
[image]
If anyone needs a template for setting up device reporting, I didn't understand the issue myself. I'm starting with Z2M. I hope it helps someone. Works for me.! :)

3DCube-KNX, 2025-08-22:
Hello
I have a GL-C-201P in CCT mode on HA with Z2MQTT, recognized as GL-C-006P.

I've configured the image settings, and the on/off and brightness settings work.
But when I try to use currentHue and currentsaturation, I get a red pop-up with an error message that doesn't give me time to copy, but says insufficient space.

Any help?

mosanwen, 2025-08-25:
> Hello I have a GL-C-201P in CCT mode on HA with Z2MQTT, recognized as GL-C-006P.
>
> I've configured the image settings, and the on/off and brightness settings work. But when I try to use currentHue [...]

Hello
 GL-C-001P - Gledopto is 5 in 1 Smart LED Controller it include 5 modes,and Modes/Outputs can switch (click Option button to switch)
GL-C-008P - RGBCCT Mode (White light on)
GL-C-007P - RGBW Mode (Yellow light on)
GL-C-003P - RGB Mode (Blue light on)
GL-C-006P - CCT Mode (Green light on)
GL-C-009P - Dimmer Mode (Red light on)
it is recognized as GL-C-006P,that is to say  you selcet a CCT mode , this mode is not support currentHue and currentsaturation. you must switch it to RGBCCT mode  or RGBW mode(It depends on what kind of light strips you use.)

You can refer to this link:https://github.com/Koenkk/zigbee2mqtt/issues/24091

3DCube-KNX, 2025-08-31:
> > Hola tengo un GL-C-201P en modo CCT en HA con Z2MQTT, reconocido como GL-C-006P.
> > He configurado la imagen, y los ajustes de encendido/apagado y brillo funcionan. Pero cuando intento usar currentHue y currentSaturation, aparece una ventana emergente roja con un mensaje de error que no me da tiempo a copiar, sino que indica que no hay suficiente espacio.
> > ¿Alguna ayuda?
> 
> Hola GL-C-001P - Gledopto es un controlador LED inteligente 5 en 1 que incluye 5 modos y los modos / salidas se pueden cambiar (haga clic en el botón Opción para cambiar) GL-C-008P - Modo RGBCCT (luz blanca encendida) GL-C-007P - Modo RGBW (luz amarilla encendida) GL-C-003P - Modo RGB (luz azul encendida) GL-C-006P - Modo CCT (luz verde encendida) GL-C-009P - Modo de atenuación (luz roja encendida) se reconoce como GL-C-006P, es decir, selecciona un modo CCT, este modo no es compatible con currentHue y currentsaturation. debe cambiarlo al modo RGBCCT o al modo RGBW (depende del tipo de tiras de luz que use).
> 
> Puedes consultar este enlace: [#24091](https://github.com/Koenkk/zigbee2mqtt/issues/24091)

Hello.
Thanks for the suggestion.
I changed it, and it now appears as GL-C-008P (RGBCCT), since I have a CCT LED strip.
I still get the error message "insufficient space" for currentHue and currentSaturation.

Best regards.