Koenkk/zigbee2mqtt #21532: Devices no longer bind

metaljay, 2024-02-22:
### What happened?

Running the latest release in docker (updated via watchtower).
My devices no longer connect and if I try to remind I get this error:

zigbee2mqtt:error%202024-02-22%2018%3A58%3A09%3A%20Failed%20to%20configure%20'Nursery%20Switch',%20attempt%202%20(Error%3A%20Bind%200x7cb94c77c9a20000/1%20genOnOff%20from%20'0x00124b001caa8553/1'%20failed%20(AREQ%20-%20ZDO%20-%20bindRsp%20after%2010000ms)%0A%20%20%20%20at%20Object.start%20(/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23)%0A%20%20%20%20at%20/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:699:28%0A%20%20%20%20at%20Queue.execute%20(/app/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20)%0A%20%20%20%20at%20Endpoint.bind%20(/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:551:13)%0A%20%20%20%20at%20setupAttributes%20(/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:54:13)%0A%20%20%20%20at%20result.configure%20(/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:126:13)%0A%20%20%20%20at%20Object.configure%20(/app/node_modules/zigbee-herdsman-converters/src/index.ts:141:25)%0A%20%20%20%20at%20Configure.configure%20(/app/lib/extension/configure.ts:120:13)%0A%20%20%20%20at%20Configure.onMQTTMessage%20(/app/lib/extension/configure.ts:55:21)%0A%20%20%20%20at%20EventEmitter.wrappedCallback%20(/app/lib/eventBus.ts:174:17))%0AZigbee2MQTT:info%20%202024-02-22%2018:58:09:%20MQTT%20publish:%20topic%20'zigbee2mqtt/bridge/response/device/configure',%20payload%20'%7B%22data%22:%7B%22id%22:%22Nursery%20Switch%22%7D,%22error%22:%22Failed%20to%20configure%20(Bind%200x7cb94c77c9a20000/1%20genOnOff%20from%20'0x00124b001caa8553/1'%20failed%20(AREQ%20-%20ZDO%20-%20bindRsp%20after%2010000ms))%22,%22status%22:%22error%22,%22transaction%22:%22g9yae-1%22%7D'

### What did you expect to happen?

Devices to work

### How to reproduce it (minimal and precise)

Appears the same as this: https://github.com/Koenkk/zigbee2mqtt/issues/21162

### Zigbee2MQTT version

Latest

### Adapter firmware version

Latest

### Adapter

ZZH

### Setup

Docker

metaljay, 2024-02-22:
Just to add, I’ve tried the latest dev image and still no fix

blindguynar, 2024-02-25:
i think i have this problem too.  I can join any new devices on my network for some reason.

closed 2024-09-23 by github-actions[bot]