Koenkk/zigbee2mqtt #21538: Ui Problem after update to 1.53.3

moshiko2312, 2024-02-23:
### What happened?

all the ui like a list

[image]

### What did you expect to happen?

i expected to get normal UI on web 

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

7.3.2.0 build 212

### Adapter

EZSP v12 Skyconnect HA

### Setup

pi 4 home assistant core on Docker compose

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14383131/log.txt)

thkn1973, 2024-02-23:
I have the same problem.
However, the problem only occurs in the Chrome Browser.
In e.g. Brave or Microsoft Edge it is ok

redpixy, 2024-02-23:
Mine was back to normal after clearing the browser cache.

gizmo69the2nd, 2024-02-23:
Same problem on Firefox for Ubuntu - 122.0.1 (64-bit)

moshiko2312, 2024-02-24:
Ok thanks I wi check with safari

Koenkk, 2024-02-24:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/21268

closed 2024-02-24 by Koenkk