Koenkk/zigbee2mqtt #21543: [Feature request]: Expose "Reconfigure" to Home Assistant or Create Native Auto-Reconfigure for Offline Devices

sirjmann92, 2024-02-23:
### Is your feature request related to a problem? Please describe

Occasionally devices will fall off the network or go "Offline". There doesn't seem to be a rhyme or reason to which devices, probably caused by temporary interference and lack of response to Ping requests (2/2). A simple Reconfigure in Z2M brings the device back. This doesn't happen very often, and is likely just the nature of a Zigbee network operating on the same frequency as so many other devices in a home (2.4) ,  which makes it difficult to find a proper solution. You can imagine the frustration if one of the devices that goes offline is one that controls lights while you're on the toilet, shower, or a dark room like a basement while you've got your hands full (wife aggro!)

### Describe the solution you'd like

As a workaround, to bring devices back online without having to manually open Z2M, locate any "Offline" devices and select "Reconfigure", it would be nice to be able to create an automation in Home Assistant that listens for devices to go "Offline" and then performs a "Reconfigure" on the device to bring it back.

### Describe alternatives you've considered

An alternative to exposing the Reconfigure as an endpoint to HA would be a native automation or feature within Z2M to continually check for devices that are offline (or at a configurable interval) and perform the Reconfigure. Both solutions would be ideal, although either would satisfy the requirement/request.

### Additional context

I've noticed there are certain endpoints available to Home Assistant now, sensors, switches, etc... like Permit Join, which makes it possible, and very handy, to do things like "Alexa, open Zigbee" with an automation that runs in HA which turns on Z2M's Permit Join with a timer to disable it after a set period (love it!)... Expanding the integration further by exposing additional features to HA would allow for more control and freedom of operation. This feature has the potential to dramatically improve the stability and reliability of a Zigbee network using Zigbee2MQTT (depending on the setup and environmental conditions).

DAVIZINH0, 2024-05-11:
hello.
this is a good idea. exist any progress or workarround to make this?
maybe in home assistant we can detect when a device is availabe (this is easy)
and then send a mqtt topic to reconfigure this device. is this posible???

efenex, 2024-05-30:
I have the same need for an automation to be able to trigger reconfigure, as I have aqara switches that will regularly reset and require a reconfigure to function normally again.

praun, 2024-08-18:
yes, very good idea this. To be honest i was kinda baffled that this wasnt already possible..! 

I have a zigbee device that sometimes gets stuck in the wrong state and then I have to go into z2m and manually reconfigue it, and it instantly pops back into the correct state. 

So i just made a quick google search, to learn how to automate this process, but sadly i just ended up here..

boredmthfkr, 2024-09-07:
A very good idea. We need this... I am tired of how many times I have to manually trigger the reconfigure for a lightbulb I have that goes offline. Another identical lightbulb, no more than 2 meters away has no issue.

GuyFullalove, 2024-10-10:
This has my vote :)

M4Marc, 2024-11-04:
Same Problem anything New about this?

Ekylypse, 2024-11-25:
You can publish to MQTT topic to do this already. 

Topic: zigbee2mqtt/bridge/request/device/configure

Payload: {"id": "friendly_name"}

Friendly name can also be ieee_address.

This information can be found in the documentation:
https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request

M4Marc, 2024-11-25:
Hey Anthony,
thank you for your answer.
Now i can try it.Hopfully it will solve my problem.
Very best wishes from Germany
Marc

⁣BlueMail for Android herunterladen ​

thebriggsy, 2025-01-29:
I could really do with this feature - I'm actually not using Z2MQTT but am using ZHA - I have a sensor which occasionally drops off (every other week or so) and the way I solve it is I press the Reconfigure in the UI. Solves it every time - I'd love to be able to do this via an automation on a regular basis.

Ekylypse, 2025-01-29:
Well, I haven't read the ZHA docs, but if you'll look at my [previous reply](https://github.com/Koenkk/zigbee2mqtt/issues/21543#issuecomment-2496518686) then you'll see it's already possible to do with an automation with Z2M. I do it with one device that's a pain in the butt.

sirjmann92, 2025-01-30:
@Ekylypse , this request is to expose a feature in Z2M to HA's API, you can't "do that already". You can, however, use your solution as a workaround to solve the same problem. It's a workaround because you can't use HA's UI to create automations, cards, buttons, etc... without additional effort. Anytime you have to manually use a script or install additional software/plugins, it's a workaround. Some people struggle more than others with scripting, or they just won't use workarounds.

I still appreciate the reply and suggestion. I didn't know it was possible. I also didn't thoroughly read the documentation for possible workarounds or alternative solutions, I just know it's not currently exposed to HA and it would be nice to have.

I'll definitely use this in the short-term for problematic devices, but it would be a bit time-exhaustive to do this for all of my devices.

Thank you!

dajomas, 2025-02-21:
I just had a couple devices that needed the reconfigure and I would have been very happy if this feature existed.

so here is my vote to add this feature.

M4Marc, 2025-04-21:
@Ekylypse I've only just been able to devote myself to the topic again and have failed to create an automation. I'm failing at the payload topic, etc. It would be great to get a concrete example. I would like to initiate a reconfiguration of one or better all  offline lamps via an automation.

rrozema, 2025-06-05:
As @Ekylypse described, even though it's not through a call to Z2M, you can already perform the action from within HA. As an example here's the action you could put into a script or automation to reconfigure a device called "friendly_name". i.e. just replace that string "friendly_name" by your device's name then execute it. I just tested it and it works.
`action: mqtt.publish
data:
  qos: "0"
  payload: "{\"id\": \"friendly_name\"}"
  topic: zigbee2mqtt/bridge/request/device/configure
  evaluate_payload: false
`