Koenkk/zigbee2mqtt #21546: Can't connect to used Hue Play Bar

Blackspell01, 2024-02-23:
### What happened?

I bought a pair of used hue play bars and did not think about not being able to connect them.
As I have no hue bridge and no Hue dimmer switch, apparently I can't factory reset them.
According to the documentation, I should be able to find them with touchlink. 
Yet z2m always returns: Detected 0 touchlink devices.
Im using the Sonoff Dongle E with the Silicon Labs chip on which touchlink should be supported.
Is there any way I could connect to the hue devices without buying a hue bridge?

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

7.3.1.0 build 176

### Adapter

Sonoff Dongle E

### Setup

Docker

danieledwardgeorgehitchcock, 2024-03-08:
How far away do you have the bars from your co-ordinator? you need the devices basically on top of one another and it usually takes several attempts. Also sometimes helps to power up the devices just after you have hit the discover button.. Touchlink is by far the flakiest if reset methods for Hue products and I'd definitely recommend you consider getting a cheap dimmer remote off of ebay :-)

Blackspell01, 2024-03-08:
The culprit was the firmware of the Sonoff Dongle. The one It comes with from ITead happens to have touchlink disabled apparently

m4cko5, 2024-03-22:
I will just reuse this topic, since I have the somewhat same problem.
The other difference is that I have flashed the Dongle-E to the Silicon Labs Multiprotocol firmware.

Since I also have a Ikea remote I tried resetting with it, this resulted in the remote being paired to the remote, but not found by Z2M.

closed 2024-10-19 by github-actions[bot]