Koenkk/zigbee2mqtt #21554: SR-ZG9080A Open and Close is wrong

Oggymator, 2024-02-24:
### What happened?

I connect the SR-ZG9080A to my shutter and have to use invert true, that it works correct. But I connect it like written in the manual, so my invert value must be false.

### What did you expect to happen?

If I use the invert true, in the complete Home Assistant the open and close state are wrong.
I can use open and close as correct command, but the state is wrong.
Also very terrible to use it in NodeRed, because als commands must be inverted.

Maybe someone could set the values correctly. So everything that is now for Invert true is actually the value for Invert false and vice versa.

See also
https://github.com/Koenkk/zigbee2mqtt/issues/18945

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20221226

### Adapter

ZigStar

### Setup

Add-on Home Assistant Intel NUC

Oggymator, 2024-03-10:
somebody can help on it?

anharald, 2024-05-05:
solution is there:
[https://github.com/Koenkk/zigbee-herdsman-converters/pull/7334](url)

Oggymator, 2024-05-05:
Yes, with last update.
Thanks!

Oggymator, 2024-08-29:
Now I found a problem.
Because of the changes, the current sate of the shutter is wrong.
If it set to 50% closed or something more or less, it is shown as open. But that is wrong, must be closed if the value is less like 100%. Only 100% must be open, all other values must be closed.

Can somebody fix it?

Oggymator, 2024-11-11:
push

closed 2025-01-19 by github-actions[bot]