Koenkk/zigbee2mqtt #21557: Philips Hue Switch V2 - No "On Hold" Option

ValiantGB, 2024-02-24:
### What happened?

I have two Philips Hue Switch V2 and have connected both to Zigbee2mqtt.  The first remote connected fine, and in automations in Home Assistant, provides the following "hold" button conditions, in addition to the usual "press" and "release" options:

"up_hold" action
"up_hold_release" action
"on_hold" action
"on_hold_release" action
"off_hold" action
"off_hold_release" action

However when I look to create an automation for the second remote in Home Assistant, none of these "hold" options are available.

### What did you expect to happen?

I expected a "hold" condition to be made available for automations.

### How to reproduce it (minimal and precise)

I tried to remove and re-pair the Hue Switch - the same limitations apply.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20210708

### Adapter

SONOFF Universal Zigbee 3.0 USB Dongle Plus Gateway

### Setup

Home Assistant OS on Intel NUC

mundschenk-at, 2024-03-02:
Did you try holding the buttons on the remote in question? If I remember correctly, the available device actions are provisioned dynamically by HA.

ValiantGB, 2024-03-03:
Yes - I did long hold / release on every button and the options now appear in Home Assistant automations - thank you!

closed 2024-03-03 by ValiantGB