Koenkk/zigbee2mqtt #21558: Supported Ikea LED2101G4 bulb pairs but is marked "unsupported". Interview fails.

james-fry, 2024-02-24:
### What happened?

I just purchased an Ikea Tradfri LED2101G4 bulb.
When I attempt to pair it with Z2M pairing succeeds but interview fails (see logs).
I have tried:
- retried a lot of times
- tried after fresh restart of Z2M addon
- tried after resetting the coordinator and restart of Z2M
- tried both in location where bulb will be postions and with the bulb 30cm from the coordinator.

Is this a problem with code? Or is my bulb DOA? 

### What did you expect to happen?

Expected it will pair successfully as it is a supported device.

### How to reproduce it (minimal and precise)

Try to pair this device.

### Zigbee2MQTT version

1.35.3

### Adapter firmware version

20230507

### Adapter

LAUNCHXL-CC1352P Evaluation board

### Setup

Addon on HAOS VM x86-64 connected by passed through USB

### Debug log

Info 2024-02-24 20:28:07Device '0x60b647fffe678b34' joined
Info 2024-02-24 20:28:07MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/event', payload '{"data":{"friendly_name":"0x60b647fffe678b34","ieee_address":"0x60b647fffe678b34"},"type":"device_joined"}'
Info 2024-02-24 20:28:07MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"message":{"friendly_name":"0x60b647fffe678b34"},"type":"device_connected"}'
Info 2024-02-24 20:28:07Starting interview of '0x60b647fffe678b34'
Info 2024-02-24 20:28:08MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/event', payload '{"data":{"friendly_name":"0x60b647fffe678b34","ieee_address":"0x60b647fffe678b34","status":"started"},"type":"device_interview"}'
Info 2024-02-24 20:28:08MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x60b647fffe678b34"},"type":"pairing"}'

then later

Error 2024-02-24 20:30:26Failed to interview '0x60b647fffe678b34', device has not successfully been paired
Info 2024-02-24 20:30:26MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/event', payload '{"data":{"friendly_name":"0x60b647fffe678b34","ieee_address":"0x60b647fffe678b34","status":"failed"},"type":"device_interview"}'

DB entry yealds no additional info:

{"id":30,"type":"Unknown","ieeeAddr":"0x60b647fffe678b34","nwkAddr":59061,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1708806491901}

james-fry, 2024-03-09:
Just updating here that I replaced the bulb with a new one today, but same issue is occuring.
Bulb joins the network, interview starts but never completes.
Bulb remains in DB but is not paired and is inoperable.
Bulb needs to be force deleted.

srett, 2024-03-09:
Do you have any repeaters? If so, try to explicitly only permit joining via one router, or the coordinator. I've had a TRV that would only pair with my coordinator, but not with my repeaters, and the opposite was true for PIR motion detector.

If that fails too and if it's not too much hassle, could you try on 1.33.x?

Otherwise, [zigbee-herdsman debug output](https://www.zigbee2mqtt.io/guide/usage/debug.html) could maybe help the devs pinpoint the problem.

james-fry, 2024-03-10:
> Do you have any repeaters? If so, try to explicitly only permit joining via one router, or the coordinator. I've had a TRV that would only pair with my coordinator, but not with my repeaters, and [...]

I do have routers and I tried pairing in place where the bulb will go (i.e. expecting to pair via routers) and I tried pairing with the bulb less than 1m from the coordiator.
I had not however explicitly forced coord or a router so tried this approach.
Unfortunately forcing pairing via coordinator didnt work:

```
Info 2024-03-10 15:17:46MQTT publish: topic 'zigbee2mqtt_cc1352p/0x003c84fffe4204c9', payload '{"device":{"friendlyName":"0x003c84fffe4204c9","ieeeAddr":"0x003c84fffe4204c9","networkAddress":21562,"type":"Unknown"},"last_seen":"2024-03-10T15:17:45+00:00"}'
Info 2024-03-10 15:17:46Device '0x003c84fffe4204c9' joined
Info 2024-03-10 15:17:46MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/event', payload '{"data":{"friendly_name":"0x003c84fffe4204c9","ieee_address":"0x003c84fffe4204c9"},"type":"device_joined"}'
Info 2024-03-10 15:17:46MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"message":{"friendly_name":"0x003c84fffe4204c9"},"type":"device_connected"}'
Info 2024-03-10 15:17:46Starting interview of '0x003c84fffe4204c9'
Info 2024-03-10 15:17:46MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/event', payload '{"data":{"friendly_name":"0x003c84fffe4204c9","ieee_address":"0x003c84fffe4204c9","status":"started"},"type":"device_interview"}'
Info 2024-03-10 15:17:46MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x003c84fffe4204c9"},"type":"pairing"}'
Error 2024-03-10 15:20:04Failed to interview '0x003c84fffe4204c9', device has not successfully been paired
Info 2024-03-10 15:20:04MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/event', payload '{"data":{"friendly_name":"0x003c84fffe4204c9","ieee_address":"0x003c84fffe4204c9","status":"failed"},"type":"device_interview"}'
Info 2024-03-10 15:20:04MQTT publish: topic 'zigbee2mqtt_cc1352p/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x003c84fffe4204c9"},"type":"pairing"}'
```

Unfortunately also same result with pairing explicitly via a router.
:(

Is the process to roll back to a specific version of Z2M (HA addon) documented anywhere?

I know this bulb model works "usually" as I already have a different one paired:

[image]

I tried pulling the logs from the z2m docker container. but there was no additional info than I get from the addon logs or logs in the Z2M UI (as pasted above with debug logging enabled).

srett, 2024-03-11:
> Is the process to roll back to a specific version of Z2M (HA addon) documented anywhere?

Sorry, no idea here, I'm using a manual install, but hopefully due to the popularity this is only a google search away, same for enabling the herdman logs in that setup.
 
I'm not a dev on z2m, I just happened to be affected by network instability in newer versions, and there seem to be a few issues opened over the past weeks that sound like similar problems, like unable to pair, devices leaving the network right after pairing, or devices becoming unresponsive but still sending updates. In many such issues, people had better success with versions before 1.34.x, and when I tried switching to 1.33.x my issues went away too, but I'm still trying to figure out what might be going on here.

closed 2024-10-09 by github-actions[bot]