Koenkk/zigbee2mqtt #21562: ZigbeeTlc devices by pvvx

robvanoostenrijk, 2024-02-25:
### Link

https://github.com/pvvx/ZigbeeTLc/

### Database entry

{"id":10,"type":"EndDevice","ieeeAddr":"0x58dd5782342d585a","nwkAddr":49589,"manufId":4417,"manufName":"Qingping","powerSource":"Battery","modelId":"CGDK2-z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,32,1026,1029,516],"outClusterList":[25,516],"clusters":{"genBasic":{"attributes":{"modelId":"CGDK2-z","manufacturerName":"Qingping","powerSource":3,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":1,"dateCode":"20240211","swBuildId":"3001-0120"}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":193,"batteryVoltage":30}},"hvacThermostat":{"attributes":{}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2302}},"msRelativeHumidity":{"attributes":{"measuredValue":6475}},"hvacUserInterfaceCfg":{"attributes":{"256":4,"257":0,"tempDisplayMode":0}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe4dad03","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe4dad03","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe4dad03","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe4dad03","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0,"manufacturerCode":null}],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":1,"dateCode":"20240211","swBuildId":"3001-0120","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1708839867942,"checkinInterval":3600}

### Comments

Second device database entry:

`{"id":22,"type":"EndDevice","ieeeAddr":"0xa4c1386f9bbcc0eb","nwkAddr":7574,"manufId":4417,"manufName":"Tuya","powerSource":"Battery","modelId":"TS0201-z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,32,1026,1029],"outClusterList":[25,516],"clusters":{"genBasic":{"attributes":{"modelId":"TS0201-z","manufacturerName":"Tuya","powerSource":3,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":1,"dateCode":"20240211","swBuildId":"3001-0120"}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":32}},"msTemperatureMeasurement":{"attributes":{"measuredValue":480}},"msRelativeHumidity":{"attributes":{"measuredValue":5305}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe4dad03","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe4dad03","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe4dad03","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x040d84fffe4dad03","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10,"manufacturerCode":null},{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null},{"cluster":1029,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100,"manufacturerCode":null}],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":1,"dateCode":"20240211","swBuildId":"3001-0120","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1708839787377,"checkinInterval":3600}}`

pvvx makes a Zigbee firmware available for a series of Thermometer / Humidity measurement devices.
All of these can be programmed with identical custom firmware.

All devices support:

- Temperature measurement (0.1 degrees accuracy)
- Humidity measurement (0.1 % accuracy)
- Temperature calibration (-12.7 degrees - +12.7 degrees)
- Humidity calibration (-12.7 % - +12.7 %)

Devices with a display support, the above plus:
- Temperature display Unit (Celcius, Fahrenheit)

Devices with extended display support, the above plus:
- Comfort indicator display (Show, Hide)
- Minimum comfort temperature (-127 - +127)
- Maximum comfort temperature (+127 - +127)
- Minimum comfort humidity (0 - 100)
- Maximum comfort humidity (0 - 100)

Below is an external definition which works for the 3 device types with some code re-use across device types.
But my knowledge of Zigbee Herdsman internals is not sufficient enough to cleanly implement this for a pull request.

Hence requesting help from the experts here to hopefully help clean this up and support all ZigbeeTlc pvvx devices in a single converter file.

Definitions are:
- temperature -> measuredValue
- Humidity -> measuredValue
- PowerCfg -> batteryVoltage
- PowerCfg -> batteryPercentageRemaining

- hvacUserInterfaceCfg -> tempDisplayMode (`attribute 0x000 - uint8`):
  [image]

- hvacUserInterfaceCfg -> scheduleProgrammingVisibility (`attribute 0x001 - uint8`):
   Comfort Indicator display
  [image]

- hvacUserInterfaceCfg  -> temperature_calibration (`attribute 0x100 - uint8`)
   Temperature calibration, in 0.1° steps.
- hvacUserInterfaceCfg  -> humidity_calibration(`attribute 0x101 - uint8`) 
   Humidity offset, in 0.1 % steps.
- hvacUserInterfaceCfg  -> comfort_temperature_min (`attribute 0x102 - uint8`) 
   Comfort parameters/Temperature minimum, in 1°C steps.
- hvacUserInterfaceCfg  -> comfort_temperature_max (`attribute 0x103 - uint8`)
   Comfort parameters/Temperature maximum, in 1°C steps.
- hvacUserInterfaceCfg  -> comfort_humidity_min(`attribute 0x104 - uint8`) 
  Comfort parameters/Humidity minimum, in 1% steps.
- hvacUserInterfaceCfg  -> comfort_humidity_max(`attribute 0x105 - uint8`) 
   Comfort parameters/Humidity maximum, in 1% steps.

### External definition

```shell
// requires zigbee2mqtt v1.34+
// external converter for ZigbeeTLc by pvvx
[... 135 lines cut ...]
        ota: ota.zigbeeOTA,
        extend: [
            TEMP_CALIBRATION,
            RH_CALIBRATION
        ],
        configure: CONFIGURE
    },
    {
        zigbeeModel: ['TS0201-z'],
        model: 'Temp & RH Monitor - TS0201',
        vendor: 'Tuya',
        description: 'Temperature & humidity sensor using pvvx ZigbeeTLc',
        fromZigbee: [fromZigbee.temperature, fromZigbee.humidity, fromZigbee.battery, fromZigbee.hvac_user_interface],
        exposes: [e.battery(), e.temperature(), e.humidity()],
        ota: ota.zigbeeOTA,
        extend: [
            TEMP_CALIBRATION,
            RH_CALIBRATION
        ],
        configure: CONFIGURE
    }
]

module.exports = definitions;
```

empireages, 2024-03-16:
Thanks for the external converter !
It negociate a LYWSD03MMC perfectly at the configure stage while the default in z2m [1.36.0] failed to do so.

Thanks

SanderGols-TomTom, 2024-07-01:
Would this also fix the issue with invalid default values?

[image]

HA log:
```log
2024-07-01 14:54:32.823 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c1388c4eabc979_comfort_temperature_min: 200 (range -50.0 - 120.0)
2024-07-01 14:54:32.823 ERROR (MainThread) [homeassistant.components.mqtt.number] Invalid value for number.0xa4c1388c4eabc979_comfort_temperature_max: 250 (range -50.0 - 120.0)
```

closed 2024-12-30 by robvanoostenrijk