Koenkk/zigbee2mqtt #21564: Unstable connection to EMQX broker

Audiobuzz, 2024-02-25:
### What happened?

Recently moved over from using mosquitto to using a kubernetes hosted EMQX instance. 

ZIgbee2mqtt seems unable to maintain a stable connection to the emqx broker whereas other mqtt clients are working fine with it. 

z2m continues as if it is still connected (logs indicate it is still publishing) but the broker logs and the other clients never receive the topic updates. The emqx has log entries for "congested connection" only for the z2m client. This error indicates the TCP buffer is full from the client not reading the socket. 

Eventually the broker reaches the keepalive timeout and kicks the client off and closes the TCP connection. Z2m doesn't notice this immediately and eventually hits the TCP timeout and reestablishes the connection. Topic data flows for about 5-10 seconds before locking up again. 

### What did you expect to happen?

When moving to the new broker, I expected z2m to pub/sub correctly and continue to work the same as other clients have done.

### How to reproduce it (minimal and precise)

start z2m and let it connect to an EMQX broker. Observe topic data initially and then stopping soon after. Eventually z2m ends up reconnecting and starting the cycle again.

### Zigbee2MQTT version

1.35.3-dev commit: 7c97902

### Adapter firmware version

20220219

### Adapter

sonoff zigbee 3.0 usb dongle plus

### Setup

Raspberry pi 3b with Ubuntu 20.04, z2m in docker-compose

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/14395865/log.txt)
[emqx@emqx-core-c8978cf8-0.emqx-headless.emqx-operator-system.svc.cluster.local-trace_zigbee2mqtt_2024-02-25.log](https://github.com/Koenkk/zigbee2mqtt/files/14395867/emqx%40emqx-core-c8978cf8-0.emqx-headless.emqx-operator-system.svc.cluster.local-trace_zigbee2mqtt_2024-02-25.log)

mundschenk-at, 2024-03-02:
Did you adjust the TCP network settings for your EMQX nodes according to the documented examples?

andersjohansson3, 2024-03-07:
I seem to have the same issue as @Audiobuzz. also recently moved both to EMQX and zigbee2mqtt.
zigbee2mqtt keeps dropping the connection to EMQX broker, filling up the log with
`Zigbee2MQTT:error 2024-03-07 23:07:32: MQTT error: read ECONNRESET
Zigbee2MQTT:info  2024-03-07 23:07:33: Connected to MQTT server
Zigbee2MQTT:info  2024-03-07 23:07:33: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:info  2024-03-07 23:07:35: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
Zigbee2MQTT:error 2024-03-07 23:07:35: MQTT error: write ECONNRESET`

@mundschenk-at can you point to these documented examples?

Audiobuzz, 2024-03-07:
> Did you adjust the TCP network settings for your EMQX nodes according to the documented examples?

I would have thought that TCP settings would affect both EMQX and mosquitto but I'm happy to give them a shot if you can point me to the required settings?

andersjohansson3, 2024-03-08:
I managed to get my zigbee2mqtt working again after a couple of re-installations, removing all config files and putting backup db and configuration back. What seemed to "fix" my issue for now was removing the locally stored icons from the configuration file. But I don't know how that could have been related with mqtt connection...

convicte, 2024-05-22:
I've faced very similar issues over the last to days, but thus far was unable to resolve them:
[image]
The coordinator being down for almost a day completely deteriorated my network in the process.
I seem to have eliminated all confounders, becuase both ESPHome, NodeRed, Nuki locks, etc. logged into the new coordinator under the same credentials as the old one just fine.
Z2m is constantly failing to connect to MQTT with an ECONNRESET error.
I maintained network sessions and passwords across MQTT brokers and everything but Z2m transitions seamlessly.

I'll make an attempt to remove the local icons, though I struggle understanding how this would help as well.

I'd greatly appreciate assistance as repairing 60+ devices is no fun....

blauhorn, 2024-05-24:
I am experiencing the same problem.
z2m runs on a raspberry pi, the MQTT broker on another one. 
Restarting the z2m -pi brought improvement.

convicte, 2024-05-24:
That is interesting to hear there is actually variability between deployments.
For me EMQX is running inside HA as an addon and both are a part of HA VM in Unraid.
I can't get Z2m to log in to EMQX MQTT at all - as seen above.

mpuff, 2024-07-03:
I moved also to emqx broker, now I get error at startup:
zigbee2mqtt z2m: MQTT error: write EPIPE

I only created a mqtt user in emqx like I had it before in mosquitto.
i will now change back to mosquitto and check if zigbee2mqtt is working again

convicte, 2024-07-03:
Same issue persists from the earlier reports.
I transitioned like this for testing multiple times and got the same error every time.

Works in mosquito and fails to work in emqx.

mpuff, 2024-07-03:
zigbee2mqtt is starting with mosquitto again

bartlebi, 2024-07-10:
I can confirm that removing the icon lines from devices.yaml solves the problem.

MrNonoss, 2024-08-08:
Another confirmation that removing icons from the configuration solved the issue.

jakobdyrby, 2024-08-17:
If you get "z2m: MQTT error: read ECONNRESET" try adjusting EMQX -> MQTT Settings -> Max Packet Size. That worked for me :)

CinciRyan, 2024-08-22:
I can confirm that like @jakobdyrby said, increasing the max packet size worked for me.  I found another article talking about it as well (https://github.com/emqx/emqx/issues/12527) and it mentioned changing it from 1M up to 256M .. that worked!
I also set Z2M to use protocol version 5 since that's what my EMQX is set to. For the MQTT integration, I left it set to 3.1.1 for the protocol version.  I did try setting it to version 5, but it will not connect.

Audiobuzz, 2024-09-10:
/facepalm moment for me!

Updating EMQX to the lates 1.40.1 and setting version: 5 in the config seems to have made things much more stable. This is without all the TCP tuning for now. but I'll monitor for a while and see how the stability goes.

D1ang, 2024-11-19:
I think he did that in zigbee2mqtt under:
settings -> MQTT

and then MQTT protocol version: set from 4 to 5.

[image]

closed 2025-01-27 by github-actions[bot]