Koenkk/zigbee2mqtt #21569: Crash while discovering multiple EGLO lights

saucekorn69, 2024-02-25:
### What happened?

After resetting 8 EGLO lights and turning them on, I am trying to add all the devices. It starts adding and some and crashes after a while the following debug message:

Zigbee2MQTT:info  2024-02-25 15:26:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13864607f205c","ieee_address":"0xa4c13864607f205c","status":"started"},"type":"device_interview"}'
Error: DatabaseEntry with ID '3' already exists
    at Database.insert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/database.ts:40:19)
    at Device.undelete (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/device.ts:430:25)
    at Controller.onDeviceJoined (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:557:20)

### Zigbee2MQTT version

1.35.3-dev commit: 4f71c1e8

### Adapter firmware version

6.10.3.0 build 297

### Adapter

POPP zigbee usb key

### Setup

Z2M on a DEBIAN 11 VM with broker / HA on NAS

### Debug log

Zigbee2MQTT:info  2024-02-25 15:26:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13864607f205c","ieee_address":"0xa4c13864607f205c","status":"started"},"type":"device_interview"}'
Error: DatabaseEntry with ID '3' already exists
    at Database.insert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/database.ts:40:19)
    at Device.undelete (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/device.ts:430:25)
    at Controller.onDeviceJoined (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:557:20)

saucekorn69, 2024-02-26:
Exatrct of database.db

{"id":3,"type":"Unknown","ieeeAddr":"0xa4c13864607f205c","nwkAddr":7217,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1708871377665}

Koenkk, 2024-02-26:
can you reproduce this? If yes, could you provide the herdsman debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/usage/debug.html) on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Koenkk, 2024-02-27:
- I don't see the `DatabaseEntry with ID 'X' already exists` in this log
- Can you check if the issue still occurs with the [new driver](https://github.com/Koenkk/zigbee2mqtt/discussions/21462)?

saucekorn69, 2024-02-27:
Yes I can check, I don't think I am running it, I'll let u know.

closed 2024-09-25 by github-actions[bot]