Koenkk/zigbee2mqtt #21570: Follow up of UNSUPPORTED_ATTRIBUTE Temp sensor ORVIBO ST30 #21541

saucekorn69, 2024-02-25:
### What happened?

I've updated Z2M and tried and again with the same device. See debug log

### Zigbee2MQTT version

1.35.3-dev commit: 4f71c1e

### Adapter firmware version

6.10.3.0 build 297

### Adapter

POPP zigbee usb key

### Setup

Z2M on a DEBIAN 11 VM with broker / HA on NAS

### Debug log

Zigbee2MQTT:error 2024-02-25 15:53:13: Failed to configure '0x5c0272fffef6d372', attempt 2 (Error: ConfigureReporting 0x5c0272fffef6d372/2 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:319:28)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:672:22)
    at Object.batteryPercentageRemaining (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:59:5)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/orvibo.ts:161:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:122:13)
    at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))
Error: DatabaseEntry with ID '6' already exists
    at Database.insert (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/database.ts:40:19)
    at Device.undelete (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/device.ts:430:25)
    at Controller.onDeviceJoined (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:557:20)

Koenkk, 2024-02-25:
Could you provide the data/database.db entry of this device?

saucekorn69, 2024-02-26:
Sure !

{"id":3,"type":"Unknown","ieeeAddr":"0xa4c13864607f205c","nwkAddr":7217,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1708871377665}

Should I make a cleanup of this file ?

Koenkk, 2024-02-26:
Should be fixed now!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

saucekorn69, 2024-02-27:
It works fine now, thanks !

closed 2024-02-27 by saucekorn69