Koenkk/zigbee2mqtt #21574: Aqara T1 Smart Ceiling Light

KuliKul, 2024-02-25:
### Link

https://www.aqara.cn/-Ceiling-Light-T1M_overview

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0x54ef4410008f0e17","nwkAddr":28925,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.light.acn031","epList":[1,2],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"61440":1,"modelId":"lumi.light.acn031","appVersion":26}},"manuSpecificLumi":{"attributes":{"128":{"type":"Buffer","data":[1,16,0,23,14,143,0,16,68,239,84,253,112,0,0,0,0,1]},"223":{"type":"Buffer","data":[22,33,2,0,23,33,0,0,12,33,3,0,24,33,0,0]},"229":{"type":"Buffer","data":[2,0,2,0,0,36,141,56,213,5,103]},"247":{"type":"Buffer","data":[117,32,254,115,35,153,153,204,76]},"65522":{"type":"Buffer","data":[170,117,19,132,9,65,7,65,0,15,12,35,1,1,0,2,10,35,8,0,12,100,11,32,2]}}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{"lightingColorCtrl":{"attributes":{"245":40959232,"currentX":19660,"currentY":39321}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":26,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1708874171751}

### Comments

I tried the add new unsupported devices, added successfully but cannot control it. The respond for everytime I try to control it as No converter available for 'state' (""). 

The js file as below:

### External definition

```shell
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {} = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.light.acn031'],
    model: 'HCXDD11LM',
    vendor: 'Aqara',
    description: 'Aqara Ceiling Lamp T1',
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;
```

Koenkk, 2024-02-25:
Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) instead?

tomknocke, 2024-03-01:
Had anybody success with that device? Actually i do not own that light, but seems to very interesting to get one in case it will be supported. Thanks

TooBri, 2024-03-02:
> Could you try with the [generated external converter](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-creating-the-external-definition) instead?

const {deviceEndpoints, light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['lumi.light.acn032'],
    model: 'lumi.light.acn032',
    vendor: 'Aqara',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), light({"colorTemp":{"range":[153,370]}}), light({"colorTemp":{"range":[153,370]},"color":true})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;

corvis, 2024-04-05:
Most likely the support for this device was added by this PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/7203

closed 2024-11-03 by github-actions[bot]